nextcloud/nextcloudpi #1485: Ownyourbits Down #1053

imk13, 2022-06-05:
**System information**

[image]

`nslookup www.ownyourbits.com`

```
;; Got SERVFAIL reply from 8.8.8.8, trying next server
Server:		1.1.1.1
Address:	1.1.1.1#53

** server can't find www.ownyourbits.com: SERVFAIL
```

ben569, 2022-06-06:
Can confirm. Is someone on the case?

riff-79, 2022-06-07:
Can confirm still down.

GNEUGNEUGNEU, 2022-06-07:
yup, down

pilepalle, 2022-06-09:
down.
Does anyone know if ownyourbits.com stays down?
If so, where can I get the RPI and odroid images?

theCalcaholic, 2022-06-10:
Ownyourbits is down, because there has been a misunderstanding with the hoster. I don't know if it will be restored.

However, nextcloudpi.com will be migrated to a new hosting solution as soon as I can get to it. Images have been regenerated and can now be found in the GH release artifacts: https://github.com/nextcloud/nextcloudpi/releases/tag/v1.47.2

closed 2022-06-10 by theCalcaholic

sunjam, 2022-06-15:
Documentation from the dead `docs.nextcloudpi.com` has migrated to the forum:

https://help.nextcloud.com/c/ncpdocs/178