nextcloud/nextcloudpi #1486: Open ports check fail due to a bug in ncp-diag file

Soiuke, 2022-06-05:
ncp-diag file

Line 79

Add "/check" after "https://portchecker.co"

Otherwise port check 80 and 443 anwser "closed" erroneously (even when these ports are correctly opened and working)

Congratulations for you exelent work and thanks to you and all ncp comunity.

dd900, 2022-06-19:
Can confirm making this edit fixes the issue of NextCloudPi Panel showing ports closed even though they were open.

schoetju, 2022-07-10:
Can confirm the bugfix too.
@Soiuke Do you mind to create a fork plus a corresponding pull-request?

BTW: This issue is the same as https://github.com/nextcloud/nextcloudpi/issues/1475 , isn't it?

Oberlurch3000, 2022-07-16:
I can also confirm that solved the issue for me. Btw: No spacebar between "https://portchecker.co" and "/check" => "https://portchecker.co/check"

theCalcaholic, 2022-07-17:
Thank you for the additional info!

I'm going to close this issue though, because it's a duplicate of #1475

closed 2022-07-17 by theCalcaholic

spiritualcrab, 2022-07-22:
> ncp-diag file
> 
> Line 79
> 
> Add "/check" after "https://portchecker.co"
> 
> Otherwise port check 80 and 443 anwser "closed" erroneously (even when these ports are correctly opened and working)
> 
> Congratulations for you exelent work and thanks to you and all ncp comunity.

I am sorry, but how to find this file in order to edit it?

LittleAlf, 2022-07-22:
/usr/local/bin/ncp-diag

Soiuke, 2022-07-27:
> Can confirm the bugfix too. @Soiuke Do you mind to create a fork plus a corresponding pull-request?
> 
> BTW: This issue is the same as #1475 , isn't it?

@schoetju I've not created a fork and pull-request because I think that this is a "minor" issue, which does not affect the main functionality of the project. As far as @theCalcaholic knows about the issue and also the proposed solution, I think that he can modify ncp-diag file in the main project/branch.

Otherwise please tell me if I must open a fork and pull-request in order to incorporate the solution to the main project.

By the other hand, I apologize for duplicating issues.

Thanks to all the ncp community. Congratulations to @nachoparker for his great work during last years, and thanks to @theCalcaholic for leading the project from now on.