nextcloud/nextcloudpi #1487: v1.47.2 image reporting

theCalcaholic, 2022-06-08:
Please report here if you have been able to support any of the images from the v1.47.2 release (successfully or not). :)

kobruc, 2022-06-09:
Hi! I am using ncp version 1.47.2 and would like to help with supporting this project. My server is productive with 4 users and one admin user

kobruc, 2022-06-09:
I using a raspberry 4 pi with 8 gb ram and one USB-SSD where the ncdata lockated. 
php 7.4.28 (cli) (built: Feb 17 2022 16:17:19) ( NTS )

Sorry my english is not so gut! Can i write on german to?

theCalcaholic, 2022-06-09:
@kobruc Thank you :)

Ja, deutsch ist auch okay. Hast du Version 1.47.2 neu als Image installiert oder von einer älteren Version geupdatet?

theCalcaholic, 2022-06-09:
Okay, bei diesem Issue geht es um Probleme mit dem [aktuellen Image](https://github.com/nextcloud/nextcloudpi/releases/tag/v1.47.2). Das wird nur relevant, wenn man das direkt einspielt.

kobruc, 2022-06-09:
Kein Problem. Ich spiele es direkt ein, nur bitte ein paar Hinweise wie und auf was und ein Link von wo ich das image downloaden soll weil ownyourbits ist offline.

ps. habe die rc* image schon gefunden 
Lg

theCalcaholic, 2022-06-10:
Ich denke, die beste Anleitung findet sich in der nextcloudpi Dokumentation - die leider auch gerade nicht erreichbar ist.

Glücklicherweise, kann man sie im Web Archive noch finden:

https://web.archive.org/web/20220426010500/https://docs.nextcloudpi.com/en/how-to-install-nextcloudpi/

kobruc, 2022-06-10:
Danke schön ! Ich installiere es gerade, werde hier alles berichten. 
Vielen Dank für deine Unterstützung!

kobruc, 2022-06-10:
Also ich habe von einem mikro SD Karte installiert ohne Probleme. Dann habe versucht den Wizard auszuführen ohne Erfolg.
nc-automount und nc-format-USB hat funktioniert aber "Move data tu USB" ging nicht mehr: orange Farbe ... Habe mehrere male versucht, neugestartet noch 4 mal versucht dann habe ich noch ein Update durchgeführt und auch eine kleinere 125 GB USB Stick versucht auch erfolgloos. Erste USB war eine 256 GB SanDisk Stick beide ohne externe Stromquelle (vielleicht das ist das Problem) Also soweit bin ich gekommen!
[image]

kobruc, 2022-06-10:
So, jetzt habs hinbekommen. Weiß nicht was war, plötzlich hat die USB akzeptiert. 

Also die Installation lauft bis jetzt stabil, auf ein raspberry pi 4 mit 4 GB Ram und ein USB als Datenspeicher mit 256 GB.
habe auch ein duckdns auch aktiviert. Ich würde es jetzt gerne über ein nginx proxy manager laufen lassen und dort lastencrypt certifikat auch erstellen. Oder soll ich es über Webpanell erstellen zuerst?

theCalcaholic, 2022-06-10:
@kobruc VIelen Dank für die Rückmeldung! Den Fehler bzgl. Imagick kannst du ignorieren, das ist Absicht (das Paket ist anfällig für Sicherheitslücken). Für Support mit Spezialkonfigurationen (reverse proxy, etc.) wende dich bitte an das Nextcloud Forum. Dort gibt es auch eine ncp-Untersektion: https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33

kobruc, 2022-06-10:
@theCalcaholic Kein Problem, mache ich sehr gern jetzt habe ich gerade Zeit und es macht auch Spass.

Inzwischen habe ich imagick und noch ein Fehler mit „default_phone_region“ auch behoben. Jetzt steht bei der Übersicht ein grüner Punkt und daneben: Alle Überprüfungen bestanden. Version von ncp ist 23.0.2

Was soll ich nocht testen bitte? Mache gerne alles was du verlangst

theCalcaholic, 2022-06-10:
Super :) Ich empfehle aber wirklich, den Fehler mit imagick *nicht* zu "beheben". Ist ein Sicherheitsrisiko.

kobruc, 2022-06-10:
> @theCalcaholic danke für den Hinweis. 

Wie soll ich jetzt imagick wieder entfernen?

dvpfig, 2022-06-14:
I'm using Odroid HC4 but I use a container to run NCP in my system. 

Can you upload this version on arm64 to Docker Hub?  
https://hub.docker.com/r/ownyourbits/nextcloudpi-arm64/tags

From there I could get it and test.

sunjam, 2022-06-15:
@theCalcaholic do you have control over the Docker Hub currently under ownyourbits/nextcloudpi?

https://hub.docker.com/r/ownyourbits/nextcloudpi/

kobruc, 2022-06-21:
@theCalcaholic  weist du vielleicht was kann das bedeuten ich habe ein paar mal die NextCloudPi_RPi_1472_06-08-22.tar.bz2 das ich von dir bekommen habe erfolgreich installiert. Jetzt seit ein paar Tagen geht nicht mehr bekomme immer die Meldung: **"NextCloudPi not yet initialized, trying again in a few seconds ..."** und es tut sich nitchts. Habe neue SD Karte genommen dan habe auch auf mit einen Mac versucht. 

Es bleibt immer hängen mit die Meldung: NextCloudPi not yet initialized, trying again in a few seconds ... ich versuche es wieder zi aktivieren, bekomme neue Passwörter aber es aktiviert sich nicht.

dvpfig, 2022-07-01:
> @dvpfig @sunjam Unfortunately, I have no access to nacho's docker repository at this time. I'll talk to him for a solution

Image for version 1.47.2 is available in DockerHub, thanks. 

Unfortunately, I did a fast test to change images from 1.46.9 to 1.47.2 and it did not work properly to start the Nextcloud instance.

My machine is Odroid HC4. 
I did not see any errors in the Logs of the NCP docker container. 
The NCP web manager is accessible, doesn't show any error. 
But the Nextcloud is unreachable. I kept the same configuration in a persistant docker volume, so there were no changes between image versions 1.46.9 and 1.47.2. 

I'll need to investigate deeper what is happening. But I wonder if there is need to change any configuration to run on 1.47.2?

Also, which Nextcloud version does NCP support? does it support Nextcloud 24?

Thanks, and hope this project continues to be supported. It has been a life-saviour so far to keep the Nextcloud instance up-and-running.

martadinata666, 2022-07-10:
the `:latest` tag only amd64 as the 1.47.2 correctly had multiarch, something missing on manifest build? or it intended?

theCalcaholic, 2022-07-12:
@dvpfig You should not have to change anything when updating to v1.47.2. However, the update process by swapping images is not well tested for ncp (except for special updates, like updating to a new version of Debian). You should use ncp auto updates (activate from `ncp-config`) instead.

theCalcaholic, 2022-07-24:
closed now that v1.48.2 is out

closed 2022-07-24 by theCalcaholic