nextcloud/nextcloudpi #1488: nextcloudpi website ist not available

Marvout, 2022-06-11:
Sorry to bother, but i cannot reach the nextcloudpi website.
So at the moment i just downloaded nextcloudpi from github,
but anything else doesnt work. Does anyone have the same problems ?

tflidd, 2022-06-11:
Yes, they work on it: https://github.com/nextcloud/nextcloudpi/issues/1485

sunjam, 2022-06-15:
Documentation has migrated from the dead `docs.nextcloudpi.com` to the forum.

https://help.nextcloud.com/c/ncpdocs/178

ajdste, 2022-06-20:
> Documentation has migrated from the dead `docs.nextcloudpi.com` to the forum.
> 
> https://help.nextcloud.com/c/ncpdocs/178

Thanks!  :)
But ...

I just tried there. The only pages resolving For Me are these:
"About" (https://help.nextcloud.com/t/about-the-nextcloudpi-documentation-category/125977)
"Index" (https://help.nextcloud.com/t/index-nextcloudpi-documentation/126448)

The links on the "Index" page brings up this:
"Oops! That page doesn’t exist or is private."

I am signed in on my Nextcloud account.

Anyone have another idea?  :)

sunjam, 2022-06-20:
> I just tried there. The only pages resolving For Me are these:

Thank you for the feedback.  I'm going through and publishing any pages that are still in private draft.  I'm not finding a lot of them, but happy to correct.  Who are you on the forum?  Truth is no one else has given any feedback yet, so it really helps.  Let me know your username there and I'll add you to the document group if you are okay with that.

ajdste, 2022-06-21:
Hey Sunjam ~~

Thank you for fixing the NextCloudPi Documentation! It works now.  :)

https://help.nextcloud.com/t/index-nextcloudpi-documentation/126448

Cheers!
aj

aadithya96, 2022-06-25:
@sunjam It still doesn't work for me. All links either resolve to the FAQ or the "How and where to  get help" section.

rababerladuseladim, 2022-07-02:
The dead domain is still contained in the Readme and wiki. Should these be updated or donyou want to wait until you have a new hosting solution?

theCalcaholic, 2022-07-07:
https://www.nextcloudpi.com is now available again. We're working on https://nextcloudpi.com and docs.nextcloudpi.com

closed 2022-07-07 by theCalcaholic