nextcloud/nextcloudpi #1489: Issue with Nextcloud 23.0.2

paco192002, 2022-06-14:
Hello all!

This is my first time submitting a report on GitHub, so in case you need more info, just ask for it. Thanks!

I've been using this system since a year ago, I have a USB SSD for the OS, and a 14TB USB HDD, connected to a Raspberry Pi 4 4GB.
System was working since 2 months ago, when the USB HDD crashed and lost the file system, so I recovered it and re-format it. 

Until then, everything is Ok, issue now is, that after using ncp config to install a Nextcloud, and doing a ncp init, to reinitialize everything, it doesn't come to live, with a big chunk of error, wich I think it may come due to a bug on the reinstallation process.
The issue is triggered on any try of login on the Nextcloud service, either if its right or wrong, it will prompt an internal server error.
Domain is https://paconube.ddns.net if someone wants to try by itself!
This is a personal project and just did it for "fun" and home and friends use, 

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.47.2
NextCloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 11. 5.15.32-v8+ (aarch64)
automount            yes
USB devices          sda sdb
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.8G/220G
rootfs usage         2.8G/220G
swapfile             /var/swap
dbdir                /var/lib/mysql/2
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "2a0c:5a82:1401:9700:52a9:9f27:2cb8:6c23",
            "1": "192.168.1.134",
            "14": "nextcloudpi",
[... 94 lines cut ...]

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"pMSxJg85HM27E4SI1e4j","level":3,"time":"2022-06-14T13:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":649,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1087,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":277,"CustomMessage":"--"}}
{"reqId":"YqiGhHxYlCm9ikTKFn3MPwAAAxI","level":2,"time":"2022-06-14T13:00:52+00:00","remoteAddr":"79.116.8.108","user":"--","app":"no app in context","method":"GET","url":"/index.php/csrftoken","message":"Could not detect any host in https:///data/htaccesstest.txt","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.39","version":""}
{"reqId":"YqiGhHxYlCm9ikTKFn3MPwAAAxI","level":2,"time":"2022-06-14T13:00:52+00:00","remoteAddr":"79.116.8.108","user":"--","app":"no app in context","method":"GET","url":"/index.php/csrftoken","message":"Could not detect any host in http:///data/htaccesstest.txt","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.39","version":""}
{"reqId":"OO3t1dFGxf5s0sUkdyVd","level":3,"time":"2022-06-14T13:05:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":649,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1087,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":277,"CustomMessage":"--"}}
{"reqId":"FMTA67ufqS2d9J3g3HFl","level":2,"time":"2022-06-14T13:09:08+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 60000 milliseconds with 1626122 bytes received (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://apps.nextcloud.com/api/v1/apps.json","userAgent":"--","version":"23.0.2.1"}
{"reqId":"YqiL8M3JTTvii4krGwthcAAAjhI","level":3,"time":"2022-06-14T13:24:00+00:00","remoteAddr":"79.116.8.108","user":"--","app":"PHP","method":"POST","url":"/index.php/login","message":"Unknown: POST Content-Length of 188 bytes exceeds the limit of 100 bytes at Unknown#0","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.39","version":"23.0.2.1"}
{"reqId":"YqiMX3xYlCm9ikTKFn3MbAAAEhE","level":3,"time":"2022-06-14T13:25:52+00:00","remoteAddr":"79.116.8.108","user":"--","app":"index","method":"POST","url":"/index.php/login","message":"Argument 1 passed to OC\\Core\\Controller\\LoginController::tryLogin() must be of the type string, null given, called in /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php on line 217 in file '/var/www/nextcloud/core/Controller/LoginController.php' line 309","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.39","version":"23.0.2.1","exception":{"Exception":"Exception","Message":"Argument 1 passed to OC\\Core\\Controller\\LoginController::tryLogin() must be of the type string, null given, called in /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php on line 217 in file '/var/www/nextcloud/core/Controller/LoginController.php' line 309","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":157,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":302,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1006,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","Line":158,"Previous":{"Exception":"TypeError","Message":"Argument 1 passed to OC\\Core\\Controller\\LoginController::tryLogin() must be of the type string, null given, called in /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php on line 217","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":217,"function":"tryLogin","class":"OC\\Core\\Controller\\LoginController","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":126,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":157,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":302,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1006,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/core/Controller/LoginController.php","Line":309},"CustomMessage":"--"}}
[... cut ...] r.php","Line":309},"CustomMessage":"--"}}
{"reqId":"YqiT_4uglXEX4mY16WN9ewAABwI","level":3,"time":"2022-06-14T13:58:24+00:00","remoteAddr":"79.116.8.108","user":"--","app":"PHP","method":"POST","url":"/index.php/login","message":"Unknown: POST Content-Length of 175 bytes exceeds the limit of 1 bytes at Unknown#0","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.63 Safari/537.36 Edg/102.0.1245.39","version":"23.0.2.1"}

```
</details>

theCalcaholic, 2022-07-16:
I'm not entirely sure, but it looks like your database might be corrupted. Do you have a recent backup that you can restore? If so, that would be the best course of action

theCalcaholic, 2022-07-21:
Closed due to inactivity. Feel free to comment and I'll open it again.

closed 2022-07-21 by theCalcaholic