nextcloud/nextcloudpi #1492: Distribution of images via Torrents

sunjam, 2022-06-30:
**Description**  
Developing a simple and reproducible distribution of Github releases for images as torrents with checksums.  See #1487 as an example.
 
**Help Needed**  
Anyone willing is welcome to help setup a simple and trustworthy system for automatically distributing images to a torrent tracker such as https://linuxtracker.org/

REAPERSbattlecry, 2022-07-01:
I created the torrents files of the current images (v. 1.47.2-rc2) and posted the [here](https://help.nextcloud.com/t/magnetlinks-for-torrentdownload-ncp/141510).  The nc forum doesn't allow .torrent files. Therefor just the magnet links.

The torrent include the image and a textfile with the checksum.
I tested the checksum before creating the torrent.

At the moment there is no scripted process, just me ;).

theCalcaholic, 2022-07-07:
@REAPERSbattlecry Thank you for your help. The issue with that is, that I can't endorse any torrents if I have no control over their creation. So I can't guarantee NCP users that they're safe to use.

That's why we would need an automated, transparent system which produces these issues

REAPERSbattlecry, 2022-07-09:
Haven't thought about that. My thought was "just the root of trust changed". It was nachoparker and now it is someone else. Everyone can check with the checksum the origin of the files. 
But so far there is no download and more important no second seeder.
I would love to see the possibility of torrent downloads but I'm unsure if there is a target audience for it ;).

sunjam, 2022-07-13:
> Haven't thought about that.

This isn't arbitrary @REAPERSbattlecry 
Re-read the original post here with the understanding this it is the only option the maintainer would consider.  It was written based on exact wording and previous conversation of what can be accepted.

Appreciate your interest, but it needs to follow the posted gameplan.  :smile_cat:

REAPERSbattlecry, 2022-07-13:
@sunjam : I'm totally fine with that and I can understand the requirements totally.
So no offence taken. 
Just wanted to share my thoughts so far.

theCalcaholic, 2022-07-16:
> By the power of an internet search:
> 
> Seems github actions allow a torrent to be automatically triggered and generated from a github release. [Details here](https://dev.to/animmouse/convert-direct-download-links-into-torrent-using-github-actions-4bng)
> 
>     * project repo - https://github.com/AnimMouse/torrent-webseed-creator

Yes, I know they do. However, it's currently not a priority for me to implement it. :)
My priorities right now are:

1. Fix image builds
2. Automate image builds
3. Update to PHP 8 and NC 24