nextcloud/nextcloudpi #1499: zram service failes since upgrade to bullseye

Vargblood, 2022-07-11:
Hi everyone,

since i have upgraded my ncp to bullseye zram service is failing. I think it's a small issue in the config file.

`
zram.service - Set up ZRAM
     Loaded: loaded (/etc/systemd/system/zram.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-07-11 07:52:58 CEST; 1h 10min ago
    Process: 506 ExecStart=/usr/local/bin/ncp-zram start (code=exited, status=1/FAILURE)
   Main PID: 506 (code=exited, status=1/FAILURE)
        CPU: 152ms

Jul 11 07:52:58 Server systemd[1]: Starting Set up ZRAM...
**Jul 11 07:52:58 Server ncp-zram[506]: /usr/local/bin/ncp-zram: Zeile 12: ( / 4)* 1024 : Syntaxfehler: Operator erwartet. (Fehlerverursachendes Zeichen ist "/ 4)* 1024>**
Jul 11 07:52:58 Server systemd[1]: zram.service: Main process exited, code=exited, status=1/FAILURE
Jul 11 07:52:58 Server systemd[1]: zram.service: Failed with result 'exit-code'.
Jul 11 07:52:58 Server systemd[1]: Failed to start Set up ZRAM.

`
Thank you.

mh254, 2023-02-19:
Same error on my Pi, any fix available?

mh254, 2023-02-19:
Ok,in my case it was the language.

Line 12 uses grep -e "^Mem:", but my language is German therefore it must be Speicher instead of Mem and now the service is working again.