nextcloud/nextcloudpi #1501: Nextcloudpi unreachable after update

coppens2022, 2022-07-14:
After Nextcloudpi was updated automatically to v1.4.8.1, I can no longer connect to the service, nor internally, nor via DNS.

theCalcaholic, 2022-07-14:
Do you still have ssh or physical access? If so, please attache the output of `ncp-report` to this issue

coppens2022, 2022-07-15:
Overnight, nextcloud updated to 23.0.6.1 and now, the problem seems to be solved. Thank you anyway!

theCalcaholic, 2022-07-16:
@coppens2022 Glad to hear that your problem is solved, though I am still wondering what might have caused it.

@TheShanMan Sounds like an entirely different issue, so I'm going to close this one. 

Still, according to your log, you're database is turned off and NC is in maintenance mode. Can you please run the following commands and attach another `ncp-dist-upgrade`?

```sh
sudo systemctl start mysql
sudo ncc maintenance:mode --off
sudo ncp-report
```

closed 2022-07-16 by theCalcaholic

TheShanMan, 2022-07-16:
[ncp-report.txt](https://github.com/nextcloud/nextcloudpi/files/9126487/ncp-report.txt)

I can access nextcloud now, although the nextcloudpi system info page still reports 80 and 443 as closed. And it's probably not related, but with httpsonly disabled, I'm not able to log in via 80 - it still switches me to https.

Thank you!

theCalcaholic, 2022-07-17:
@TheShanMan The former is indeed unrelated (see #1486) and the latter is expected, because that's a security feature of the TLS certificates (called "HSTS"). It causes your browser to only accept TLS connections if it has every connected to the website via TLS. To connect to port 80, you need to delete the site from the browsers history (e.g. on FIrefox: Right-click in History and choose "Forget about this site")