nextcloud/nextcloudpi #1506: "ncp install error update-rc.d" when running Debian bash installer

BstAA, 2022-07-17:
<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-version: Datei oder Verzeichnis nicht gefunden
Nextcloud is not installed - only a limited number of commands are available
NextCloudPi version
OS                   Debian GNU/Linux 11. 5.10.0-16-amd64 (x86_64)
automount            no
USB devices          none
datadir               (doesn't exist)
rootfs usage         2,0G/19G
swapfile             /dev/sda5
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            ens33
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               14min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun Jul 17 17:41:08.556356 2022] [mpm_event:notice] [pid 2620:tid 139985404853568] AH00489: Apache/2.4.54 (Debian) configured -- resuming normal operations
[Sun Jul 17 17:41:08.556447 2022] [core:notice] [pid 2620:tid 139985404853568] AH00094: Command line: '/usr/sbin/apache2'
[Sun Jul 17 17:43:16.711617 2022] [mpm_event:notice] [pid 2620:tid 139985404853568] AH00492: caught SIGWINCH, shutting down gracefully
[Sun Jul 17 17:43:16.768995 2022] [mpm_event:notice] [pid 19885:tid 140240971205952] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Sun Jul 17 17:43:16.769020 2022] [core:notice] [pid 19885:tid 140240971205952] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: '/var/log/mysql/*.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: '/nextcloud.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden

```
</details>

-->

**System information**
Debian 11.4 fresh Text-only Install
Install via bash Installer

Error during install:
```
+ update-rc.d systemd-resolved enable
update-rc.d: error: cannot find a LSB script for systemd-resolved
```
see enclosed ncp-report.

theCalcaholic, 2022-07-17:
Sorry and thanks for reporting! Should be fixed now

closed 2022-07-17 by theCalcaholic

BstAA, 2022-07-17:
THX! Tested and running :-) 
Reason? Worked for me yesterday. 
Klaus

theCalcaholic, 2022-07-17:
Reason: I broke the installer today while trying to fix it for another scenario (Raspberry Pi image generation) :P