nextcloud/nextcloudpi #1508: cannot restore a snapshot

bassman56, 2022-07-21:
I use version 1.48.2. I can create manual snapshots, but I cannot restore them. This is the output of the restore operation (I inserted a echo directive before the failing command).

[ nc-restore-snapshot ] (Thu Jul 21 20:53:00 CEST 2022)
Create a readonly snapshot of '/mnt/nccloud/ncdata' in '/mnt/nccloud/ncdata/../ncp-snapshots/autobackup_2022-07-21_205301'
snapshot autobackup_2022-07-21_205301 generated
Maintenance mode enabled
btrfs subvolume delete /mnt/nccloud/ncdata/data
ERROR: Not a Btrfs subvolume: Invalid argument

Problem: the subvolume is actually /mnt/nccloud/ncdata, NOT /mnt/nccloud/ncdata/data. so the script seems to determine the subvolume of the data directory wrongly?

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.48.2
NextCloudPi image    NextCloudPi_07-18-22
OS                   Debian GNU/Linux 11. 5.10.0-16-amd64 (x86_64)
automount            no
USB devices          sda 
datadir              /mnt/nccloud/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      140M/977G
rootfs usage         8,3G/46G
swapfile             /var/swap
dbdir                /var/lib/mysql/
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp89s0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:52

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "3": "nextcloudpi",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "11": "92.117.160.52",
            "1": "192.168.1.112",
[... 88 lines cut ...]
```

{"reqId":"avuGEQhq6MBVN9ZR4TMc","level":2,"time":"2022-07-20T16:14:46+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/data/tmp is not present or writable","userAgent":"--","version":"23.0.2.1"}
{"reqId":"YtgrhsfgeVNuN32fEseZUwAABgI","level":4,"time":"2022-07-20T16:21:26+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"webdav","method":"REPORT","url":"/remote.php/dav/files/ncp/","message":"Das System befindet sich im Wartungsmodus.","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:91.0) Gecko/20100101 Firefox/91.0","version":"23.0.2.1","exception":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":339,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"}}
{"reqId":"Ytgri8fgeVNuN32fEseZVgAABwI","level":4,"time":"2022-07-20T16:21:31+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Das System befindet sich im Wartungsmodus.","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:91.0) Gecko/20100101 Firefox/91.0","version":"23.0.2.1","exception":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":339,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"}}
{"reqId":"YtgrjsfgeVNuN32fEseZWQAACwI","level":4,"time":"2022-07-20T16:21:34+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Das System befindet sich im Wartungsmodus.","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:91.0) Gecko/20100101 Firefox/91.0","version":"23.0.2.1","exception":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":339,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"}}
{"reqId":"EKz0CuJpi2CQqY57ShU5","level":3,"time":"2022-07-20T17:00:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"stream_socket_enable_crypto(): Peer certificate CN=`nextcloud2.fritz.box' did not match expected CN=`127.0.0.1' at /var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/StreamBuffer.php#94","userAgent":"--","version":"23.0.2.1","exception":{"Exception":"Error","Message":"stream_socket_enable_crypto(): Peer certificate CN=`nextcloud2.fritz.box' did not match expected CN=`127.0.0.1' at /var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/StreamBuffer.php#94","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/StreamBuffer.php","line":94,"function":"stream_socket_enable_crypto"},{"file":"/var/www/nextcloud/3rdparty/swiftmailer/swiftmailer/lib/classes/Swift/Transport/EsmtpTransport.php","line":348,"function":"startTLS","class":"Swift_Transport_StreamBuffer","type":"->"},{"file":"/var/www/nextcloud [... cut ...] extcloud/apps/dav/lib/Server.php","line":339,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"}}

```
</details>

theCalcaholic, 2022-07-21:
Your datadirectory seems to be ill configured.

Can you check `sudo ncc config:system:get datadirectory` to ensure that it is `/mnt/nccloud/ncdata` and, if it isn't, correct that with `sudo ncc config:system:set datadirectory --value=/mnt/nccloud/ncdata`?

bassman56, 2022-07-21:
Hi Tobias,

thanks for the quick reaction.

Actually sudo ncc config:system:get datadirectory results in
/mnt/nccloud/ncdata/d ata, although I specified /mnt/nccloud/ncdata when
setting the data directory in the ncp-web interface.

I applied your proposal ncc config:system:set datadirectory
--value=/mnt/nccloud/ncdata, then had to manually move all file 1 level up
from /mnt/nccloud/ncdata/data/ to /mnt/nccloud/ncdata, after that the
nextcloud works again.

 

Now the system works fine. I created some snapshots, added a test file to a
user directory, restored to the snapshot a few minutes earlier, and the test
file was gone! :)

Thanks for the help!

 

But where is the root cause? So nc-datadir might be broken adding a /data
to the path? (I would not like to try too much in order not to kill my
nextcloud..)

Thanks again

Christian

 

<https://github.com/nextcloud/nextcloudpi/issues/1508#issuecomment-119185010
GPSVANCNFSM54IWYQFA> .
<https://github.com/notifications/beacon/AAGO6UUDEO3OL7FW3FW4PTTVVGPSVA5CNFS
M54IWYQFKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOI4FDA6Q

bassman56, 2022-07-22:
not fully understood but working

closed 2022-07-22 by bassman56