nextcloud/nextcloudpi #1513: Broken Packages

glenn-sorrentino, 2022-07-27:
I'm installing NCP on a Pi4 with a fresh Raspberry Pi OS. Install fails when dependencies are failed to download: 

-----

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 smbclient : Depends: samba-libs (= 2:4.13.13+dfsg-1~deb11u3) but 2:4.13.13+dfsg-1~deb11u4 is to be installed
             Depends: libwbclient0 (= 2:4.13.13+dfsg-1~deb11u3) but 2:4.13.13+dfsg-1~deb11u4 is to be installed
E: Unable to correct problems, you have held broken packages.

theCalcaholic, 2022-07-27:
At first glance that sounds like an upstream issue... How are you installing NCP? Are you using the official raspberry pi image from https://github.com/nextcloud/nextcloudpi/releases or are you installing with the curl installer? In the latter case: What image are you using as base?

Id3aFly, 2022-07-27:
Same for me, I am using a clean and fresh Debian installation and run the script as provided in the Readme:  

`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`

Id3aFly, 2022-07-27:
I am using latest Debian11 with all packages upgraded.

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

glenn-sorrentino, 2022-07-27:
Same as @Id3aFly - using curl. Downloading the RPi image from your link now.

> At first glance that sounds like an upstream issue... How are you installing NCP? Are you using the official raspberry pi image from https://github.com/nextcloud/nextcloudpi/releases or are you installing with the curl installer? In the latter case: What image are you using as base?

Same as @Id3aFly - using curl. Downloading the RPi image from your link now.

Id3aFly, 2022-07-27:
I installed the correct packages now manually.

sudo apt install libwbclient0=2:4.13.13+dfsg-1~deb11u3
sudo apt install samba-libs=2:4.13.13+dfsg-1~deb11u3

glenn-sorrentino, 2022-07-27:
The image in the link above fixed the install issue 👍

closed 2022-07-27 by glenn-sorrentino

Id3aFly, 2022-07-27:
What image do you mean?

glenn-sorrentino, 2022-07-27:
> What image do you mean?

https://github.com/nextcloud/nextcloudpi/releases 
scroll to Assets, download and unzip

Id3aFly, 2022-07-27:
Ah, I see. I was thinking of image as visual image as jpg. ;)

theCalcaholic, 2022-07-28:
@Id3aFly @glenn-sorrentino Please give me more information on what base images you have been using (@Id3aFly e.g. the one from https://www.debian.org/download or are you using any kind of hosting provider with prepared images; @glenn-sorrentino Have you been using Raspbian or something else)? Otherwise I can't try to reproduce this issue or look into the cause.

Id3aFly, 2022-07-28:
@theCalcaholic Yes, I just took the debian-11.4.0-amd64-netinst.iso from the debian page. Then I installed it via VirtualBox.

theCalcaholic, 2022-07-28:
@Id3aFly I couldn't reproduce this issue with the latest installer. Could you try again with `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/devel/install.sh | BRANCH=devel bash`?

Id3aFly, 2022-08-01:
RIght now I don't have my device with me and can only reproduce at some point in the future. I'll try then out for sure.

Id3aFly, 2022-09-09:
Just as a comment, even if its already quite late: Install works with latest ncpi version without any issues ;) Thank you!