nextcloud/nextcloudpi #1516: Odroid HC2 "NextCloudPi_OdroidHC2_07-25-22.tar.gz" no function

pilepalle, 2022-07-27:
Hi all, I just tried to rebuild my cloud platform Odroid HC2 with "NextCloudPi_OdroidHC2_07-25-22.tar.gz". Unfortunately this fails as I can't get to the activation page. With an old version "NextCloudPi_OdroidHC2_01-08-20" it works without problems. Am I doing something wrong here or does the current image not work with the Odroid HC2?

theCalcaholic, 2022-07-28:
Hi, thanks for reporting. I'll try to check the current image (that's not so easy as we don't yet have a reliable process of checking the images without the required hardware). To help me out, you could verify that the checksum of your downloaded image matches the one from the release page.

In the meantime, if an older image works for you, you can just use that and update to the latest version. However, in the versions before, there was a bug when moving the data directory, so make sure to update first and then move your data directory.

pilepalle, 2022-07-28:
Hi, the hash matches the one on the website. 529cee648b5f621cd5bbf398a8fa56ef

theCalcaholic, 2022-07-28:
Thank you, that probablz means, the image is indeed broken

pilepalle, 2022-08-05:
Hi, is there any information about when a new working image for the HC2 will be available? 
I can offer you my replacement Odroid XU4 for testing. How can I send it to you? Are you from Germany?

theCalcaholic, 2022-08-05:
@pilepalle That would actually amazing. I suspect there to be an issue with all the 32bit images that might be an upstream issue with armbian - but this theory is hard to verify for me without any hardware to test on.
I'm working on tests that don't rely on the correct hardware being available - but unsurprisingly they fail (probably because the images are broken). 

Yes, I'm from Germany. Are you in the NCP telegram or matrix group? You could contact me there (I'm theCalcaholic on both platforms) in order to receive my address.

theCalcaholic, 2022-08-17:
FYI: As 32bit is less and less supported by Nextcloud and I have trouble building 32bit armbian images at the moment, have decided to drop support for the prebuilt armbian images for now. Depending on how much echo this will cause, I might reconsider that decision.

It is, of course, still possible to install NCP via curl installer on those devices.

closed 2022-08-17 by theCalcaholic

pilepalle, 2022-11-21:
Hello, is there still an image somewhere that works on the Odroid HC2. It doesn't have to be the latest one. I would just like to continue using my two Odroid HC2s. Unfortunately, I no longer have an executable image. Is there a possibility to download it somewhere? Unfortunately, the NextCloudPi_OdroidHC2_07-24-22 image doesn't work either. I can't get an activation page.