nextcloud/nextcloudpi #1517: First Use Wizard: failing when moving data directory to USB

glenn-sorrentino, 2022-07-27:
```
[ nc-automount ] (Wed Jul 27 19:30:33 BST 2022)
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /lib/systemd/system/nc-automount.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /lib/systemd/system/nc-automount-links.service.
automount enabled

[ nc-format-USB ] (Wed Jul 27 19:30:38 BST 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0xe8e0db5e00 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Wed Jul 27 19:30:43 BST 2022)
Maintenance mode enabled
moving data directory from /var/www/nextcloud/data to /media/myCloudDrive/ncdata...
BTRFS filesystem detected
Create subvolume '/media/myCloudDrive/ncdata'
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value logfile set to string /media/myCloudDrive/ncdata/data/nextcloud.log
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value tempdirectory set to string /media/myCloudDrive/ncdata/data/tmp
Maintenance mode disabled
/usr/local/etc/library.sh: line 215: $3: unbound variable
```

glenn-sorrentino, 2022-07-27:
Running a second time:

```
[ nc-datadir ] (Wed Jul 27 19:31:21 BST 2022)
INFO: /media/myCloudDrive/ncdata is not empty. Creating backup /media/myCloudDrive/ncdata-07-27-22.1658946682
{"reqId":"qQu2kchyhngqTRUBOPrv","level":2,"time":"2022-07-27T18:31:23+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
{"reqId":"IIqJqbexk9cL7lDXkKvF","level":2,"time":"2022-07-27T18:31:24+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/23/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:164
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands()
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```

glenn-sorrentino, 2022-07-27:
And a third time:

```
[ nc-datadir ] (Wed Jul 27 19:36:52 BST 2022)
{"reqId":"Jw1aZ0TmL80Tr40XRCsw","level":2,"time":"2022-07-27T18:36:53+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
Error reading data directory. Is NextCloud running and configured?
```

theCalcaholic, 2022-07-28:
Hey, please add the information requested in the issue template (especially the output of `ncp-report`). That issue should already be fixed in the latest version.

glenn-sorrentino, 2022-07-28:
> Hey, please add the information requested in the issue template (especially the output of `ncp-report`). That issue should already be fixed in the latest version.

pi@nextcloudpi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.48.3
NextCloudPi image    NextCloudPi_07-25-22
OS                   Debian GNU/Linux 11. 5.10.92-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata/data (doesn't exist)
rootfs usage         2.5G/30G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase (doesn't exist)
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               12:28

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:87
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /var/www/nextcloud/lib/private/DB/Connection.php(237): Doctrine\DBAL\Connection->executeQuery()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(287): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /var/www/nextcloud/lib/private/AppConfig.php(361): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /var/www/nextcloud/lib/private/AppConfig.php(126): OC\AppConfig->loadConfigValues()
#7 /var/www/nextcloud/lib/private/AppConfig.php(317): OC\AppConfig->getApps()
#8 /var/www/nextcloud/lib/private/legacy/OC_App.php(982): OC\AppConfig->getValues()
#9 /var/www/nextcloud/lib/private/Server.php(692): OC_App::getAppVersions()
#10 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#11 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#13 /var/www/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#15 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(182): OC\AppFramework\Utility\SimpleContainer->get()
#16 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#19 /var/www/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#20 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#21 /var/www/nextcloud/lib/private/Server.php(1079): OC\AppFramework\Utility\SimpleContainer->get()
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
[... 44 lines cut ...]
[Wed Jul 27 20:35:51.176855 2022] [mpm_event:notice] [pid 728:tid 547881624640] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Wed Jul 27 20:35:51.176986 2022] [core:notice] [pid 728:tid 547881624640] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/myCloudDrive/ncdata/data/nextcloud.log' for reading: No such file or directory

```
</details>

theCalcaholic, 2022-07-28:
Okay, that shouldn't happen enymore. Can you 

1. give me the output of `cat /usr/local/bin/CONFIG/nc-datadir.sh`
2. run `ncp-update fix/ncdatadir-unbound-variable` and then try again?

Please perform the steps in this order

glenn-sorrentino, 2022-07-28:
>     1. give me the output of `cat /usr/local/bin/CONFIG/nc-datadir.sh`

cat: /usr/local/bin/CONFIG/nc-datadir.sh: No such file or directory

>     2. run `ncp-update fix/ncdatadir-unbound-variable` and then try again?

```
[ nc-automount ] (Thu Jul 28 16:37:31 PDT 2022)
automount enabled

[ nc-format-USB ] (Thu Jul 28 16:37:36 PDT 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0xe8e0db5e00 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Thu Jul 28 16:37:41 PDT 2022)
{"reqId":"jBZwOZnskXPZywJGXh8s","level":2,"time":"2022-07-28T23:37:42+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
Error reading data directory. Is NextCloud running and configured?
```

theCalcaholic, 2022-07-29:
> cat: /usr/local/bin/CONFIG/nc-datadir.sh: No such file or directory
Damnit, I forgot a directory. It would have been `/usr/local/bin/ncp/CONFIG/nc-datadir.sh`

In your log I can recognize a bug with nc-format-USB (it doesn't recreate the tmp directory if it is located on the USB drive). 
Please run
```sh
sudo mkdir /media/myCloudDrive/ncdata/data/tmp` && sudo chown www-data: /media/myCloudDrive/ncdata/data/tmp
```
and try again

glenn-sorrentino, 2022-07-29:
> > cat: /usr/local/bin/CONFIG/nc-datadir.sh: No such file or directory
> > Damnit, I forgot a directory. It would have been `/usr/local/bin/ncp/CONFIG/nc-datadir.sh`
> 
> In your log I can recognize a bug with nc-format-USB (it doesn't recreate the tmp directory if it is located on the USB drive). Please run
> ```shell
> sudo mkdir /media/myCloudDrive/ncdata/data/tmp` && sudo chown www-data: /media/myCloudDrive/ncdata/data/tmp
> ```

```
pi@nextcloudpi:~ $ sudo mkdir /media/myCloudDrive/ncdata/data/tmp && sudo chown www-data: /media/myCloudDrive/ncdata/data/tmp
mkdir: cannot create directory ‘/media/myCloudDrive/ncdata/data/tmp’: No such file or directory
```

glenn-sorrentino, 2022-07-29:
> ```shell
> sudo mkdir /media/myCloudDrive/ncdata/data/tmp` && sudo chown www-data: /media/myCloudDrive/ncdata/data/tmp
> ```

I manually created the directory. Here's the new output:

```
[ nc-datadir ] (Fri Jul 29 14:23:07 PDT 2022)
{"reqId":"kfCJe27gcPIwsQxTHAVx","level":3,"time":"2022-07-29T21:23:09+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/media/myCloudDrive/ncdata/data/data_dir_writability_test_62e44fbd19be0.tmp): failed to open stream: Permission denied at /var/www/nextcloud/lib/private/legacy/OC_Util.php#809","userAgent":"--","version":"23.0.6.1","exception":{"Exception":"Error","Message":"fopen(/media/myCloudDrive/ncdata/data/data_dir_writability_test_62e44fbd19be0.tmp): failed to open stream: Permission denied at /var/www/nextcloud/lib/private/legacy/OC_Util.php#809","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":809,"function":"fopen"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":157,"function":"checkServer","class":"OC_Util","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}
{"reqId":"kfCJe27gcPIwsQxTHAVx","level":3,"time":"2022-07-29T21:23:09+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/media/myCloudDrive/ncdata/data/nextcloud.log): failed to open stream: Permission denied at /var/www/nextcloud/lib/private/Log/File.php#84","userAgent":"--","version":"23.0.6.1"}
Error reading data directory. Is NextCloud running and configured?
```

nembrinj, 2022-08-07:
I have the same problem: fresh install of nextcloupi, following the steps to change data storage to USB drive leads to same behavior as above. I tried the above solution (ncp-update + mkdir /data/tmp) with same result.

noticed that ownership of data dir changes after running the nc-datadir script, resulting in permission denied when trying to cd from the ssh session terminal (which sounds strange to me)

pi@nextcloudpi:/media/backup/nextcloud-data $ ls
data
pi@nextcloudpi:/media/backup/nextcloud-data $ cd data
-bash: cd: data: Permission denied
pi@nextcloudpi:/media/backup/nextcloud-data $ 

here the ncp-report output:

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.48.1
NextCloudPi image    NextCloudPi_07-25-22
OS                   Debian GNU/Linux 11. 5.10.92-v8+ (aarch64)
automount            yes
USB devices          sda sdb 
datadir              /media/backup/nextcloud-data/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      862G/1.8T
rootfs usage         2.8G/15G
swapfile             /var/swap
dbdir                /media/USBdrive/nextcloud
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:15

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:164
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands()
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>
[... 26 lines cut ...]

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"6Ol0o6AZ9vSaljUE1si2","level":4,"time":"2022-08-07T21:38:03+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Could not boot notify_push: Connection refused","userAgent":"--","version":"23.0.6.1","exception":{"Exception":"RedisException","Message":"Connection refused","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":132,"function":"pconnect","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":172,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/var/www/nextcloud/apps/notify_push/lib/AppInfo/Application.php","line":62,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":160,"function":"OCA\\NotifyPush\\AppInfo\\{closure}","class":"OCA\\NotifyPush\\AppInfo\\Application","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":118,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":127,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":457,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":435,"function":"queryNoFallback","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":87,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":106,"function":"array_map"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":114,"function":"buildClass","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":131,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":463,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":435,"function":"queryNoFallback","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/Simpl [... cut ...] w/nextcloud/lib/private/Console/Application.php","line":119,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/RedisFactory.php","Line":132,"CustomMessage":"Could not boot notify_push: Connection refused"}}

```
</details>

dekzi, 2022-08-09:
Also facing this issue even after clean reinstalls and the solutions posted here. Identical logs to Glenn on the first, second, and third attempt to move the directory, attempted on multiple pi's and USB devices.

nembrinj, 2022-08-10:
Identical 1st, 2nd, and 3rd logs also in my case. is there a workaround in terms of configuration?

dekzi, 2022-08-11:
@theCalcaholic I'm unfamiliar with NextCloudPi's development workflow, but shouldn't this be considered a higher priority bug? This issue has directly broken core functionality, seemingly universally, and yet there's been almost no visible progress for 2 weeks.
I don't mean to target the developers or maintainers, I'm just extremely confused about the status of this issue.

nembrinj, 2022-08-12:
@dekzi I agree that this issue makes NextCloudPI completely unusable for my needs. I can wait another week, but I will then need to switch back to something working.

@theCalcaholic I am happy to help trying out release candidates or modifying the nc-datadir.sh script, but I need some input on what could be the problem

let us know :-)

theCalcaholic, 2022-08-13:
@nembrinj @dekzi @glenn-sorrentino This should already be fixed in the latest version. I was not able to reproduce this issue with latest master on a VM (using the curl installer).
However, there was a problem with my new, automated release pipeline and somehow an older image got attached to the 1.48.3 release (compare #1525). I fixed this for the RaspberryPi image (by uploading the correct image which was created, but not published by the release pipeline). If you have a Raspberry Pi, please (re-)download the image from the v1.48.3 release and check if the error persists.

I'll check and replace the other images soon if they have the same problem

nembrinj, 2022-08-13:
@theCalcaholic Thanks! will do today and let you know...

nembrinj, 2022-08-13:
@theCalcaholic  I managed to set up the data dir using the nc-datadir webUI. So up to now, everything working! thanks a lot for the support!

I'll let you know if I have problems setting up my users' accounts/backup/etc but I don't see why it should go wrong now.

theCalcaholic, 2022-08-14:
Good to hear!

closed 2022-08-14 by theCalcaholic

glenn-sorrentino, 2022-08-15:
Still isn't fixed for me.

gsaurabhr, 2022-08-30:
Same issue with me with ncp 1.49.0

hanselpeter, 2022-09-16:
same for me. using nextcloudpi v1.49.0.
During first initialisation using the wizard and while trying to move the data dir to the usb after setting up of nextcloud.
Nextcloud is not usable for me because of this.