nextcloud/nextcloudpi #1525: md5

sarimsa, 2022-08-09:
Hi
I want to newly setup a NextCloudPi and downloaded at https://github.com/nextcloud/nextcloudpi/releases/
04a966f98d7d29d191e65227e717253d  NextCloudPi_RPi_07-25-22.tar.gz

I calculate md5 as 77ECF533A80AACEFE8BAD43C4F9615D8
- is my download wrong or the hash?
- what is the risk to use the download anyway?
Thanks
s

theCalcaholic, 2022-08-09:
Hey, thank you for reporting. You're right, there seems to have been an issue with my release pipeline. The image you had been downloading still had a bug that is fixed on the latest version. I've replaced the image with in the release with the correct one.

closed 2022-08-09 by theCalcaholic

glenn-sorrentino, 2022-08-15:
This still isn't fixed for me :/