nextcloud/nextcloudpi #1526: Odroid XU4 -

nacho-domatix, 2022-08-09:
Trying to give my Odroid XU4 a second life (it lived before with nextcloudpi too) I found an image of [Debian Bullseye OS](https://forum.odroid.com/viewtopic.php?f=96&t=44932) for the device and used the script with curl. And everything went apparently “almost” fine (just got a “Failed to start MariaDB 10.5.15 database server.” but after reboot everything was ok).

The issue is that when I type the IP, standard Nextcloud window appears. No trace of NCP (cli or web using port 4443), and no idea what’s wrong.

Anyone had this issue before? Any help would be appreciated. Many thanks in advance
[image]
.

theCalcaholic, 2022-08-09:
Hi, thanks for reporting. Can you please attach the exact command you used for installing NCP and the output of `sudo ncp-report`?

nacho-domatix, 2022-08-09:
Hi @theCalcaholic, many thanks or the reply.

I used 
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`

Also:

```
root@odroid-bullseye:~# sudo ncp-report
sudo: ncp-report: command not found
```
As far as I see, there is no ncp-whatever command installed on the system.

theCalcaholic, 2022-08-10:
@nacho-domatix  In that case, the installation probably failed. I don't suppose you still have its output? With a bit of luck you might find some clue in the file `/var/log/ncp.log`

Normally, I would tell you to try one of the readily built images for your board - however, there is currently some kind of upstream issue that prevents me from verifying if the 32bit armbian images are working. 

Actually, you could help me with that by testing the image from the following link:

https://storage.googleapis.com/ncp-public-share/2824533501-odroidxu4-ncp.zip

If you'd like to help, you can try to run the image on your Odroid XU4 and give me the output of the following commands:

```sh
sudo systemctl status apache2
sudo apache2ctl -V
```

nacho-domatix, 2022-08-10:
```
root@odroid-bullseye:~# cat /var/log/ncp.log
[ nc-nextcloud ] (Tue 09 Aug 2022 04:27:41 PM UTC)
Downloading Nextcloud 23.0.6...
Installing  Nextcloud 23.0.6...
Creating possible missing Directories
chmod Files and Directories
chown Directories
chmod/chown .htaccess
Setting up database...
Setting up Apache...
INFO: Metrics enabled: no
Enabling site nextcloud.
To activate the new configuration, you need to run:
  systemctl reload apache2
Module proxy already enabled
Considering dependency proxy for proxy_http:
Module proxy already enabled
Enabling module proxy_http.
Considering dependency proxy for proxy_wstunnel:
Module proxy already enabled
Enabling module proxy_wstunnel.
To activate the new configuration, you need to run:
  systemctl restart apache2
Created symlink /etc/systemd/system/multi-user.target.wants/notify_push.service → /etc/systemd/system/notify_push.service.
Setting up system...
Don't forget to run nc-init

```

That is the output. I wasn't aware of every line during the installation, but seemed to work ok.

Regarding the image of your link, I couldn't make it work. The Odroid's IP shown on the router is not answering through ssh or web. I can try again later, but doesn't look good at first glance.

nacho-domatix, 2022-08-11:
So I tried 3 different installations with different cards and drives and the image provided was 100% functional. When I was ready to use it and start to sync, I had some fun with "expected filesize xxx but read xxx", apparently related to this: https://help.nextcloud.com/t/tmp-directory-permissions-after-restore-of-nextcloudpi/59742/8 but tmp directory was set properly in /var/www/nextcloud/config/config.php After a reboot, everything was fine. 

Sometimes, not sure when, Odroid tries to boot from the hard drive (it has to do something with the format I believe), so better don't connect at first boot. Then following ncp instructions, every reboot I did is handled by the sdcard.

Everything works perfect and smooth, I love your work guys. Thanks for all the amazing work and contributions.

theCalcaholic, 2022-08-12:
@nacho-domatix Glad to hear that you could resolve your issue!

closed 2022-08-12 by theCalcaholic