nextcloud/nextcloudpi #1531: Nextcloud 25 Support

sunjam, 2022-08-12:
[First beta of 25 released](https://help.nextcloud.com/t/first-beta-of-nextcloud-25/143635/) as of 8/12.
- https://github.com/nextcloud/appstore/issues/917 for appstore listings.

- https://github.com/nextcloud/nextcloudpi/issues/1447 for NextcloudPi

sunjam, 2022-08-12:
- https://github.com/nextcloud/server/issues/32595 might not be included by the time 25 is released, but it is in development now.

sunjam, 2022-09-20:
```
Running nc-update-nextcloud
Current   Nextcloud version 24.0.3.2
Available Nextcloud version 25.0.0.beta7
Download Nextcloud 25.0.0.beta6...
Error downloading
Clean up...
Done. Press any key...
```
Download fails when attempting to update to any beta of 25 from ncp-config or 4443

sunjam, 2022-09-22:
Interesting, that functionality was previously available to at least tell you "Your Nextcloud is the latest release" with a green check mark.  I understand that in the end it isn't strictly ncp related.  Thanks for the clarification.

theCalcaholic, 2022-09-22:
It's basically a specific nextcloud app that has been disabled

sunjam, 2022-09-22:
As if by magic, the built-in Updater works again.  I am unsure of why it wasn't rendering, but it certainly works now.  :champagne: 
[image]

theCalcaholic, 2022-09-22:
The magic is you updating NC manually :D
ncp's updater disables the NC update app after completing the update

sunjam, 2022-09-22:
```
Running nc-update-nextcloud
Current   Nextcloud version 25.0.0.12
Available Nextcloud version 25.0.0RC1
/usr/local/etc/library.sh: line 408: [: 0RC1: integer expression expected
```

sunjam, 2022-10-08:
Upgrade to 25..0.0rc3 fails with `Segmentation fault (core dumped)`

sirdrakeistda, 2022-10-09:
@sunjam Have you fixed the segmentation fault? There is a similar issue with this: https://github.com/nextcloud/nextcloudpi/issues/1615 
If you have any idea how to fix it would be nice to get some information. The error happened for me with NC24.0.6, so this doesn't seem to be only related to NC25 beta. Is it possible to be related to the php7.4 -> php8.1 update? Maybe thats the reason it can't be reproduced with a fresh container because the fresh container might have already php8.1.

theCalcaholic, 2022-10-09:
> Edit: this is all on ncp-internal-arm64:3146515542 Changing back to latest. **Disregard this post if it isn't helpful.**
> 
> Upgrade to `25.0.0rc3` fails with `Segmentation fault (core dumped)` from `24.0.6`
> 
> `/usr/local/bin/ncc: line 3: 1664079 Segmentation fault ${SUDO} php /var/www/nextcloud/occ "$@"`
> 
> https://bin.disroot.org/?28fe9b9c4eceb513#F6KJ66JMruLyKss5MSVeqJCFfza1rt6hTcQFtADYyJ73 https://bin.disroot.org/?da3cb5397c520f23#5wuvjEu6utzyx89Uy6dcQkXH582hpMDAx1EDsr7At3Ay
> 
> NextcloudPi fails to understand it can update from 24.0.5 to 25.0.0rc3, so:
> 
>     * Switch branch to beta
> 
>     * Update from 24.0.5 to 24.0.6
> 
>     * Switch branch to stable
> 
>     * Switch branch to beta
> 
>     * Update from 24.0.6 to 25.0.0rc3 become available
> 
> 
> Interesting, update process works from a fresh container. Wasn't able to reproduce the segmentation fault. https://bin.disroot.org/?51421b62908b3561#F3tFoXTCE8WH5EwhfFwwoK2hWo7EVpZj7qYQoxZSQi3B

@sirdrakeistda  is right: The segmentation fault occurs if you have upgraded php and not cleared the pre-8.1 opcache.

`sudo bash -c /usr/local/etc/library.sh; clear_opcache` should fix the issue

sunjam, 2022-10-14:
Updater bug has been fixed, so update process is now as simple as updating to `25.0.0rc5`

Seems to be smooth sailing.  Let me know if you want anything specific tested.  Papercut issues I've encountered so far have been resolved.

sirdrakeistda, 2022-10-19:
I installed nextcloud 25.0.0. After update I got an blank page (firefox) or an http 500 error (chrome) after login. Most apps didn't worked and showed a blank page (including the start page with the dashboard, files, ... I don't know why, but polls and Phonetrack did work).  The app settings were reachable and usable, occ commands work without problem. After some research I noticed, that this problem was caused by the deck app. Disabling Deck fixed the issue.
https://github.com/nextcloud/deck/issues/3978
I hope this is helpful for others with the same issue.

REAPERSbattlecry, 2022-10-20:
ncp 1.50.3 on Debian 11 VM  with Curl installer. 
Fresh install upgraded to NC25.0.0. seem to have no problems. 
I installed several apps including Talk, Deck, Snappymail, Mail, Groupfolders.
My impression is, that it is a little bit slower than NC24...

Will check further.

phil-lipp, 2022-10-21:
I just updated from 24.0.6 (with php8.1) to 25.0.0. on an existing instance. So far everything's working fine and navigating through the NC interface feels snappier than ever!
One thing I noticed is that running nc-previews throws the following error:
```
[Symfony\Component\Console\Exception\CommandNotFoundException]  
  Command "preview:pre-generate" is not defined.                  
                                                                  
  Did you mean one of these?                                      
      notification:generate                                       
      preview:repair                                              
      preview:reset-rendered-texts 
```

surfrock66, 2022-11-19:
Just a heads up, the fix for missing indicies is usually this:

` sudo -u www-data php ./occ  db:add-missing-indices`

sunjam, 2022-12-06:
Release client available for testing if anyone can test for 32-bit

https://help.nextcloud.com/t/rc-3-of-nc-25-0-2/151387

sunjam, 2022-12-09:
25.0.2 has been released to nc stable.  Anyone able to test?  Especially appreciated if you can test the update on 32-bit hardware.  Thanks!

janvlug, 2022-12-09:
I'm running Nextcloud server 25.0.2 on an old 32 bit Synology, it works. I had to modify a file to get it working on 25.0.1.
I'm looking forward to Nextcloud 25 support by nextcloudpi, because I want to migrate to a bit faster hardware running this project.

REAPERSbattlecry, 2023-01-15:
Did some trials with devel-branch v1.51.0 and the update from NC24.0.8 to 25.0.2.
It generally works but never got it working without having a "service unavailable" on the whole webserver (Nextcloud and ncp-UI). 
It is solvable with the new clear-op-cache-tool (Thanks for that I think @ZendaiOwl) but it have to be done in the cli (ncp-config).

Is there a way to trigger the clear-op-cache after the update by default?

ZendaiOwl, 2023-01-15:
@REAPERSbattlecry 

The op_clear_cache tool should be made by @theCalcaholic 🙂 so I'll let him answer that

theCalcaholic, 2023-01-19:
> [With the 25.0.3 release PHP 7.4 has reached EOL and is no longer supported moving forward.](https://help.nextcloud.com/t/third-maintenance-release-for-hub-3-is-out-get-your-update/153857)
> - [#34492](https://github.com/nextcloud/server/pull/35737)
> - [#34997](https://github.com/nextcloud/server/pull/34997)
> - [#35737](https://github.com/nextcloud/server/pull/34492)

Good thing we upgraded PHP when moving to NC 24 ☺️

svenb1234, 2023-01-24:
I pulled the :latest docker image and got the following:

```
Setup notify_push (attempt 1/5)

✓ redis is configured

✓ push server is receiving redis messages

✓ push server can load mount info from database

✓ push server can connect to the Nextcloud server

✓ push server is a trusted proxy

✓ push server is running the same version as the app

  configuration saved

Nextcloud version 24.0.8.2. NextCloudPi version v1.51.0

Back up current instance...

check free space...

clean up old backup /data/docker-startup-backups/nextcloud-bkp_20221104_1667597843.tar.gz

Maintenance mode enabled

backup database...

backup files...
```

So nextcloud seems to be at 24 still for the 1.51.0 latest docker image.

theCalcaholic, 2023-01-24:
@svenb1234 So far far looks alright. Did the update complete after a while or did it abort?

theCalcaholic, 2023-01-24:
@svenb1234 Yes, until you run the update to nc 25 from the web UI

theCalcaholic, 2023-01-24:
Now implemented in v1.51.0

closed 2023-01-24 by theCalcaholic

svenb1234, 2023-01-24:
Isn'T the whole point of containers that one does update by pulling a new container?

The update does not work either if one does not specify the exact version. 0 used to work in order to update to the latest nextcloud version available.

[image]

sunjam, 2023-01-24:
Correct, but ncp has never followed container practices.  Don't expect this
to change, but improvements to the docker process have been in gradual
development.

theCalcaholic, 2023-01-25:
> Isn'T the whole point of containers that one does update by pulling a new container?
> 
> The update does not work either if one does not specify the exact version. 0 used to work in order to update to the latest nextcloud version available.
> 
> [image]

Yes, you update NextcloudPi by pulling a new container. But we support multiple NC versions with each NCP version, so updating users Nextcloud version would just be rude (consider that people might rely on NC apps or customizations that aren't compatible with the latest version of NC yet)

ZendaiOwl, 2023-01-25:
It looks like the suggestion is what is filled in the image, did you fill in a '0' in the field or did you only hit apply?

svenb1234, 2023-01-25:
The 0 is the default input which you get if you enter nothing. This used to pick the latest nextcloud version available for NCP.

With this "nothing" in the input field I clicked on apply.

svenb1234, 2023-01-25:
Still misleading. NCP did never show the latest version available in that field. If there was nothing in the input field NCP checked - after one clicked on apply - for the latest available version, i.e. while showing the progress in the big textarea below apply.

theCalcaholic, 2023-01-25:
> Still misleading. NCP did never show the latest version available in that field. If there was nothing in the input field NCP checked - after one clicked on apply - for the latest available version, i.e. while showing the progress in the big textarea below apply.

Exactly, that's how it's supposed to work and how it worked in our tests. It's not misleading - it's a bug :)

theCalcaholic, 2023-01-25:
Ah, @ZendaiOwl you're right. The field is actually empty. We should adjust the hint, probably. @svenb1234 The updater never worked with an empty input, but it worked with an input of "0". What we are seeing in your screenshot is a placeholder, while the field is actually empty. This is indeed misleading - the placeholder should maybe adjusted to something like "24.0.8 or 0"