nextcloud/nextcloudpi #1532: [REOPENED] First Use Wizard: failing when moving data directory to USB

glenn-sorrentino, 2022-08-15:
[This issue ](https://github.com/nextcloud/nextcloudpi/issues/1517)was closed prematurely. 

Still failing to move datadir using both the wizard and the admin panel.

glenn-sorrentino, 2022-08-15:
[ nc-automount ] (Mon Aug 15 09:14:14 BST 2022)
automount enabled

[ nc-format-USB ] (Mon Aug 15 09:14:21 BST 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0xe8e0db5e00 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Mon Aug 15 09:14:26 BST 2022)
{"reqId":"0h5wDzzX24wAWIh5BsZp","level":2,"time":"2022-08-15T08:14:27+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
Error reading data directory. Is NextCloud running and configured?

glenn-sorrentino, 2022-08-15:
<details>
<summary>NextCloudPi diagnostics</summary>

```

{"reqId":"sifAYK4KJaKbqvMcBwnb","level":2,"time":"2022-08-15T08:18:07+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
NextCloudPi version  v1.48.3
NextCloudPi image    NextCloudPi_07-25-22
OS                   Debian GNU/Linux 11. 5.10.92-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata/data (doesn't exist)
rootfs usage         2.3G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{"reqId":"MPfzzCMw0RGkSN9Zv4DE","level":2,"time":"2022-08-15T08:18:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/myCloudDrive/ncdata/data/tmp is not present or writable","userAgent":"--","version":"23.0.6.1"}
Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/23/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:164
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands()
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Aug 15 09:05:03.017979 2022] [mpm_event:notice] [pid 845:tid 548455515200] AH00493: SIGUSR1 received.  Doing graceful restart
[Mon Aug 15 09:05:03.037375 2022] [ssl:warn] [pid 845:tid 548455515200] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 1.1.1n  15 Mar 2022, version currently loaded is OpenSSL 1.1.1k  25 Mar 2021) - may result in undefined or erroneous behavior
[Mon Aug 15 09:05:03.040575 2022] [ssl:warn] [pid 845:tid 548455515200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Aug 15 09:05:03.040780 2022] [ssl:error] [pid 845:tid 548455515200] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=fv-az132-400 / issuer: CN=fv-az132-400 / serial: 188C6D7B852C5B077F697C339E9E93E4C1E851BF / notbefore: Jul 25 20:23:36 2022 GMT / notafter: Jul 22 20:23:36 2032 GMT]
[Mon Aug 15 09:05:03.040800 2022] [ssl:error] [pid 845:tid 548455515200] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Aug 15 09:05:03.043204 2022] [mpm_event:notice] [pid 845:tid 548455515200] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1k configured -- resuming normal operations
[Mon Aug 15 09:05:03.043254 2022] [core:notice] [pid 845:tid 548455515200] AH00094: Command line: '/usr/sbin/apache2'
[Mon Aug 15 09:12:45.130643 2022] [proxy_fcgi:error] [pid 2931:tid 548244353408] [client 192.168.1.121:49782] AH01067: Failed to read FastCGI header
[Mon Aug 15 09:12:45.130746 2022] [proxy_fcgi:error] [pid 2931:tid 548244353408] (104)Connection reset by peer: [client 192.168.1.121:49782] AH01075: Error dispatching request to :4443: 
[... 9 lines cut ...]
[Mon Aug 15 09:12:52.504350 2022] [mpm_event:notice] [pid 738:tid 548593083456] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1k configured -- resuming normal operations
[Mon Aug 15 09:12:52.504460 2022] [core:notice] [pid 738:tid 548593083456] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/myCloudDrive/ncdata/data/nextcloud.log' for reading: No such file or directory

```
</details>

theCalcaholic, 2022-08-15:
Hey, sorry to hear that. Please feel me exactly how you reinstalled NCP

theCalcaholic, 2022-08-15:
From your logs it appears like you didn't reinstall NCP at all - is that correct?

During the previous attempts to move the data directory, it was broken. There is unfortunately no good way I can fix that via update. Therefore it would be necessary to reinstall NCP

glenn-sorrentino, 2022-08-15:
> @nembrinj @dekzi @glenn-sorrentino I fixed this for the RaspberryPi image (by uploading the correct image which was created, but not published by the release pipeline). If you have a Raspberry Pi, please (re-)download the image from the v1.48.3 release and check if the error persists.

I downloaded the Pi image [from here](https://github.com/nextcloud/nextcloudpi/releases) and flashed using Etcher.

glenn-sorrentino, 2022-08-16:
I just tried using curl but still no luck.

dekzi, 2022-08-16:
Also still facing this on a fresh reinstall with the updated Pi image on the releases page.

theCalcaholic, 2022-08-16:
Okay, since I can't reproduce the issue, it would be of tremendous help if you could describe all the steps you took from installing NCP until moving the data directory (including exact commands where applicable).

theCalcaholic, 2022-08-16:
Did you, by any chance, try to restore NCP from a backup? The error that you are seeing should not be possible on a fresh installation...

theCalcaholic, 2022-08-16:
I have adjusted nc-format-USB to avoid accidental breakage of the system. Could you try running `ncp-update fix/nc-format-usb-breaking-tmp` on a fresh NCP system **before** running anything else apart from the activation (not even the first run wizard) and then try again?

glenn-sorrentino, 2022-08-16:
> I have adjusted nc-format-usb to avoid accidental breakage of the system. Could you try running ncp-update fix/nc-format-usb-breaking-tmp on a fresh NCP system before running anything else apart from the activation (not even the first run wizard) and then try again?

@theCalcaholic Running into a new issue with... here are my steps:

1. Format SD and external drive
2. Install Raspberry Pi OS using their Installer app
3. Log in a run: `sudo apt update && sudo apt -y dist-upgrade && sudo apt -y autoremove && curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash`
4. Reboot
5. Log in to the webpanel and start the wizard
6. Automount 👍
7. **[NEW]** Format 👎

Report 👇
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.49.0
NextCloudPi image    NextCloudPi_08-16-22
OS                   Debian GNU/Linux 11. 5.15.56-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.4G/30G
rootfs usage         4.4G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    24.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "<!DOCTYPE HTML PUBLIC \"-\/\/IETF\/\/DTD HTML 2.0\/\/EN\">\n<html><head>\n<title>302 Found<\/title>\n<\/head><body>\n<h1>Found<\/h1>\n<p>The document has moved <a href=\"https:\/\/icanhazip.com\/\">here<\/a>.<\/p>\n<\/body><\/html>",
            "1": "192.168.1.96",
            "14": "nextcloudpi"
        },
[... 79 lines cut ...]
```

{"reqId":"Yvv1r8fER1rrueWV3nAK3wAAAFQ","level":2,"time":"2022-08-16T19:53:44+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"HEAD","url":"/remote.php/dav/files/glenn//.SeedVaultAndroidBackup/1657155229227/0H3BlLXua9xBR29yHKyp9vB42nIzWFTqzKnL-1wOYP8","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv1ycfER1rrueWV3nAK4AAAAFU","level":2,"time":"2022-08-16T19:54:10+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"MKCOL","url":"/remote.php/dav/uploads/glenn/e17b48be422e3eab5bc7112b3a19f819","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv14-zk-Bifr2NnEa6FKAAAAJg","level":2,"time":"2022-08-16T19:54:36+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/uploads/glenn/e17b48be422e3eab5bc7112b3a19f819","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv1_sfER1rrueWV3nAK4QAAAFc","level":2,"time":"2022-08-16T19:55:03+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"HEAD","url":"/remote.php/dav/files/glenn//.SeedVaultAndroidBackup/1657155229227/63F7a7t9aidmrNFUPeISHTxstmxQ3g1iV8wJ89xDiHo","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv2Hezk-Bifr2NnEa6FKQAAAIE","level":2,"time":"2022-08-16T19:55:34+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"MKCOL","url":"/remote.php/dav/uploads/glenn/a98445cf49bdce451e5e8af6dec026eb","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv2Nuzk-Bifr2NnEa6FKgAAAII","level":2,"time":"2022-08-16T19:56:00+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/uploads/glenn/a98445cf49bdce451e5e8af6dec026eb","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv2Uezk-Bifr2NnEa6FKwAAAIU","level":2,"time":"2022-08-16T19:56:26+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"HEAD","url":"/remote.php/dav/files/glenn//.SeedVaultAndroidBackup/1657155229227/Y436PfNZpkVwqAWGGNaRnJvfY0X-FNQ-iqRy75wrfnk","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}
[... cut ...] ","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"Yvv4ZcGVUDBYN__8lfi7fwAAAA0","level":2,"time":"2022-08-16T20:05:18+00:00","remoteAddr":"192.168.1.99","user":"--","app":"core","method":"HEAD","url":"/remote.php/dav/files/glenn//.SeedVaultAndroidBackup/1657155229227/UyOt_3-iu2TU_vYvI7smAyAAee221PjnujGMMWYnyFk","message":"Login failed: 'glenn' (Remote IP: '192.168.1.99')","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.21.0","version":"24.0.3.2","data":{"app":"core"}}

```
</details>

glenn-sorrentino, 2022-08-16:
Hm, appears to work now despite the error?

theCalcaholic, 2022-08-16:
At least the log ouput looks good. Allthough I don't know why it didn't work when you ran nc-format-usb the first time.

Btw, when you run `ncp-update` after the installation, you overwrote the changes from the fix-branch again - but those didn't change core functionality anyways (just an extra check to make sure the data directory isn't overwritten).

closed 2022-08-19 by glenn-sorrentino

lSSBl, 2022-08-19:
Also in the v1.49.0 still not working:

---------------------------------------------------------------------------------------------------------------------
[ nc-automount ] (Fri Aug 19 19:13:15 CEST 2022)
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /lib/systemd/system/nc-automount.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /lib/systemd/system/nc-automount-links.service.
automount enabled

[ nc-format-USB ] (Fri Aug 19 19:13:25 CEST 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0x5752affe00 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Fri Aug 19 19:13:37 CEST 2022)
Maintenance mode enabled
moving data directory from /var/www/nextcloud/data to /media/USBdrive/ncdata...
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value logfile set to string /media/USBdrive/ncdata/data/nextcloud.log
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value tempdirectory set to string /media/USBdrive/ncdata/data/tmp
Maintenance mode disabled
---------------------------------------------------------------------------------------------------------------------

The System:
Fresh RPi3 with PiDrive in NextCloud case
Fresh 32GB SD
DL image NextCloudPi_RaspberryPi_v1.49.0.zip from github.com
extract
Burn image with etcher on win10
power up RPi with SD
IP in browser
activation
entering NextCloudPI with login ncp / pass
activation wizard
   -> USB yes -> OK
   -> automount -> OK
   -> format -> OK
   -> transfer DATA -> ERROR fail during data transfer - see logfile above

Sorry, thats not funny.
Same an 1.48.3 or 1.47.2

jannissi, 2022-08-20:
Hi there, ich also have the still the same problem. 
I tried:
- v 1.48.3 (a few weeks ago)
- v 1.48.3 yesterday
- v 1.49.0
i tried everything a few times.
My Nextcloud isn´t running for a month now and i need to restore my Backup to reach my Files again.
Is there a opportunity to download an older version form may, whitch defenetly works?

Thanks.

theCalcaholic, 2022-08-20:
@lSSBl Your log doesn't show any errors (the statements about maintenance mode are fine, it's just information no error). Are you sure that moving the data directory failed? If so, can you please describe what doesn't work and add the output of `ncp-report`?

@jannissi Same request: please explain, what's not working as expected and add the output of `ncp-report`. I don't have any images available except those on the GH release page

lSSBl, 2022-08-20:
Hmmm,

that is a view, I did`t expect, becouse on an second (running system) I`ve running NextClous 19 since 3 years and no updates untill now. there everything tuirned green during the wizard, no yellow or red colors.

I`ll check the actual image next week. I`m on holiday right now.

What I can say is, that the installation-wizard turned green by the first two parts (automount and format) and yellow after moving nc-datadir and on the left side the error occours.

Cheers from DE

jannissi, 2022-08-21:
when i set up a new Nextcloudpi (1.49.0), i use the "First start Wizard". After i reach "Move data ti USB" this appears (i always have to click "Format USB" twice) :

_[ nc-automount ] (Sun Aug 21 12:31:16 BST 2022)
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /lib/systemd/system/nc-automount.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /lib/systemd/system/nc-automount-links.service.
automount enabled

[ nc-format-USB ] (Sun Aug 21 12:31:24 BST 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0x3a3817d5600 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Error: Partition(s) 1 on /dev/sda have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

[ nc-format-USB ] (Sun Aug 21 12:31:39 BST 2022)
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0x3a3817d5600 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Sun Aug 21 12:31:45 BST 2022)
Maintenance mode enabled
moving data directory from /var/www/nextcloud/data to /media/myCloudDrive/ncdata...
BTRFS filesystem detected
Create subvolume '/media/myCloudDrive/ncdata'
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value logfile set to string /media/myCloudDrive/ncdata/data/nextcloud.log
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value tempdirectory set to string /media/myCloudDrive/ncdata/data/tmp
Maintenance mode disabled_

Sorry, where can i find ncp-report?

theCalcaholic, 2022-08-21:
@jannissi Your log doesn't contain any errors either. 

That means, there's either some silent error that doesn't show any error message or everything went fine and the exit code doesn't reflect the actual script execution status.
I'll have to investigate this further...

@jannissi  In order to execute the command `sudo ncp-report`, you will need to ssh into your ncp instance (see [this documenation article](https://help.nextcloud.com/t/how-to-enable-ssh-using-ncp-config-or-ncp-web/126317))

glenn-sorrentino, 2022-10-11:
No, I reinstalled the new Pi image. It was the version you suggested, but the file name didn't appear to change (7/25).

Sent from ProtonMail mobile

-------- Original Message --------