nextcloud/nextcloudpi #1542: PHP8.0 Support needed

Jannik44, 2022-08-18:
Hello, is it planned to support php8.0 within the next weeks? We really need it, more and more apps arent running with php 7 anymore, for example my password manager app, it crashes without php8.0 and a manual upgrade failed, so is this planned in the next time?

theCalcaholic, 2022-08-21:
Yes, that's actually the next item on my todo list :)

I'm surprised, this happens, though. NC 23 is still supported and doesn't support PHP 8 iirc. That would mean that these apps don't respect the support cycle of NC itself... 🤔

Oberlurch3000, 2022-08-24:
I've already updated to php 8 due to the passwords app ,@Jannik44  mentioned, requires it. I thing everything works ok, but that's for my private cloud. The only thing I noticed is the ncp diagnostics show php was down. I assume this issue should also go away then, right? So just wait for an ncp update.

theCalcaholic, 2022-08-24:
@Oberlurch3000 I can't promise that, but very likely, yes.
I recommend to disable auto updates for now and ensure you have a backup before upgrading ncp (and then NC for the first time).

The difficult thing with the php upgrade is, that I need to ensure, that all the config files are migrated and the settings pertained (that caused issues with uploading larger files in the past, for example).

Therefore I'm migrating all relevant config files to a new system that has been in place for a while but not yet used by all config files.

I'll give you more specific instructions on the safest way to upgrade once I'm done

theCalcaholic, 2022-08-25:
@Oberlurch3000 Alright, so I have a working upgrader now at the `feature/php8` branch.
The best way for you to upgrade would be to first revert your manual upgrade, that means:

1. Remove all php8.1 packages
2. Remove the apt respository that you presumably added from /etc/apt/sources.list or /etc/apt/sources.list.d
3. Run the following command to reinstall php7.4:
```sh
apt-get install -y --no-install-recommends -t "bullseye" php-{common,igbinary,redis} "php7.4" "php7.4}"-{curl,gd,fpm,cli,opcache,mbstring,xml,zip,fileinfo,ldap,intl,bz2,json,common,readline,mysql,bcmath,gmp}
```
4. Restart php-fpm and apache2 to apply the change: `service apache2 restart; service php8.1-fpm stop; service php7.4-fpm start`
5. Now, update to the feature branch with: `ncp-update feature/php8`
6. Then, upgrade nextcloud to version 24.0.4 and php to version 8.1 with `ncp-update-nc 24.0.4`

Best of luck!

Oberlurch3000, 2022-08-25:
@theCalcaholic Ok, backup done. Now I'll see if it works for me. My instance is already on 24.0.4. Is that a problem?

theCalcaholic, 2022-08-25:
Yeah, it won't do the upgrade in this case. It's probably better to disable automatic nc updates (you can keep automatic ncp updates) and wait for the next NC release.

Oberlurch3000, 2022-08-25:
I've no automatic updates enabled. I'm just a curious and impatient guy :D.
EDIT: Oh and I didn't mention using aarch64 architecture! I think that would have been important....?!

I get this when performing ```ncp-update feature/php8```:
<details>
<summary>OUTPUT</summary>
 ```

root@nextcloudpi:/home/pi# ncp-update feature/php8
INFO: updating to development branch 'feature/php8'
Downloading updates
Performing updates
ls: cannot access '/etc/php/8.1/fpm/conf.d/*-opcache.ini': No such file or directory
ls: cannot access '/etc/php/8.1/cli/conf.d/*-opcache.ini': No such file or directory
remote: Enumerating objects: 13884, done.
remote: Counting objects: 100% (10612/10612), done.
remote: Compressing objects: 100% (2620/2620), done.
remote: Total 10194 (delta 7731), reused 9778 (delta 7330), pack-reused 0
Receiving objects: 100% (10194/10194), 2.42 MiB | 6.79 MiB/s, done.
Resolving deltas: 100% (7731/7731), completed with 266 local objects.
From https://github.com/nextcloud/nextcloudpi
 * [new tag]         v0.0.1     -> v0.0.1
 * [new tag]         v0.0.7     -> v0.0.7
 * [new tag]         v0.1.0     -> v0.1.0
 * [new tag]         v0.10.0    -> v0.10.0
 * [new tag]         v0.10.1    -> v0.10.1
 * [new tag]         v0.11.0    -> v0.11.0
 * [new tag]         v0.11.1    -> v0.11.1
 * [new tag]         v0.12.0    -> v0.12.0
 * [new tag]         v0.12.1    -> v0.12.1
 * [new tag]         v0.12.10   -> v0.12.10
 * [new tag]         v0.12.11   -> v0.12.11
 * [new tag]         v0.12.12   -> v0.12.12
 * [new tag]         v0.12.13   -> v0.12.13
 * [new tag]         v0.12.14   -> v0.12.14
 * [new tag]         v0.12.16   -> v0.12.16
 * [new tag]         v0.12.17   -> v0.12.17
 * [new tag]         v0.12.18   -> v0.12.18
 * [new tag]         v0.12.19   -> v0.12.19
 * [new tag]         v0.12.2    -> v0.12.2
 * [new tag]         v0.12.3    -> v0.12.3
 * [new tag]         v0.12.4    -> v0.12.4
 * [new tag]         v0.12.5    -> v0.12.5
 * [new tag]         v0.12.6    -> v0.12.6
 * [new tag]         v0.12.7    -> v0.12.7
 * [new tag]         v0.12.8    -> v0.12.8
 * [new tag]         v0.12.9    -> v0.12.9
 * [new tag]         v0.13.0    -> v0.13.0
 * [new tag]         v0.14.0    -> v0.14.0
 * [new tag]         v0.14.1    -> v0.14.1
 * [new tag]         v0.14.2    -> v0.14.2
 * [new tag]         v0.14.3    -> v0.14.3
 * [new tag]         v0.14.4    -> v0.14.4
 * [new tag]         v0.14.5    -> v0.14.5
 * [new tag]         v0.14.6    -> v0.14.6
 * [new tag]         v0.14.7    -> v0.14.7
 * [new tag]         v0.14.8    -> v0.14.8
 * [new tag]         v0.14.9    -> v0.14.9
 * [new tag]         v0.15.0    -> v0.15.0
 * [new tag]         v0.15.1    -> v0.15.1
 * [new tag]         v0.15.2    -> v0.15.2
 * [new tag]         v0.16.0    -> v0.16.0
 * [new tag]         v0.16.1    -> v0.16.1
 * [new tag]         v0.16.2    -> v0.16.2
 * [new tag]         v0.17.0    -> v0.17.0
 * [new tag]         v0.17.1    -> v0.17.1
 * [new tag]         v0.17.10   -> v0.17.10
 * [new tag]         v0.17.11   -> v0.17.11
 * [new tag]         v0.17.12   -> v0.17.12
 * [new tag]         v0.17.13   -> v0.17.13
[... 773 lines cut ...]
 * [new tag]         v1.6.3     -> v1.6.3
 * [new tag]         v1.6.4     -> v1.6.4
 * [new tag]         v1.6.5     -> v1.6.5
 * [new tag]         v1.6.6     -> v1.6.6
 * [new tag]         v1.6.7     -> v1.6.7
 * [new tag]         v1.7.0     -> v1.7.0
 * [new tag]         v1.7.1     -> v1.7.1
 * [new tag]         v1.8.0     -> v1.8.0
 * [new tag]         v1.8.1     -> v1.8.1
 * [new tag]         v1.8.2     -> v1.8.2
 * [new tag]         v1.8.3     -> v1.8.3
 * [new tag]         v1.8.4     -> v1.8.4
 * [new tag]         v1.9.0     -> v1.9.0
 * [new tag]         v1.9.1     -> v1.9.1
 * [new tag]         v1.9.2     -> v1.9.2
 * [new tag]         v1.9.3     -> v1.9.3
 * [new tag]         v1.9.4     -> v1.9.4
 * [new tag]         v1.9.5     -> v1.9.5
 * [new tag]         v1.9.6     -> v1.9.6
 * [new tag]         v1.9.7     -> v1.9.7
 * [new tag]         v1.9.8     -> v1.9.8
NextCloudPi updated to version v1.49.0

```
<details>

theCalcaholic, 2022-08-26:
@Oberlurch3000 The output looks good, but the PHP upgrade will only be performed on the next nextcloud upgrade (performed with `ncp-update-nc <nc-version>`).

theCalcaholic, 2022-08-26:
@Jannik44 You could try the php upgrade if you like. Just follow the instructions above (quoted for your convenience).

> @Oberlurch3000 Alright, so I have a working upgrader now at the `feature/php8` branch. The best way for you to upgrade would be to first revert your manual upgrade, that means:
> 
>     1. Remove all php  and php8.1 packages (`apt remove --purge php*`)
> 
>     2. Remove the apt respository that you presumably added from /etc/apt/sources.list or /etc/apt/sources.list.d
> 
>     3. Run the following command to reinstall php7.4:
> 
> 
> ```shell
> apt-get install --no-install-recommends php-{common,igbinary,redis} "php7.4" "php7.4"-{curl,gd,fpm,cli,opcache,mbstring,xml,zip,fileinfo,ldap,intl,bz2,json,common,readline,mysql,bcmath,gmp}
> ```
> 
>     4. Restart php-fpm and apache2 to apply the change: `service apache2 restart; service php7.4-fpm start`
> 
>     5. Now, update to the feature branch with: `ncp-update feature/php8`
> 
>     6. Then, upgrade nextcloud to version 24.0.4 and php to version 8.1 with `ncp-update-nc 24.0.4`
> 
> 
> Best of luck!

theCalcaholic, 2022-08-29:
Update: This feature will soon get merged to devel (see #1554 )

theCalcaholic, 2022-08-29:
@Jannik44 Sorry, in my previous comment, I had forgotten, that you didn't install php 8 yet (which makes things a lot easier).
You can upgrade simply with the following two commands:

```sh
sudo ncp-update feature/php8
sudo ncp-update-nc 24.0.4
```

zedocrob, 2022-09-11:
```
sudo ncp-update feature/php8
sudo ncp-update-nc 24.0.4
```
Thanks, worked fine for me too, can we still keep `nc-autoupdate-ncp` active or manually update to this branch `ncp-update feature/php8` for next NCP updates?

theCalcaholic, 2022-09-16:
Please don't use the feature/php8 branch anymore. Changes are now in devel.

If you updated to the feature/php8 branch from any version of NCP prior to v1.49.1, you need to do the following

```sh
ncp-update
ncp-update devel
```

Afterwards, run the nc-limits app from ncp-config (CONFIG -> nc-limits).
Now you should be fine and can leave auto updates enabled

theCalcaholic, 2022-09-22:
PHP 8 support is now in master.

closed 2022-09-22 by theCalcaholic

Oberlurch3000, 2022-09-24:
So just updating NCpi via ```ncp-config``` is all that's nessesary from now on, right?

theCalcaholic, 2022-09-24:
> So just updating NCpi via `ncp-config` is all that's nessesary from now on, right?

Precisely. However, the new images aren't ready yet because of an issue with my deployment pipeline (cross architecture tests fail) which means that docker installations can't be updated yet

Oberlurch3000, 2022-09-24:
> > So just updating NCpi via `ncp-config` is all that's nessesary from now on, right?
> 
> Precisely. However, the new images aren't ready yet because of an issue with my deployment pipeline (cross architecture tests fail) which means that docker installations can't be updated yet

This happens if I perform the updates on a system with php8 already installed:
```
root@nextcloudpi:/home/pi# ncp-config
Downloading updates
Performing updates
Failed to stop php7.4-fpm.service: Unit php7.4-fpm.service not loaded.
Error while applying update 1.50.0. Exiting...
```

sirdrakeistda, 2022-09-24:
> > So just updating NCpi via `ncp-config` is all that's nessesary from now on, right?
> 
> Precisely. However, the new images aren't ready yet because of an issue with my deployment pipeline (cross architecture tests fail) which means that docker installations can't be updated yet

I installed 1.50.0, but php 8 is not activated (still php 7.4). I have the normal ncp version (no docker). According to the changelog php should work. Do I have to run something else too and not only `sudo ncp-update`?

derdershat, 2022-09-24:
I have the same message as @Oberlurch3000 but I updated php7.3 to php7.4 in july and get

> ncp-update
Downloading updates
Performing updates
Failed to stop php7.3-fpm.service: Unit php7.3-fpm.service not loaded.
Error while applying update 1.50.0. Exiting...

> whereis php7.3
php7: /usr/bin/php7.4 /usr/share/php7.4-readline /usr/share/php7.4-zip /usr/share/php7.4-mysql /usr/share/php7.4-mbstring /usr/share/php7.4-gmp /usr/share/php7.4-curl /usr/share/php7.4-common /usr/share/php7.4-bcmath /usr/share/php7.4-intl /usr/share/php7.4-json /usr/share/php7.4-xml /usr/share/php7.4-gd /usr/share/php7.4-opcache

I have nextcloudpi 24.0.4.1 installed on a dietpi debian

theCalcaholic, 2022-09-25:
Manually updating your PHP version will cause these kinds of issues.
The safest approach would be to downgrade first and then follow ncp's supported upgrade path (i. e. `ncp-dist-upgrade` to upgrad PHP 7.3 -> 7.4 and then `ncp-update-nc` to update PHP 7.4 -> 8.1).

derdershat, 2022-09-25:
@theCalcaholic 
I searched my issue and found that after my dietpi dist-upgrade I followed a tutorial to update from 7.3 to 7.4 but that of course didn't changed ncp files. (think I missed ncp-dist-upgrade like you sayed and did an ncp-update at this time)

I found that /usr/local/etc/ncp.cfg was the culprit. There php_version, release, etc. had the old values.
Changed it and after ncp-update and ncp-update-nc everything is fine and works with php 8.1 :wink:

thank you for your help and work on ncp
Grüße

Oberlurch3000, 2022-09-28:
> @theCalcaholic I searched my issue and found that after my dietpi dist-upgrade I followed a tutorial to update from 7.3 to 7.4 but that of course didn't changed ncp files. (think I missed ncp-dist-upgrade like you sayed and did an ncp-update at this time)
> 
> I found that /usr/local/etc/ncp.cfg was the culprit. There php_version, release, etc. had the old values. Changed it and after ncp-update and ncp-update-nc everything is fine and works with php 8.1 😉
> 
> thank you for your help and work on ncp Grüße

Could you explain what you mean with old values?
I'm currently on php 8.0 but my ncp doesn't recognize and shows php wasn't working.
Since I get the same error message like you. Do you think changing the php version in .../ncp.cfg and performing ncp-update 
 ncp-update-nc would solve my issue? I'm not sure if our problem is kind of the same.

theCalcaholic, 2022-09-29:
@Oberlurch3000 If you upgrade PHP manually, NCP won't know about it and try to work with a wrong version of PHP.
Also, ncp's configuration for various PHP features won't be applied.
You can try to tell ncp about the PHP version you have installed, by editing /usr/local/etc/ncp.cfg with the correct PHP version, but the second issue still remains.

If you are lucky, however, editing ncp.cfg will let the PHP upgrade succeed when you run `ncp-update-nc`.
No guarantees though

derdershat, 2022-09-29:
@Oberlurch3000 

for my case I moved late to bullseye and did this with apt dist-upgrade and not with ncp-dist-upgrade.

so my ncp.cfg didn't updated and as I was on php 7.4 in the meantime I got `Failed to stop php7.3-fpm.service: Unit php7.3-fpm.service not loaded` 

old ncp.cfg:

`{
  "nextcloud_version": "24.0.5",
  "php_version": "7.3",
  "release": "buster",
  "release_issue": [
    "Debian GNU/Linux 9",
    "Raspbian GNU/Linux 9"
  ]
}`

switched to ncp.cfg:

`{
  "nextcloud_version": "24.0.5",
  "php_version": "7.4",
  "release": "bullseye",
  "release_issue": [
    "Debian GNU/Linux 11",
    "Raspbian GNU/Linux 11"
  ]
}`

and could update with `ncp-update`

as @theCalcaholic expalined, you could be lucky too if you change your `php_version": "7.4"` in ncp.cfg to `  "php_version": "8.0",` and `ncp-update-nc` to the latest version.

Oberlurch3000, 2022-10-01:
@theCalcaholic so since my ncp was expecting php 7.4 the workflow should look like?:
1. remove php 8.0
2. reinstall php 7.4
3. disable passwords app since it will stop NC from woking correctly without php 8
4. confirm NC is working with php 7.4
5. perform ˋˋˋncp-update-ncˋˋˋ (this will update NC and php?)

That's all?
I've already installed the latest ncp 1.50.1. Is this problematic? I was expecting this to solve my issue.

theCalcaholic, 2022-10-01:
@Oberlurch3000 sounds about right.
Alternatively you could edit `/usr/local/etc/ncp.cfg` and set the php version in there to 8.0

Allthough I'd assume, that the process you suggested is more reliable.

Oberlurch3000, 2022-10-06:
So this is confusing... I tried to go the easy way first just editing the ```ncp.cfg``` file to php 8.0 and then ```ncp-update-nc```.
Since this didn'd work and NC was completly down I restored the backup.
After starting the cloud again the problem php-check showing php was down is gone.

I'm pretty sure I restarted the whole raspi last time I changed the ```ncp.cfg``` but maybe just editing the php version and rebooting is enough. After restoring the backup NC had to configure a few things while accessing the cloud first time via browser. That could also have solved the issue.

githubtefo, 2023-06-25:
Hi there - it's been a while since I postponed this update.
I am in in **v23.0.2.1** with **NCPi 1.52.0**. I tried to update from the Panel but it is sending me to the terminal because I need to upgrade PHP.
What would be the safest way to move forward, please? `ncp-dist-upgrade` and then `ncp-update-nc`? I am also having warnings saying I should update to bullseye available.
Thank you in advance.