nextcloud/nextcloudpi #1543: Backups Failing and other issues...

glenn-sorrentino, 2022-08-19:
Running nc-backup gives the output:
```
tar: data/.../files_versions: Cannot stat: Input/output error
tar: data/files_encryption: Cannot stat: Input/output error
tar: Exiting with failure status due to previous errors
error generating backup
```

<details>
<summary>NextCloudPi diagnostics</summary>

```

PHP Notice:  fwrite(): write of 1580 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 4683 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 1268 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
NextCloudPi version  v1.49.0
NextCloudPi image    NextCloudPi_08-16-22
OS                   Debian GNU/Linux 11. 5.15.56-v8+ (aarch64)
automount            yes
USB devices          sdb sdc 
datadir              /media/myCloudDrive/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      247G/932G
rootfs usage         5.6G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    24.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               33min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

PHP Notice:  fwrite(): write of 2945 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 3106 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 2942 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 3103 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 1607 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
PHP Notice:  fwrite(): write of 4683 bytes failed with errno=5 Input/output error in /var/www/nextcloud/lib/private/Log/File.php on line 89
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
[... 70 lines cut ...]
[Thu Aug 18 21:46:42.871032 2022] [mpm_event:notice] [pid 787:tid 548303582272] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Thu Aug 18 21:46:42.871171 2022] [core:notice] [pid 787:tid 548303582272] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: error reading '/media/myCloudDrive/ncdata/data/nextcloud.log': Input/output error

```
</details>

theCalcaholic, 2022-08-20:
Looks like your disk is broken. You can see that it's not just nc-backup showing input/output errors,  (compare nextcloud logs for example). 
Did the system work before? Did something happen since then?

closed 2022-08-20 by theCalcaholic

glenn-sorrentino, 2022-08-20:
SD or SSD? I doubt the SSD is broken. Everything was working without issue before.

Sent from ProtonMail mobile

-------- Original Message --------

theCalcaholic, 2022-08-20:
No, the SSD (whatever contains your data directory). If you're lucky, it's not a hardware, but a filesystem issue that could be corrected.

I/O errors are definitely caused by the disk or filesystem, though

glenn-sorrentino, 2022-08-20:
what's your recommendation? format and try again?

Sent from ProtonMail mobile

-------- Original Message --------

glenn-sorrentino, 2022-08-21:
Seems odd for two SSDs to go down at the same time?

Sent from ProtonMail mobile

-------- Original Message --------

glenn-sorrentino, 2022-08-23:
Just posting here some progress:

- I formatted the drives as exFAT, used nc-format-USB to format as BTRFS, both successfully.
- Using the wizard I moved the data-dir, though it gave the same error as last time it also worked like last time.
- Run nc-fix-permissions successfully
- Log in to admin account successfully
- Add and enable 2FA, encryption module
- Add user account
**- Log in and test upload a few files - FAIL**
     - Expected filesize of 252202 bytes but read (from Nextcloud client) and wrote (to Nextcloud storage) 8192 bytes. Could either be a network problem on the sending side or a problem writing to the storage on the server side.
     **- Add <IfModule reqtimeout_module>  RequestReadTimeout header=0  RequestReadTimeout body=0 </IfModule> to apache2.conf**
- Reboot 
- Try uploading again - Succeed
- Auth into the desktop app, successful
- Delete files locally, verified deleted on server.
- Perform backup, successful
- Log in to nextcloud on phone
    - Turn on automatic uploads 
    - Uploading. So good so far.