nextcloud/nextcloudpi #1549: HPB / notify_push doesn't start automatically after reboot

Oberlurch3000, 2022-08-23:
After rebooting the system I always have to start the HPB manually via ```systemctl```.
Is there an error or setting I miss out?

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.49.0
NextCloudPi image    NextCloudPi_03-24-22
OS                   Debian GNU/Linux 11. 5.15.32-v8+ (aarch64)
automount            yes
USB devices          sda sdb
datadir              /media/Nextcloud/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      1.9T/4.6T
rootfs usage         4.4G/59G
swapfile             /var/swap
dbdir                /media/Nextcloud/ncdatabase
Nextcloud check      ok
Nextcloud version    24.0.4.1
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "oberlurch3000.de",
            "1": "192.168.178.52",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "20": "192.168.178.31",
            "21": "oberlurch3000.de",
            "22": "raspberrypi",
            "12": "oberlurch3000.de",
            "3": "oberlurch3000.de",
            "": "raspberrypi",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.4.1",
        "overwrite.cli.url": "https:\/\/oberlurch3000.de\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
[... 111 lines cut ...]
```

{"reqId":"YwSUJSKNvvBokPZOV52yqAAAAAk","level":3,"time":"2022-08-23T08:48:09+00:00","remoteAddr":"145.14.232.132","user":"Carsten","app":"index","method":"GET","url":"/core/preview?fileId=1055559&x=64&y=64&a=true","message":"Could not create folder","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.102 Safari/537.36 Edg/104.0.1293.63","version":"24.0.4.1","exception":{"Exception":"OCP\\Files\\NotPermittedException","Message":"Could not create folder","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Files/AppData/AppData.php","line":156,"function":"newFolder","class":"OC\\Files\\Node\\Folder","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/Storage/Root.php","line":74,"function":"newFolder","class":"OC\\Files\\AppData\\AppData","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/Generator.php","line":573,"function":"newFolder","class":"OC\\Preview\\Storage\\Root","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/Generator.php","line":134,"function":"getPreviewFolder","class":"OC\\Preview\\Generator","type":"->"},{"file":"/var/www/nextcloud/lib/private/Preview/Generator.php","line":111,"function":"generatePreviews","class":"OC\\Preview\\Generator","type":"->"},{"file":"/var/www/nextcloud/lib/private/PreviewManager.php","line":212,"function":"getPreview","class":"OC\\Preview\\Generator","type":"->"},{"file":"/var/www/nextcloud/core/Controller/PreviewController.php","line":169,"function":"getPreview","class":"OC\\PreviewManager","type":"->"},{"file":"/var/www/nextcloud/core/Controller/PreviewController.php","line":142,"function":"fetchPreview","class":"OC\\Core\\Controller\\PreviewController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":225,"function":"getPreviewByFileId","class":"OC\\Core\\Controller\\PreviewController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":133,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":172,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":298,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1023,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/Files/Node/Folder.php","Line":164,"CustomMessage":"--"}}
{"reqId":"YwSYFc3dtQte8pJkJ4-2jwAAAAE","level":2,"time":"2022-08-23T09:04:21+00:00","remoteAddr":"143.92.32.36","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"143.92.32.36\" tried to access using \"109.250.217.72\" as host.","userAgent":"<title>(.*?)<\\/title>","version":"24.0.4.1","data":{"app":"core"}}
{"reqId":"YwScYnUomwrVZK-OAu_GxgAAAAU","level":2,"time":"2022-08-23T09:22:42+00:00","remoteAddr":"92.53.65.52","user":"--","app":"core","method":"GET","url":"/remote/fgt_lang?lang=/../../../..//////////dev/cmdb/sslvpn_websession","message":"Trusted domain error. \"92.53.65.52\" tried to access using \"109.250.217.72:443\" as host.","userAgent":"Python-urllib/3.8","version":"24.0.4.1","data":{"app":"core"}}
{"reqId":"YwSoIxWIgs4_glbZ5C0fOAAAAAI","level":3,"time":"2022-08-23T10:12:53+00:00","remoteAddr":"145.14.232.132","user":"Carsten","app":"mail","method":"GET","url":"/apps/mail/api/messages/1653/body","message":"Could not load message","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.102 Safari/537.36 Edg/104.0.1293.63","version":"24.0.4.1","exception":{"Exception":"OCA\\Mail\\Exception\\ServiceException","Message":"Could not load message","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/mail/lib/Controller/MessagesController.php","line":248,"function":"getImapMessage","class":"OCA\\Mail\\Service\\MailManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":225,"function":"getBody","class":"OCA\\Mail\\Controller\\MessagesController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":133,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":172,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":298,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1023,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/apps/mail/lib/Service/MailManager.php","Line":187,"Previous":{"Exception":"OCP\\AppFramework\\Db\\DoesNotExistException","Message":"Message does not exist","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/mail/lib/Service/MailManager.php","line":181,"function":"find","class":"OCA\\Mail\\IMAP\\MessageMapper","type":"->"},{"file":"/var/www/nextcloud/apps/mail/lib/Controller/MessagesController.php","line":248,"function":"getImapMessage","class":"OCA\\Mail\\Service\\MailManager","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":225,"function":"getBody","class":"OCA\\Mail\\Controller\\MessagesController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":133,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":172,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":298,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1023,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/apps/mail/lib/IMAP/MessageMapper.php","Line" [... cut ...] :"Mozilla/5.0 (Windows NT 5.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.90 Safari/537.36","version":"24.0.4.1","data":{"app":"core"}}
{"reqId":"YwUE6OggtqXMBLyE8ltxVwAAAAI","level":2,"time":"2022-08-23T16:48:40+00:00","remoteAddr":"64.62.197.92","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"64.62.197.92\" tried to access using \"109.250.217.72\" as host.","userAgent":"--","version":"24.0.4.1","data":{"app":"core"}}

```
</details>

Oberlurch3000, 2022-08-24:
Ok, got it solved by adding the following lines to ˋˋˋnano /etc/systemd/system/notify_push.serviceˋˋˋ after ˋˋˋExecStart=ˋˋˋ:
ˋˋˋRestart=on-failureˋˋˋ
ˋˋˋRestartSec=10ˋˋˋ
The HPB was starting up before my domain was available so it run into an error. Adding this lines tries it again until all services are fully running.

closed 2022-08-24 by Oberlurch3000

theCalcaholic, 2022-08-28:
Thanks for your solution. A similar configuration should probably be the default

theCalcaholic, 2022-09-26:
Implemented in the upcoming version v1.50.1