nextcloud/nextcloudpi #1553: Proxmox image not working?

nacho-domatix, 2022-08-27:
[NextCloudPi_LXD_v1.49.0.tar.gz](https://github.com/nextcloud/nextcloudpi/releases/download/v1.49.0/NextCloudPi_LXD_v1.49.0.tar.gz)

I'm trying to install/use this image in a clean Proxmox server. Version was 7.0 upgraded to 7.2 in order to get full support of Debian 11.

I download (wget) the image in XXX already works as a image, then I proceed to create the container using the image and the output message is:

`()
Formatting '/var/lib/vz/images/1011/vm-1011-disk-0.raw', fmt=raw size=8589934592 preallocation=off
Creating filesystem with 2097152 4k blocks and 524288 inodes
Filesystem UUID: e707c089-0c1d-44a0-9adc-f96723613c0a
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632
extracting archive '/var/lib/vz/template/cache/NextCloudPi_LXD_v1.49.0.tar.gz'
Total bytes read: 1620920320 (1.6GiB, 128MiB/s)
Architecture detection failed: open '/bin/sh' failed: No such file or directory

Falling back to amd64.
Use `pct set VMID --arch ARCH` to change.
/etc/os-release file not found and autodetection failed, falling back to 'unmanaged'
TASK OK`

I tried `pct set 1001 --arch arm64` and other architectures, but none of them worked.

Also failed with the script under Debian 11, but that should be another issue.

theCalcaholic, 2022-08-27:
@nacho-domatix Hi, thanks for the report. Can you please describe exactly how you tried to run the image? I'm not familiar with Proxmox, but for the LXD image it shouldn't matter very much how you try to run it.

ZendaiOwl, 2022-08-27:
@theCalcaholic 
Unfortunately it does matter how you run it, it also matters how you unpack the image to get it working on proxmox 🙏

@nacho-domatix 
I have working NCP VM on my Proxmox, you can find the steps to get it to work in this article [here](https://help.nextcloud.com/t/making-it-possible-nextcloudpi-lxd-on-proxmox/129476) on the forum with me, michuvon and others have made posts on how we got it working

nacho-domatix, 2022-08-27:
@theCalcaholic there's actually just one way to do it, image template to create a container in Proxmox:

[image]

@ZendaiOwl before I tried I read both links, and unfortunatelly they're not updated, they kinda of useless with the latest version of the image.

theCalcaholic, 2022-08-27:
You're trying to create an LXC container from a LXD image - I don't think that can work...

I'll send you an LXC image to try

nacho-domatix, 2022-09-06:
> You're trying to create an LXC container from a LXD image - I don't think that can work...
> 
> I'll send you an LXC image to try

Any news on this? I wasn't able to make it work under Proxmox yet... :(

nacho-domatix, 2022-09-14:
I'm still trying to make it work from scratch, but running the script after a clean installation of Debian fails. Apparently everything works fine, but after finishing I'm unable to enter the "start page". IP leads me to a page where "domain is not trusted" and when I add the IP in /var/www/nextcloud/config/config.php there is already a Nextcloud instance working (no idea what password).

Trying to "restart configuration" (through ncp-config) it gives me this:

Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Username is invalid because files already exist for this user
Nextcloud is not installed - only a limited number of commands are available

                                                                
  There are no commands defined in the "background" namespace.  
                                                                

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  

And after finished, I'm able to get into pi configuration (4443 port) but it says "HPB service is down", and nextcloud seems to be not working...

Not sure what to do with a clean Debian under Proxmox... :(

theCalcaholic, 2022-09-15:
Hey, sorry for not getting back to you. 
Have you tried following this guide in the wiki?

https://help.nextcloud.com/t/nextcloudpi-vm-on-proxmox/126444

I have no experience with proxmox myself, but from what you describe, it sounds like you have successfully installed ncp but can't access the admin interface, because it's only available from the same network.
I recommend you access it via ssh and use `ncp-config` to configure domain and trusted IP address

nacho-domatix, 2022-09-15:
Hi @theCalcaholic, many thanks for the response.
I tried that guide too, even tho it's a little bit outdated (debian 10 is no longer supported). 
I did also what you said, enter via ssh and use ncp-config, but when the process is finished I have no longer a Nextcloud instance installed (ask me about database path to install it from Nextcloud, not NCPi).
Shouldn't be different from install it from a non-virtual Debian, I'm not really sure what's going on... :/

theCalcaholic, 2022-09-15:
Can you please attach the output of the installer and the output of ncp-report after the installation finished?

nacho-domatix, 2022-09-15:
Hi, I made another fresh install. 

`Obj:1 http://ftp.debian.org/debian bullseye InRelease
Obj:2 http://security.debian.org bullseye-security InRelease
Obj:3 http://ftp.debian.org/debian bullseye-updates InRelease
Des:4 http://ftp.debian.org/debian bullseye/main Translation-es [318 kB]
Descargados 318 kB en 0s (822 kB/s)
Leyendo lista de paquetes...
Leyendo lista de paquetes...
Creando árbol de dependencias...
Leyendo la información de estado...
ca-certificates ya está en su versión más reciente (20210119).
Se instalarán los siguientes paquetes adicionales:
  git-man liberror-perl
Paquetes sugeridos:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn
Paquetes recomendados:
  patch
Se instalarán los siguientes paquetes NUEVOS:
  git git-man liberror-perl lsb-release sudo
0 actualizados, 5 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Se necesita descargar 8.473 kB de archivos.
Se utilizarán 42,7 MB de espacio de disco adicional después de esta operación.
Des:1 http://ftp.debian.org/debian bullseye/main amd64 liberror-perl all 0.17029-1 [31,0 kB]
Des:2 http://ftp.debian.org/debian bullseye/main amd64 git-man all 1:2.30.2-1 [1.827 kB]
Des:3 http://ftp.debian.org/debian bullseye/main amd64 git amd64 1:2.30.2-1 [5.527 kB]
Des:4 http://ftp.debian.org/debian bullseye/main amd64 lsb-release all 11.1.0 [27,9 kB]
Des:5 http://ftp.debian.org/debian bullseye/main amd64 sudo amd64 1.9.5p2-3 [1.059 kB]
Descargados 8.473 kB en 0s (18,0 MB/s)
Seleccionando el paquete liberror-perl previamente no seleccionado.
(Leyendo la base de datos ... 
(Leyendo la base de datos ... 5%
(Leyendo la base de datos ... 10%
(Leyendo la base de datos ... 15%
(Leyendo la base de datos ... 20%
(Leyendo la base de datos ... 25%
(Leyendo la base de datos ... 30%
(Leyendo la base de datos ... 35%
(Leyendo la base de datos ... 40%
(Leyendo la base de datos ... 45%
(Leyendo la base de datos ... 50%
(Leyendo la base de datos ... 55%
(Leyendo la base de datos ... 60%
(Leyendo la base de datos ... 65%
(Leyendo la base de datos ... 70%
(Leyendo la base de datos ... 75%
(Leyendo la base de datos ... 80%
(Leyendo la base de datos ... 85%
(Leyendo la base de datos ... 90%
(Leyendo la base de datos ... 95%
(Leyendo la base de datos ... 100%
(Leyendo la base de datos ... 19045 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../liberror-perl_0.17029-1_all.deb ...
Desempaquetando liberror-perl (0.17029-1) ...
Seleccionando el paquete git-man previamente no seleccionado.
Preparando para desempaquetar .../git-man_1%3a2.30.2-1_all.deb ...
Desempaquetando git-man (1:2.30.2-1) ...
Seleccionando el paquete git previamente no seleccionado.
Preparando para desempaquetar .../git_1%3a2.30.2-1_amd64.deb ...
Desempaquetando git (1:2.30.2-1) ...
Seleccionando el paquete lsb-release previamente no seleccionado.
Preparando para desempaquetar .../lsb-release_11.1.0_all.deb ...
Desempaquetando lsb-release (11.1.0) ...
Seleccionando el paquete sudo previamente no seleccionado.
Preparando para desempaquetar .../sudo_1.9.5p2-3_amd64.deb ...
Desempaquetando sudo (1.9.5p2-3) ...
Configurando liberror-perl (0.17029-1) ...
Configurando sudo (1.9.5p2-3) ...
Configurando git-man (1:2.30.2-1) ...
[... 2738 lines cut ...]
<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"zmFkf1NZJbXK5i8dNuYI","level":3,"time":"2022-09-15T14:50:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":663,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1104,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":282,"CustomMessage":"--"}}
{"reqId":"dcSHZbEzdbHW91Y137WU","level":3,"time":"2022-09-15T14:55:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":663,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1104,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":282,"CustomMessage":"--"}}
{"reqId":"0viN9vLqPcjt6R9MtoHH","level":3,"time":"2022-09-15T15:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":663,"function":"checkInstalled","class":"OC","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1104,"function":"init","class":"OC","type":"::"},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":282,"CustomMessage":"--"}}
[... cut ...]
{"reqId":"FEXSsJkra1OIMJOzdO4Y","level":2,"time":"2022-09-15T18:32:48+00:00","remoteAddr":"192.168.1.222","user":"--","app":"core","method":"GET","url":"/index.php/js/core/merged-template-prepend.js?v=5dc31c1b-0","message":"Trusted domain error. \"192.168.1.222\" tried to access using \"192.168.1.91\" as host.","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:104.0) Gecko/20100101 Firefox/104.0","version":"24.0.3.2","data":{"app":"core"}}

```
</details>
`
And that's all I can say. Sorry for the screenshots, are in spanish, but I guess that won't be a problem.

I never had this issues before (Debian 10) using same procedure.

theCalcaholic, 2022-09-16:
Are you sure that the installer was run on a fresh debian 11 VM? It looks like NCP has been installed there previously

nacho-domatix, 2022-09-16:
Yes, totally, 100%. Several times actually.

theCalcaholic, 2022-09-22:
Just FYI: Now that I pushed the NC 24/PHP 8 update, I should have more time to look into this

nacho-domatix, 2022-09-22:
Great!

I recently discovered this project https://github.com/tteck/Proxmox

Maybe you guys can work together:

https://github.com/tteck/Proxmox/discussions/353

nacho-domatix, 2022-09-23:
It's done!!
`bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/ct/nextcloudpi-v4.sh)"`
I'll try today and let you know.

nacho-domatix, 2022-09-23:
Ok, it works! Following the instructions:
`sudo ncp-config`
Go to config arrow_right nc-trusted-domains, add `0.0.0.0` or a static NextCloudPi IP
`sudo service apache2 restart`
And there you go... 🎉🎉
(I guess that instructions works well only with the script, but I tried with the whole LXC script - in the previous link)

You can close this issue anytime :)

ZendaiOwl, 2022-09-23:
Closing this issue since a #resolution was found :) 

Thank you @nacho-domatix :pray:

closed 2022-09-23 by ZendaiOwl

ZendaiOwl, 2022-09-28:
@nacho-domatix 
I went through my notes that I've kept since I started tinkering with a RPi4 & Proxmox etc. and found the method I used to install the NCP LXD image on my Proxmox ^^. 

I imported the LXD image as a VM, I even kept snapshots 

[image]

Now I don't know if I kept notes of the entire process or not, there is quite a lot though but spread out over 2 or 3 txt documents, if I remember correctly ..