nextcloud/nextcloudpi #1556: [NCP] Internal Server Error &  1146 Table 'nextcloud.oc_mounts' doesn't exist`

kmpfwrst, 2022-09-03:
Hello,
since a few days I cannot access my nextcloudpi via hte DNS nor the IP. But it is possible to enter the admin panel.
I am getting the error "Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log."

Withe following command I get:
`ncc notify_push:self-test
 An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'nextcloud.oc_mounts' doesn't exist`

I used google and internal search but didn'T find a solution for this bug.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.49.0
NextCloudPi image    NextCloudPi_01-14-19
OS                   Raspbian GNU/Linux 10. 5.10.103-v7+ (armv7l)
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      693G/932G
rootfs usage         3.5G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    23.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              192.168.178.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               59min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "***REMOVED SENSITIVE VALUE***",
            "4": "***REMOVED SENSITIVE VALUE***",
            "20": "192.168.178.1",
            "2": "***REMOVED SENSITIVE VALUE***",
[... 78 lines cut ...]
```

2022-09-03 11:57:48 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-09-03 11:57:48 0 [Note] InnoDB: Number of pools: 1
2022-09-03 11:57:48 0 [Note] InnoDB: Using generic crc32 instructions
2022-09-03 11:57:48 0 [Note] InnoDB: Initializing buffer pool, total size = 1G, instances = 1, chunk size = 128M
2022-09-03 11:57:48 0 [Note] InnoDB: Completed initialization of buffer pool
2022-09-03 11:57:48 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-09-03 11:57:49 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2022-09-03 11:57:49 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-09-03 11:57:49 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-09-03 11:57:49 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-09-03 11:57:49 0 [Note] InnoDB: 10.3.34 started; log sequence number 15738077585; transaction id 61195219
2022-09-03 11:57:49 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2022-09-03 11:57:49 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-09-03 11:57:49 0 [Note] Server socket created on IP: '127.0.0.1'.
2022-09-03 11:57:49 0 [Note] Reading of all Master_info entries succeeded
2022-09-03 11:57:49 0 [Note] Added new Master_info '' to hash table
2022-09-03 11:57:49 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.34-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
2022-09-03 11:57:49 0 [Note] InnoDB: Buffer pool(s) load completed at 220903 11:57:49
2022-09-03 11:57:55 10 [Warning] Aborted connection 10 to db: 'nextcloud' user: 'ncadmin' host: 'localhost' (Got an error reading communication packets)

```
</details>

damadis, 2022-09-18:
Hi kmpfwurst,

i'm having the same Problem after My NCP Instance updated itself on saturday morning from Nextcloud Version 23.0.6.1 to 24.0.3, but ran into an problem with the not installed PHP7.4. The Rollback failed and Nextcloud remained in maintenance mode. I switched it off in the NCP Admin Panel, but i cant access my Nextcloud. Since then i'm searching for a hint or solution to get it back running.

The /var/log/ncp.log contained this:
```
Sat 17 Sep 06:25:03 CEST 2022 - Running /etc/cron.daily/ncp-autoupdate...
[ nc-update-nc-apps-auto ]
checking for updates...
[ncp-update-nc]
Current   Nextcloud version 23.0.6.1
Available Nextcloud version 24.0.3
Download Nextcloud 24.0.3...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20220917_1663389339.tar.gz generated
Maintenance mode disabled
Install Nextcloud 24.0.3...
Fix permissions...
Upgrade...
This version of Nextcloud requires at least PHP 7.4<br/>You are currently running 7.3.31-1~deb10u1. Please update your PHP version.Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20220917_1663389339-23.0.6.1.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20220917_1663389339-23.0.6.1.tar.gz...
restore files...
restore database...
ERROR 2006 (HY000) at line 697: MySQL server has gone away
Error restoring nextcloud database
Cleanup...
Rollback failed! Data left at /var/www/ncp-data.ADJT3Q

```

theCalcaholic, 2022-09-26:
Hi, sorry for the late reply. There's been a lot going on lately with the v1.50.0 release that kept me busy.

First of all: The cause for your issues will be prevented going forward by a check in the ncp-update-nc script. However, unfortunately this doesn't help you anymore.

In your case I recommend to restore the backup that has been created before the attempted update (`var/www//nextcloud-bkp_20220917_1663389339.tar.gz` in @damadis' case; @kmpfwrst, please check your /var/log/ncp.log for the exact path).
To do so, first make sure that mariadb is running (`sudo service mariadb start`) and then run `sudo ncp-restore /var/www/nextcloud-bkp_20220917_1663389339.tar.gz`. Hopefully the restore will succeed this time

damadis, 2022-09-27:
@theCalcaholic Thanks for your reply. I allready restored the backup and my NC runs fine again. I deactivated the autoupdate_nc for now and wait until the problem with PHP8 in #1542 is resolved to make the upgrade to bullseye

kmpfwrst, 2022-09-27:
Sorry I dont get it - what excatly do you mean "the exact path". I say i cannot find the back up. ALready tried it before. Do you know how I can access my data on the external drive? If i can access it i just reinstall everything.

ZendaiOwl, 2022-09-27:
He means that the location of your backup and the exact path where it should be located is written in the ncp.log 🙂

You should be able to access the files by connecting the drive to another Linux computer and accessing it that way. You should be able to move the files as root, it can potentially mess up the permissions a little, if that happens it's easily fixed with the chown or chmod command

theCalcaholic, 2022-09-29:
@kmpfwrst Connect to your ncp instance using ssh or by connecting a keyboard and a screen.

Check the output of `sudo less /var/log/ncp.log` It should contain at least one line like this:
```
backup /data/docker-startup-backups/nextcloud-bkp_20220928_1664409578.tar.gz generated
```
This file contains your configuration, database and nextcloud installation. It should also contain your data if you didn't move the data directory.
If you moved the data directory, it should still be in the location you moved it to.

closed 2023-01-21 by theCalcaholic