nextcloud/nextcloudpi #1558: Failed to start Rotate log files NC24

zedocrob, 2022-09-07:
**System information**

zedocrob, 2022-09-11:
Looks like i found the problem:
`/etc/logrotate.d/ufw` was malformed

```
/var/log.hdd/ufw.log
{
        rotate 4
        weekly
maxsize2M          (NOT OK! deleted this line)
        maxsize 2M (and ALIGN LIKE THIS)
        missingok
        notifempty
        compress
        delaycompress
        sharedscripts
        postrotate
                invoke-rc.d rsyslog rotate >/dev/null 2>&1 || true
        endscript
}
```

logrotate,service starting fine now

theCalcaholic, 2022-09-14:
Thanks for the report! Will be fixed in v1.49.2 or v1.50.0 (depending on how fast I manage to release the latter)

theCalcaholic, 2022-09-26:
Should be fixed in v1.50.0

closed 2022-09-26 by theCalcaholic