nextcloud/nextcloudpi #1564: First Use Wizard: failing when moving data directory to USB (again)

hanselpeter, 2022-09-16:
Using nextcloudpi with image version 1.49.0
System was already completly updated to latest verions (nextcloud and OS).

had it all set up and tried to move data to a new external SSD (connected via USB) and it failed. After that I thought to myself it would be better to start again from scratch. I fromatted the sdcard and flashed the imag again. Using the wizard right at the beginning to move the data to my USB drive i formated the USB drive. But during execution of nc-datadir i got the following error:

`[ nc-automount ] (Fri Sep 16 18:35:45 BST 2022)
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /lib/systemd/system/nc-automount.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /lib/systemd/system/nc-automount-links.service.
automount enabled

[ nc-format-USB ] (Fri Sep 16 18:36:01 BST 2022)
/dev/sda: 2 bytes were erased at offset 0x000001fe (dos): 55 aa
Drive sda formatted successfuly and labeled myCloudDrive

[ nc-datadir ] (Fri Sep 16 18:36:13 BST 2022)
Maintenance mode enabled
moving data directory from /var/www/nextcloud/data to /media/USBdrive/ncdata...
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value logfile set to string /media/USBdrive/ncdata/data/nextcloud.log
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

System config value tempdirectory set to string /media/USBdrive/ncdata/data/tmp
Maintenance mode disabled`

Unfortunately because of this nextcloud is not usable for me.

dekzi, 2022-09-16:
[What Tobias said](https://github.com/nextcloud/nextcloudpi/issues/1532#issuecomment-1221545661) in #1532 seems to be correct, meaning that the data directory is actually being moved without any issues, other than not reporting its success. This does mean that you should be able to move onto the next step in the wizard by clicking "external access" at the top, despite the warning. I've had no issues after trying this myself.

Also, please keep in mind that although #1532 and #1517 were prematurely closed, creating a third duplicate issue will only further decentralize discussion of this problem, and make finding a solution harder.

theCalcaholic, 2022-09-16:
I think I do understand the cause for this issue now. As @dekzi said, it shouldn't actually break anything.
I'll fix it next week (don't have access to a computer right now).

theCalcaholic, 2022-09-28:
[Patch is now in devel](https://github.com/nextcloud/nextcloudpi/commit/6d71e53a7023784275649927667edce82c1d052f)

theCalcaholic, 2022-09-29:
And now in master

closed 2022-09-29 by theCalcaholic