nextcloud/nextcloudpi #1568: Docker-compose fails to restart

sunjam, 2022-09-22:
# Steps to re-create
1.  `docker-compose up -d` using the official compose script
2.  `nc-update` for ncp or nextcloud to a newer version
3. `docker-compose down`
4. `docker-compose up`

```
$ docker-compose up
Creating network "nextcloudpi_default" with the default driver
Creating nextcloudpi ... done
Attaching to nextcloudpi
nextcloudpi    | Starting PHP-fpm
nextcloudpi    | Starting Apache
nextcloudpi    | bash: /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh: No such file or directory
nextcloudpi    | /usr/local/etc/library.sh: line 220: ${3:}: bad substitution
nextcloudpi    | Starting Redis
nextcloudpi    | Starting Cron
nextcloudpi    | Starting Postfix
nextcloudpi    | ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)
nextcloudpi    | Init done
```

sunjam, 2022-09-22:
I've also tried this compose fail I adapted from the run command.

```
version: '3.3'
services:
    nextcloudpi:
        container_name: nextcloudpi
        image: ownyourbits/nextcloudpi
        ports:
            - '4443:4443'
            - '443:443'
            - '80:80'
        volumes:
            - '/path/to/ncdata:/data'
        restart: unless-stopped
```

ZendaiOwl, 2022-09-23:
So I'll leave a report of what I've found so far when reproducing this and also testing out what it might be 

The update of NCP and upgrade of NC works within the docker container, after using `docker-compose down && docker-compose up -d`, stopping the running container and then starting it up again it is back on the previous version of NCP v1.48.3 with PHP 7.4 and NC v.24.0.4 upgrade still remaining and the docker instance can no longer locate the `/usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh` file makinng the database crash. Creating this file and restarting the container with `docker-compose restart nexxtclodupi` resolves this bit but then there is an upgrade for NC and the apps waiting that needs to be done with `ncc upgrade`.

<details>
<summary>Missing file</summary>

```bash
└࿓❯ sudo docker-compose logs
Attaching to nextcloudpi
nextcloudpi    | Starting PHP-fpm
nextcloudpi    | Starting Apache
nextcloudpi    | bash: /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh: No such file or directory
nextcloudpi    | Starting mariaDB
nextcloudpi    | 2022-09-23  2:36:30 0 [Note] mysqld (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 220 ...
nextcloudpi    | 2022-09-23  2:36:30 0 [Warning] Can't create test file /var/lib/mysql/5b901702e8e8.lower-test
nextcloudpi    | mysqld: Can't change dir to '/var/lib/mysql/' (Errcode: 2 "No such file or directory")
nextcloudpi    | 2022-09-23  2:36:30 0 [ERROR] Aborting
```

</details>

<details>
<summary>After adding the missing file</summary>

```bash
└࿓❯ sudo docker exec -it nextcloudpi bash
root@5b901702e8e8:/# nano /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh
bash: nano: command not found
root@5b901702e8e8:/# cat > /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh <<'EQ'
> #! /bin/bash

set -e
source /usr/local/etc/library.sh

if [[ "$1" == "--defaults" ]]
then
  echo "INFO: Restoring template to default settings" >&2
  DB_DIR=/var/lib/mysql
else
  if [[ "$DOCKERBUILD" -eq 1 ]]
  then
    echo "INFO: Docker build detected." >&2
    DB_DIR=/data-ro/database
  elif is_docker
  then
    echo "INFO: Docker container detected." >&2
    DB_DIR=/data/database
  else
    DB_DIR="$(source "${BINDIR}/CONFIG/nc-database.sh"; tmpl_db_dir)"
  fi
fi

# configure MariaDB (UTF8 4 byte support)
cat <<EOF
[mysqld]
datadir = ${DB_DIR?}
EOF
> EQ
root@5b901702e8e8:/# cat /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh
#! /bin/bash

set -e
source /usr/local/etc/library.sh

if [[ "$1" == "--defaults" ]]
then
  echo "INFO: Restoring template to default settings" >&2
  DB_DIR=/var/lib/mysql
else
[... 674 lines cut ...]
data filesystem      ext2/ext3
data disk usage      412G/3.6T
rootfs usage         412G/3.6T
swapfile             /dev/sda3
dbdir                /data-ro/database
Nextcloud check      ok
Nextcloud version    24.0.4.1
HTTPD service        down
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              172.22.0.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1:15
```

ZendaiOwl, 2022-09-23:
Also found unresolved environment variable in `/etc/apache2/apache2.conf`

```bash
root@5b901702e8e8:/# cat /etc/apache2/apache2.conf
# This is the main Apache server configuration file.  It contains the
# configuration directives that give the server its instructions.
# See http://httpd.apache.org/docs/2.4/ for detailed information about
# the directives and /usr/share/doc/apache2/README.Debian about Debian specific
# hints.
#
#
# Summary of how the Apache 2 configuration works in Debian:
# The Apache 2 web server configuration in Debian is quite different to
# upstream's suggested way to configure the web server. This is because Debian's
# default Apache2 installation attempts to make adding and removing modules,
# virtual hosts, and extra configuration directives as flexible as possible, in
# order to make automating the changes and administering the server as easy as
# possible.

# It is split into several files forming the configuration hierarchy outlined
# below, all located in the /etc/apache2/ directory:
#
#	/etc/apache2/
#	|-- apache2.conf
#	|	`--  ports.conf
#	|-- mods-enabled
#	|	|-- *.load
#	|	`-- *.conf
#	|-- conf-enabled
#	|	`-- *.conf
# 	`-- sites-enabled
#	 	`-- *.conf
#
#
# * apache2.conf is the main configuration file (this file). It puts the pieces
#   together by including all remaining configuration files when starting up the
#   web server.
#
# * ports.conf is always included from the main configuration file. It is
#   supposed to determine listening ports for incoming connections which can be
#   customized anytime.
#
# * Configuration files in the mods-enabled/, conf-enabled/ and sites-enabled/
#   directories contain particular configuration snippets which manage modules,
#   global configuration fragments, or virtual host configurations,
#   respectively.
#
#   They are activated by symlinking available configuration files from their
#   respective *-available/ counterparts. These should be managed by using our
#   helpers a2enmod/a2dismod, a2ensite/a2dissite and a2enconf/a2disconf. See
#   their respective man pages for detailed information.
#
# * The binary is called apache2. Due to the use of environment variables, in
#   the default configuration, apache2 needs to be started/stopped with
#   /etc/init.d/apache2 or apache2ctl. Calling /usr/bin/apache2 directly will not
#   work with the default configuration.

# Global configuration
#

#
# ServerRoot: The top of the directory tree under which the server's
# configuration, error, and log files are kept.
#
# NOTE!  If you intend to place this on an NFS (or otherwise network)
# mounted filesystem then please read the Mutex documentation (available
# at <URL:http://httpd.apache.org/docs/2.4/mod/core.html#mutex>);
# you will save yourself a lot of trouble.
#
# Do NOT add a slash at the end of the directory path.
[... 431 lines cut ...]
#
# Note that the use of %{X-Forwarded-For}i instead of %h is not recommended.
# Use mod_remoteip instead.
#
LogFormat "%v:%p %h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" vhost_combined
LogFormat "%h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" combined
LogFormat "%h %l %u %t \"%r\" %>s %O" common
LogFormat "%{Referer}i -> %U" referer
LogFormat "%{User-agent}i" agent

# Include of directories ignores editors' and dpkg's backup files,
# see README.Debian for details.

# Include generic snippets of statements
IncludeOptional conf-enabled/*.conf

# Include the virtual host configurations:
IncludeOptional sites-enabled/*.conf

# vim: syntax=apache ts=4 sw=4 sts=4 sr noet
ServerName localhost
<IfModule mod_security2.c>
  SecServerSignature " "
</IfModule>
```

ZendaiOwl, 2022-09-23:
I'll also add, before I go to bed, whenever I try to run the :latest image it keeps appending "ncp_" to the volumes, it also creates "ncp_network", which it did not do previously

ZendaiOwl, 2022-09-23:
@sunjam @theCalcaholic 
So I went through a couple more testing steps today, I haven't quite reached something fully conclusive yet but I'll add the information I've gathered so far here in the issue with my conclusions at the end and I have to stop here for now. 

I need to return to my schoolwork :pray: 
I hope this will give some more clarity around the situation however and maybe someone knows a fix for it other than the conclusion I've currently reached, which you can read at the end of this comment. :smile_cat: 

#### NEW TESTS

#### docker-compose.yml - 1st version
```
services:
  nextcloudpi:
    cap_add:
    - SYS_ADMIN
    command: 192.168.178.15
    container_name: nextcloudpi
    devices:
    - /dev/fuse:/dev/fuse
    environment:
      APACHE_LOCK_DIR: /var/lock/apache2
      APACHE_LOG_DIR: /var/log/apache2
      APACHE_PID_FILE: /var/run/apache2.pid
      APACHE_RUN_DIR: /var/run/apache2
      APACHE_RUN_GROUP: www-data
      APACHE_RUN_USER: www-data
      HOSTNAME: berrybox
    image: ownyourbits/nextcloudpi:latest
    ports:
    - published: 80
      target: 80
    - published: 443
      target: 443
    - published: 4443
      target: 4443
    restart: unless-stopped
    volumes:
    - ncdata:/data:rw
    - database:/var/lib/mysql:rw
    - /etc/localtime:/etc/localtime:ro
version: '3.3'
volumes:
  ncdata:
    external: false
  database:
    external: false
```
#### docker-compose.yml - 2nd version - Incorrect configuration, don't use
```
services:
  nextcloudpi:
    cap_add:
    - SYS_ADMIN
    command: 192.168.178.15
    container_name: nextcloudpi
    devices:
    - /dev/fuse:/dev/fuse
    environment:
      APACHE_LOCK_DIR: /var/lock/apache2
      APACHE_LOG_DIR: /var/log/apache2
      APACHE_PID_FILE: /var/run/apache2.pid
      APACHE_RUN_DIR: /var/run/apache2
      APACHE_RUN_GROUP: www-data
      APACHE_RUN_USER: www-data
    image: ownyourbits/nextcloudpi:latest
    ports:
    - published: 80
      target: 80
    - published: 443
      target: 443
    - published: 4443
[... 459 lines cut ...]
nextcloudpi    | 2022-09-23 15:34:19 0 [Note] InnoDB: Buffer pool(s) dump completed at 220923 15:34:19
nextcloudpi    | 2022-09-23 15:34:19 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
nextcloudpi    | 2022-09-23 15:34:19 0 [Note] InnoDB: Shutdown completed; log sequence number 1227102; transaction id 2315
nextcloudpi    | 2022-09-23 15:34:19 0 [Note] mysqld: Shutdown complete
nextcloudpi    | 
nextcloudpi    | Stopping MariaDB database server: mariadbd.
nextcloudpi    | Starting MariaDB database server: mariadbd.
nextcloudpi    | Stopping redis-server: redis-server.
nextcloudpi    | Starting redis-server: redis-server.
nextcloudpi    | Starting notify_push daemon
nextcloudpi    | Configuring Domain
nextcloudpi    | System config value trusted_domains => 11 set to string XX.XXX.XXX.XX
nextcloudpi    | System config value trusted_domains => 1 set to string 172.27.0.2
nextcloudpi    | System config value trusted_domains => 14 set to string 09d3b333b2ec
nextcloudpi    | System config value trusted_domains => 6 set to string 192.168.178.15
nextcloudpi    | Nextcloud version 24.0.3.2. NextCloudPi version v1.49.1
nextcloudpi    | Init done
```

* Successfully created a new container again

* Current conclusion
By running the upgrade internally within the Docker container, you can no longer stop the container and start it again as it pulls from the image in the docker registry, which still uses the previous version, and so this creates the current conflict you can see in the workflow I've outlined with the tests. 

So my take is, that you should take down the previous container, remove the old image and pull the new one once it has been properly updated and pushed to the Docker registry.

ZendaiOwl, 2022-09-23:
Since I had a restored version with a working docker-compose file again I attempted the `ncp-update` & `ncp-update-nc 24.0.4` one last time to see what happens. 

* Issues
* The backup function creates a backup containing a typo: `backup /var/www//nextcloud-bkp_20220923_1663943738.tar.gz generated` - Notice the 2nd "//" in the path OR it is only the debug message in the terminal that shows the 2nd "/" since later at the end of the upgrade logs it is properly formatted with the path `Backup stored at /data/nextcloud/data/ncp-update-backups/nextcloud-bkp_20220923_1663943738-24.0.3.2.tar.gz`

<details>
<summary>Complete upgrade log of ncp-update & ncp-update-nc 24.0.4</summary>

```bash
└࿓❯ sudo docker exec -it nextcloudpi bash
root@09d3b333b2ec:/# php -v
PHP 7.4.30 (cli) (built: Jul  7 2022 15:51:43) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies
    with Zend OPcache v7.4.30, Copyright (c), by Zend Technologies
root@09d3b333b2ec:/# ncp-update
Downloading updates
Performing updates
Clearing opcache...
Done.
mkdir: cannot create directory '/run/php': File exists
NextCloudPi updated to version v1.50.0
root@09d3b333b2ec:/# nc-update 24.0.4
bash: nc-update: command not found
root@09d3b333b2ec:/# nc-update       
bash: nc-update: command not found
root@09d3b333b2ec:/# nc-update-nc 24.0.4
bash: nc-update-nc: command not found
root@09d3b333b2ec:/# ncp-update-nc 24.0.4
Current   Nextcloud version 24.0.3.2
Available Nextcloud version 24.0.4
Download Nextcloud 24.0.4...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20220923_1663943738.tar.gz generated
Maintenance mode disabled
Install Nextcloud 24.0.4...
Restore data...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: calendar
Disabled incompatible app: contacts
Disabled incompatible app: news
Disabled incompatible app: notes
Disabled incompatible app: notify_push
Disabled incompatible app: tasks
Updating <circles> ...
Updated <circles> to 24.0.1
Starting code integrity check...
Finished code integrity check
Update successful
Turned off maintenance mode
Resetting log level
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
[... 342 lines cut ...]
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"24.0.4.1","data":[]}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Check encryption key format","userAgent":"--","version":"24.0.4.1","data":{"app":"updater"}}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"24.0.4.1","data":[]}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove old dashboard app config data","userAgent":"--","version":"24.0.4.1","data":{"app":"updater"}}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"24.0.4.1","data":[]}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:00+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"24.0.4.1","data":{"app":"updater"}}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:01+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"24.0.4.1","data":[]}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:01+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"24.0.4.1","data":{"app":"updater"}}
{"reqId":"tZniDydXfo8HZeXEhsRR","level":1,"time":"2022-09-23T16:45:01+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispa [... cut ...]
### Restarting the container with the latest docker-compose I wrote in my comment above does work, even if it rolls back to the previous version within the docker registry, the actual files and programs that have been upgraded are still upgraded. As evidenced by Nextcloud asking you to run `occ upgrade` when the container starts up again. 

After running this upgrade I have access to the Nextcloud website still, meaning that restarting the container does work after an internal upgrade, however as I said in my earlier comment above, it will restore the container to the previous configuration of the image in the docker registry again.

facorazza, 2022-09-23:
> So my take is, that you should take down the previous container, remove the old image and pull the new one once it has been properly updated and pushed to the Docker registry OR you can update it internally, but you cannot take it down or restart the container at all if this is done, because it will roll it back to the previous configuration of the image in the docker registry

That's always been the case. With docker-compose you can stop the container and preserve the version, but if you run docker compose down and are using an older image it will go back to that image's version.

ZendaiOwl, 2022-09-23:
@facorazza Awesome, thanks for confirmation :D there was a discussion about it in the Matrix chat and I didn't know for sure, which is why I started testing it out with the current upgrade issue  ^^

sunjam, 2022-09-23:
I tried the docker-compose file recommended (docker-compose.yml - 1st version) and I noticed the php 8.1 upgrade isn't performed doesn't happen when updating to 1.50 and then updating NextcloudPi to 24.0.5 from `ncp-config` on current image.

ZendaiOwl, 2022-09-24:
Try running `ncp-update-nc $NCVER` if you want to update Nextcloud within your container. You shouldn't have automatic updates turned on unless you're pushing your image to docker hub which you can then pull from when you're starting the container. Docker updates by pulling the new image, stopping the old and starting the new

https://help.nextcloud.com/t/how-to-update-nextcloudpi-docker-container/29282/2?u=zendaiowl

ZendaiOwl, 2022-09-26:
Just to add if someone else comes by reading, if you upgrade PHP inside docker and the image isn't upgraded on the docker hub, you will revert back to 7.4 when pulling the image again during a container restart and end up with a broken container configuration.

theCalcaholic, 2022-09-26:
> Just to add if someone else comes by reading, if you upgrade PHP inside docker and the image isn't upgraded on the docker hub, you will revert back to 7.4 when pulling the image again during a container restart and end up with a broken container configuration.

The underlying issue is exactly that. It shouldn't be possible to use `ncp-update` to update on docker as it might lead to undefined behavior. This will be addressed in the upcoming v1.50.1 release

theCalcaholic, 2022-09-26:
Related: #1574

ZendaiOwl, 2022-09-26:
> The underlying issue is exactly that. It shouldn't be possible to use ncp-update to update on docker as it might lead to undefined behavior. 

Yes, this is the recommended way and also the best I think, for users to not upgrade docker internally at all

One possible solution, for users looking to automate the docker upgrade, would be to make a Cron script that checks .. I don't know, once a day? Once a month? And when triggered runs the upgrade commands in order for the user, perhaps at a certain time at night maybe? 🙏

theCalcaholic, 2022-09-29:
This has now been fixed by refusing to update containers using ncp-update (without explicit override)

closed 2022-09-29 by theCalcaholic