nextcloud/nextcloudpi #1570: ncp-dist-upgrade: Debian 11 Bullseye packages don't include PHP 8.1 yet

mwip, 2022-09-25:
Using `ncp-dist-upgrade` to update from Debian 10 buster to Debian 11 bullseye, I stumbled across a problem with PHP. As read from `/usr/local/etc/ncp-recommended.cfg`, the suggested updates would aim to install PHP 8.1. Yet, from the [bullseye package repos](https://packages.debian.org/search?suite=bullseye&searchon=names&keywords=php), we can see, that Debian currently ships PHP 7.4 for bullseye.

Manually setting `ncp-recommended` (a.k.a. `new_cfg`) to point to PHP 7.4 results in a borked update. The OS works fine, but NC(P) does not.

**System information (before upgrade)**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.0
OS                   Debian GNU/Linux 10. 5.14.0-odroid-arm64 (aarch64)
automount            no
USB devices          none
datadir              /media/icy/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      2.0T/3.6T
rootfs usage         13G/55G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    23.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              192.168.2.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               4min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "2003:c7:8712:cc00:21e:6ff:fe49:38d",
            "1": "192.168.2.11",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "20": "cloud.m-weigand.de",
            "21": "hc4",
            "22": "192.168.2.11",
            "3": "localhost",
            "": "localhost",
            "14": "hc4.hc4"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
[... 98 lines cut ...]
```

{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item \u2018A form of colonialism\u2019: Activists demand climate reparations for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664096466","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Palestinian comic fans denounce Marvel\u2019s \u2018Sabra\u2019 as propaganda for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664093523","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Zelenskyy \u2018shocked\u2019 by Israel\u2019s failure to give Ukraine weapons for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664092091","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Typhoon Noru \u2018intensifying explosively\u2019 as it hits Philippines for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664091003","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Criticism grows louder over Russia\u2019s chaotic troop mobilisation for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664090965","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Italy election: Will far-right prevail for first time since WII? for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664083176","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Iran pledges \u2018decisive action\u2019 as Mahsa Amini protests continue for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664081485","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Russia\u2019s Lavrov vows \u2018full protection\u2019 for any annexed territory for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664077987","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item N Korea fires ballistic missile ahead of US-S Korea drills for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664065893","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Russians defy Putin\u2019s mobilisation push, hundreds arrested for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664054085","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Added item Could Lebanon\u2019s economic collapse create new humanitarian crisis? for feed Al Jazeera \u2013 Breaking News, World News and Video from Al Jazeera lastmodified: 1664050706","userAgent":"--","version":"23.0.6.1"}
{"reqId":"kxGk7NOglgIFwl9cfPcL","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"extract","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"read access : https://dwaves.de/feed/ into a feed instance (feed class : FeedIo\\Feed)","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"start reading https://dwaves.de/feed/","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":1,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"no 'modifiedSince' parameter given, setting it to 01/01/1970","userAgent":"--","version":"23.0.6.1"}
{"reqId":"mWORiq29aSIJdh6Nvh03","level":1,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"hitting https://dwaves.de/feed/","userAgent":"--","version":"23.0.6.1"}
{"reqId":"kxGk7NOglgIFwl9cfPcL","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Json","userAgent":"--","version":"23.0.6.1"}
{"reqId":"kxGk7NOglgIFwl9cfPcL","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Atom","userAgent":"--","version":"23.0.6.1"}
{"reqId":"kxGk7NOglgIFwl9cfPcL","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Rss","userAgent":"--","version":"23.0.6.1"}
{"reqId":"kxGk7NOglgIFwl9cfPcL","level":0,"time":"2022-09-25T14:16:19+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Rdf","userAgent":"--","version":"23.0.6.1"}

```
</details>

DKX47, 2022-09-27:
Guess I hit that same bug. This morning the automated update (to NCP Version v1.50.0) failed installing php8.1, Rollback failed, all my files seemed lost. That was an exiting Morning :)
NCP / NC runs now with php7.3, I guess. I have my files in backup but don't know what to do, to get the php thingy back on track...

```
Di 27. Sep 06:25:02 CEST 2022 - Running /etc/cron.daily/ncp-autoupdate...
Downloading updates
Performing updates
Clearing opcache...
Done.
Running nc-autoupdate-nc
[ nc-autoupdate-nc ] (Di 27. Sep 06:26:35 CEST 2022)
automatic Nextcloud updates enabled
automatic Nextcloud updates enabled

Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade
NextCloudPi updated to version v1.50.0
[ncp-update-nc]
Current   Nextcloud version 24.0.3.2
Available Nextcloud version 24.0.5
Download Nextcloud 24.0.5...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20220927_1664252816.tar.gz generated
Maintenance mode disabled
Install Nextcloud 24.0.5...
Restore data...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: apporder
Disabled incompatible app: calendar
Disabled incompatible app: contacts
Disabled incompatible app: notes
Disabled incompatible app: notify_push
Disabled incompatible app: tasks
Updating <circles> ...
Updated <circles> to 24.0.1
Updating <notifications> ...
Updated <notifications> to 2.12.1
Starting code integrity check...
Finished code integrity check
Update successful
Turned off maintenance mode
Resetting log level
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Check indices of the oc_jobs table.
Check indices of the oc_direct_edit table.
Done.
Check columns of the comments table.
Done.
Check primary keys.
Done.
All tables already up to date!
[... 100 lines cut ...]
ERROR: Conf php7.3-fpm does not exist!
Failed to start php7.3-fpm.service: Unit php7.3-fpm.service not found.
PHP upgrade has been successfully reverted
Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20220927_1664252816-24.0.3.2.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20220927_1664252816-24.0.3.2.tar.gz...
restore files...
restore database...
No datadir found in backup
Maintenance mode disabled
Starting scan for user 1 out of 3 (xxx
Starting scan for user 2 out of 3 (xxx)
Starting scan for user 3 out of 3 (xxx)
+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 13      | 0     | 00:00:01     |
+---------+-------+--------------+
sed: can't read /etc/php/7.3/mods-available/opcache.ini: No such file or directory
Cleanup...
Rollback failed! Data left at /var/www/ncp-data.YX5vqX
 
```

ZendaiOwl, 2022-09-27:
Well, one of the first things you need to do is upgrade to Bullseye I'd say. 

If you have backups, of both your settings in the backup tar and your user files, I'd say just reinstall it fresh with Bullseye and restore it from there. 

Restoring the backup tar first, then moving your files back in and scanning for external files with `ncc` or `occ` command, see ncc/occ list for which one 🙂

DKX47, 2022-09-27:
Maybe I'm reading the log wrong but isn't the installer trying to fetch php8.1 from the bullseye repos without success?

theCalcaholic, 2022-09-28:
Fix is now in master

closed 2022-09-28 by theCalcaholic

DKX47, 2022-10-01:
@theCalcaholic 
On 1.50.2 I still get the message "Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade" after login. When executing that command I still get some "error" messages:

```Update to bullseye available. Type '>>> ATTENTION <<<
This is a dangerous process that is only guaranteed to work properly if you
have not made manual changes in the system. Backup the SD card first and
proceed at your own risk.

Note that this is not a requirement for NCP to continue working properly.
The current distribution will keep receiving updates for some time.

Do you want to continue? [y/N]
y
Maintenance mode enabled
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done                 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'php7.3-tokenizer' for regex 'php7.3'
Note, selecting 'php7.3-calendar' for regex 'php7.3'
Note, selecting 'php7.3-fileinfo' for regex 'php7.3'
Note, selecting 'php7.3-readline' for regex 'php7.3'
Note, selecting 'php7.3-ftp' for regex 'php7.3'
Note, selecting 'php7.3-ctype' for regex 'php7.3'
Note, selecting 'php7.3-common' for regex 'php7.3'
Note, selecting 'php7.3-iconv' for regex 'php7.3'
Note, selecting 'php7.3-phar' for regex 'php7.3'
Note, selecting 'php7.3-pdo' for regex 'php7.3'
Note, selecting 'php7.3-exif' for regex 'php7.3'
Note, selecting 'php7.3-gettext' for regex 'php7.3'
Note, selecting 'php7.3-shmop' for regex 'php7.3'
Note, selecting 'php7.3-sockets' for regex 'php7.3'
Note, selecting 'php7.3-sysvmsg' for regex 'php7.3'
Note, selecting 'php7.3-sysvsem' for regex 'php7.3'
Note, selecting 'php7.3-sysvshm' for regex 'php7.3'
Note, selecting 'php7.3-posix' for regex 'php7.3'
Virtual packages like 'php7.3-fileinfo' can't be removed
E: Unable to locate package php7.3-curl
E: Couldn't find any package by glob 'php7.3-curl'
E: Couldn't find any package by regex 'php7.3-curl'
E: Unable to locate package php7.3-gd
E: Couldn't find any package by glob 'php7.3-gd'
E: Couldn't find any package by regex 'php7.3-gd'
E: Unable to locate package php7.3-fpm
E: Couldn't find any package by glob 'php7.3-fpm'
E: Couldn't find any package by regex 'php7.3-fpm'
E: Unable to locate package php7.3-cli
E: Couldn't find any package by glob 'php7.3-cli'
E: Couldn't find any package by regex 'php7.3-cli'
E: Unable to locate package php7.3-opcache
E: Couldn't find any package by glob 'php7.3-opcache'
E: Couldn't find any package by regex 'php7.3-opcache'
E: Unable to locate package php7.3-mbstring
E: Couldn't find any package by glob 'php7.3-mbstring'
E: Couldn't find any package by regex 'php7.3-mbstring'
E: Unable to locate package php7.3-xml
E: Couldn't find any package by glob 'php7.3-xml'
E: Couldn't find any package by regex 'php7.3-xml'
E: Unable to locate package php7.3-zip
E: Couldn't find any package by glob 'php7.3-zip'
E: Couldn't find any package by regex 'php7.3-zip'
E: Unable to locate package php7.3-ldap
[... 49 lines cut ...]
Reading package lists... Done                 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'php7.4-common' instead of 'php7.4-fileinfo'
php7.4 is already the newest version (7.4.30-1+deb11u1).
php7.4 set to manually installed.
Suggested packages:
  php-pear
The following NEW packages will be installed:
  php7.4-ldap
The following packages will be DOWNGRADED:
  php7.4-bz2 php7.4-cli php7.4-common php7.4-curl php7.4-fpm php7.4-gd
  php7.4-intl php7.4-json php7.4-mbstring php7.4-opcache php7.4-xml php7.4-zip
0 upgraded, 1 newly installed, 12 downgraded, 0 to remove and 54 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.
Something went wrong. Fix it and try again
```

Seems like the PHP Components have different versions. Should I use the "--allow-downgrades" to get back to 7.3?

theCalcaholic, 2022-10-01:
@DKX47 Since you are probably going to install PHP 8 afterwards, you can try --alow-downgrades without much risk, imo.

DKX47, 2022-10-08:
Just a quick note: The System was broken beyond repair, I ended up reinstalling. Thank you for your Support anyway :)