nextcloud/nextcloudpi #1575: Adding a "Warning" label

sunjam, 2022-09-26:
**Description**  
Adding a `Warning` label (Github version of tags) for issues known to cause breaking changes in NextcloudPi, so we can offer a perma-link which directs to those known issues an admin should be aware of if curious on potential issues with the updater, etc.

Thanks either way for considering.
Related issues #1574 #1570 etc.

theCalcaholic, 2022-09-26:
I like that idea!

REAPERSbattlecry, 2022-09-26:
I'm not sure if everyone is reading the github/forum posts before upgrading. As far as I can see most of the users are running their ncp-instance on autoupdate (many devices are still on buster).

Is it possible or maybe possible in advance to your idea to choose in the webUI between stable and testing update. And in testing the devel-branch gets used permanently. 
Maybe we find more people to do testing on different instances.

But despite all my critics I like that idea as well.

theCalcaholic, 2022-09-26:
@REAPERSbattlecry This is not meant as a tool for users, but for people who answer questions, like in the forums.
Makes it easier for me to communicate with people like @sunjam or @ZendaiOwl 

Regarding the idea to offer a "testing" branch in the web UI, I have to think about that. It's not exactly trivial to implement with the current update mechanism

theCalcaholic, 2022-09-26:
I have created the label: https://github.com/nextcloud/nextcloudpi/labels/disruption

closed 2022-09-26 by theCalcaholic

ZendaiOwl, 2022-09-26:
That's perfect 🥳