nextcloud/nextcloudpi #1577: docker: update to 1.50 and restart of the image removes all users but ncp from the db

svenb1234, 2022-09-27:
### PROBLEMS

After updating to 1.50 and restarting the docker image all users are gone from nextcloud. Not their data. But ncc user:list only lists ncp.

I will try to find out more details, but just in case somebody comes across the same issue, please add to this report.

Maybe related, when I go to my nextcloud server I see:
[image]

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
NextCloudPi version  v1.50.0
NextCloudPi image    NextCloudPi_docker_09-14-22
OS                   Debian GNU/Linux 11. 5.14.21-150400.24.21-default (x86_64)
automount            no
USB devices          sdc 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      1.4T/3.6T
rootfs usage         17G/30G
swapfile             /dev/md3
dbdir                /data-ro/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              172.17.0.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "***REMOVED SENSITIVE VALUE***",
[... 93 lines cut ...]
```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"YzKEM6bwiLY7z7isARAeGwAADgE","level":3,"time":"2022-09-27T05:03:47+00:00","remoteAddr":"93.132.152.129","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/calendars/sven/persnlich/","message":"WRONGPASS invalid username-password pair","userAgent":"DAVx5/4.2.3.2-ose (2022/08/31; dav4jvm; okhttp/4.10.0) Android/7.1.2","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"WRONGPASS invalid username-password pair","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":143,"function":"auth","class":"Redis","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":2082,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":122,"function":"getLockingProvider","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":454,"function":"__construct","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1445,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":602,"function":"boot","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":1111,"function":"init","class":"OC","type":"::"},{"file":"/data/nextcloud/remote.php","line":114,"args":["/data/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/RedisFactory.php","Line":143,"CustomMessage":"--"}}
{"reqId":"YzKEM6bwiLY7z7isARAeGwAADgE","level":3,"time":"2022-09-27T05:03:47+00:00","remoteAddr":"93.132.152.129","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/calendars/sven/persnlich/","message":"NOAUTH Authentication required.","userAgent":"DAVx5/4.2.3.2-ose (2022/08/31; dav4jvm; okhttp/4.10.0) Android/7.1.2","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"NOAUTH Authentication required.","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/data/nextcloud/lib/private/App/InfoParser.php","line":58,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/App/AppManager.php","line":506,"function":"parse","class":"OC\\App\\InfoParser","type":"->"},{"file":"/data/nextcloud/lib/private/App/AppManager.php","line":521,"function":"getAppInfo","class":"OC\\App\\AppManager","type":"->"},{"file":"/data/nextcloud/lib/private/legacy/OC_App.php","line":568,"function":"getAppVersion","class":"OC\\App\\AppManager","type":"->"},{"file":"/data/nextcloud/lib/private/legacy/OC_App.php","line":891,"function":"getAppVersion","class":"OC_App","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Util.php","line":1225,"function":"shouldUpgrade","class":"OC_App","type":"::"},{"file":"/data/nextcloud/lib/public/Util.php","line":579,"function":"needUpgrade","class":"OC_Util","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":116,"function":"needUpgrade","class":"OCP\\Util","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":78,"function":"initTemplateEngine","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":324,"function":"__construct","class":"OC_Template","type":"->"},{"file":"/data/nextcloud/remote.php","line":86,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/remote.php","line":168,"function":"handleException"}],"File":"/data/nextcloud/lib/private/Memcache/Redis.php","Line":48,"CustomMessage":"--"}}
[... cut ...] function":"initTemplateEngine","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":324,"function":"__construct","class":"OC_Template","type":"->"},{"file":"/data/nextcloud/remote.php","line":86,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/remote.php","line":168,"function":"handleException"}],"File":"/data/nextcloud/lib/private/Memcache/Redis.php","Line":48,"CustomMessage":"--"}}

```
</details>

svenb1234, 2022-09-27:
Comment was too long:

{"reqId":"YzKEl40M96rgjHR8DlHppgAASQA","level":3,"time":"2022-09-27T05:05:27+00:00","remoteAddr":"93.132.152.129","user":"--","app":"index","method":"POST","url":"/index.php/apps/notify_push/pre_auth","message":"No such file or directory","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"No such file or directory","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":137,"function":"pconnect","class":"Redis","type":"->"},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":2082,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Files/View.php","line":122,"function":"getLockingProvider","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":454,"function":"__construct","class":"OC\\Files\\View","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1445,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":602,"function":"boot","class":"OC\\Server","type":"->"},{"file":"/data/nextcloud/lib/base.php","line":1111,"function":"init","class":"OC","type":"::"},{"file":"/data/nextcloud/index.php","line":34,"args":["/data/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/RedisFactory.php","Line":137,"CustomMessage":"--"}}
{"reqId":"YzKEl40M96rgjHR8DlHppgAASQA","level":3,"time":"2022-09-27T05:05:27+00:00","remoteAddr":"93.132.152.129","user":"--","app":"core","method":"POST","url":"/index.php/apps/notify_push/pre_auth","message":"No such file or directory","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"No such file or directory","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":137,"function":"pconnect","class":"Redis","type":"->"},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContain [... cut ...] CP\\Util","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":78,"function":"initTemplateEngine","class":"OC_Template","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Template.php","line":324,"function":"__construct","class":"OC_Template","type":"->"},{"file":"/data/nextcloud/index.php","line":74,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"}],"File":"/data/nextcloud/lib/private/Memcache/Redis.php","Line":48,"CustomMessage":"--"}}

BstAA, 2022-09-27:
Some problem with me.  Additional problem see new issue.

theCalcaholic, 2022-09-27:
This seems to be caused by an incomplete update. Please restore the backup that was created during nc-update (should be in `/path/to/your/data/directory/ncp-update-backups`) with `sudo ncp-restore /path/to/backup.tar.gz`

theCalcaholic, 2022-09-27:
`ncp-update` should not be used to update docker instances of NCP. Unfortunately this was not prevented so far (but will be from version v1.50.1 on).

See also https://github.com/nextcloud/nextcloudpi/issues/1568

To fix this, please restart the container and restore the latest backup (a backup should have been created during the latest nc update, which you can find in `/data/nextcloud/data/ncp-update-backups`).

svenb1234, 2022-09-27:
> To fix this, please restart the container and restore the latest backup (a backup should have been created during the latest nc update, which you can find in `/data/nextcloud/data/ncp-update-backups`).

I do not have that directory. Only the daily backup configured in the web panel.

Tommy-01, 2022-09-28:
> `ncp-update` should not be used to update docker instances of NCP. Unfortunately this was not prevented so far (but will be from version v1.50.1 on).
>
> See also #1568
>
> To fix this, please restart the [...]

Hi, can you please tell how to do the restore? Or is the there an  how-to? Until now i just find guides that use the ncp configuration site. But i cant use this, because i cant login after the update.   
Thank you!

svenb1234, 2022-09-28:
In the docker console of your container 
`ncp-restore /data/nextcloud/data/ncp-update-backups/backup file xy`

However, in my container that path does not exist.

svenb1234, 2022-09-28:
Sorry for closing by mistake.

theCalcaholic, 2022-09-28:
I finally found the cause for this. Unfortunately, the old database is gone. You will need to restore a backup after following these steps:

```sh
sed -i 's|/data-ro|/data|' /etc/mysql/mariadb.conf.d/90-ncp.cnf
chown -R mysql: /data/database
service mariadb restart
```

Now restore a backup from the web panel. A dataless backup is enough (files should not have been lost)

ZendaiOwl, 2022-09-28:
Adding this bit of information here as well, since it is related

I found another issue on v1.50.1 using the `:latest` tag, wrote to you in Matrix chat

The :latest tag is only built using the amd64 platform tag, use the version itself when pulling from the docker hub to get the correct architecture

Should not be too much trouble to correct however

theCalcaholic, 2022-09-28:
@ZendaiOwl That was actually the v1.49.1 image, not the v1.50.1 one, but yeah - it's corrected now

ZendaiOwl, 2022-09-28:
Oh? The tag ID of the image did say it was 1.50.1 at the time, anyway, great that it's corrected :pray: Thank you

theCalcaholic, 2022-09-29:
Should finally be fixed in the pending v1.50.2 release (build is running)

closed 2022-09-29 by theCalcaholic