nextcloud/nextcloudpi #1580: docker: error on msysql config after pulling new 1.50 image (can't connect to MySQL server)

BstAA, 2022-09-27:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.49.1
NextCloudPi image    NextCloudPi_docker_09-14-22
OS                   Debian GNU/Linux 11. 5.4.203-1-pve (x86_64)
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           no
data filesystem      zfs
data disk usage      47G/3.4T
rootfs usage         13G/395G
swapfile             /dev/md1
dbdir                 (doesn't exist)
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
Gateway              172.18.0.1
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               4:25

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /data/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /data/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /data/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /data/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(281): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data/nextcloud/lib/private/AppConfig.php(411): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data/nextcloud/lib/private/AppConfig.php(176): OC\AppConfig->loadConfigValues()
#7 /data/nextcloud/lib/private/AppConfig.php(367): OC\AppConfig->getApps()
#8 /data/nextcloud/lib/private/legacy/OC_App.php(983): OC\AppConfig->getValues()
#9 /data/nextcloud/lib/private/Server.php(713): OC_App::getAppVersions()
#10 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#11 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#13 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#15 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(184): OC\AppFramework\Utility\SimpleContainer->get()
#16 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#17 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#19 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#20 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
[... 63 lines cut ...]
{"reqId":"J0giNO9lut8VzjsvDosc","level":1,"time":"2022-09-09T12:50:30+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"23.0.6.1","data":{"app":"updater"}}
{"reqId":"J0giNO9lut8VzjsvDosc","level":1,"time":"2022-09-09T12:50:34+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"23.0.6.1","data":{"app":"updater"}}
{"reqId":"J0giNO9lut8VzjsvDosc","level":1,"time":"2022-09-09T12:50:34+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"24.0.3.2","data":{"app":"updater"}}
{"reqId":"J0giNO9lut8VzjsvDosc","level":1,"time":"2022-09-09T12:50:34+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"24.0.3.2","data":{"app":"updater"}}
{"reqId":"J0giNO9lut8VzjsvDosc","level":1,"time":"2022-09-09T12:50:34+02:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"24.0.3.2","data":{"app":"updater"}}
{"reqId":"YzKV6vgIpTZx2EesDkuWwQAASgE","level":2,"time":"2022-09-27T08:19:23+02:00","remoteAddr":"192.168.10.58","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/files/Klaus/","message":"Login failed: 'klaus' (Remote IP: '192.168.10.58')","userAgent":"Mozilla/5.0 (Windows) mirall/3.6.0stable-Win64 (build 20220906) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"YzKV6_gIpTZx2EesDkuWygAAVAE","level":2,"time":"2022-09-27T08:19:23+02:00","remoteAddr":"192.168.10.58","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/files/Klaus/","message":"Login failed: 'klaus' (Remote IP: '192.168.10.58')","userAgent":"Mozilla/5.0 (Windows) mirall/3.6.0stable-Win64 (build 20220906) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.3.2","data":{"app":"core"}}
{"reqId":"YzLNHvgIpTZx2EesDkuWzQAASAE","level":2,"time":"2022-09-27T12:14:54+02:00","remoteAddr":"192.168.10.58","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: klaus (Remote IP: 192.168.10.58)","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.3.2","data":[]}
{"reqId":"YzLNR_gIpTZx2EesDkuW0AAATQE","level":2,"time":"2022-09-27T12:15:36+02:00","remoteAddr":"192.168.10.58","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: ncp (Remote IP: 192.168.10.58)","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.3.2","data":[]}
{"reqId":"YzLNXS_QPehc6yRHeXN_AQAAFgc","level":2,"time":"2022-09-27T12:15:59+02:00","remoteAddr":"192.168.10.58","user":"--","app":"no app in context","method":"POST","url":"/index.php/login","message":"Login failed: ncp (Remote IP: 192.168.10.58)","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.3.2","data":[]}

```
</details>

**Docker Log**
```
Starting PHP-fpm
Starting Apache
bash: /usr/local/etc/ncp-templates/mysql/90-ncp.cnf.sh: No such file or directory
/usr/local/etc/library.sh: line 220: ${3:}: bad substitution
Starting Redis
Starting Cron
Starting Postfix
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)
```

theCalcaholic, 2022-09-27:
Hi!

There is no new 1.50 image - how have you been pulling it?

BstAA, 2022-09-27:
> Hi!
> 
> There is no new 1.50 image - how have you been pulling it?

1.  `docker-compose pull / docker-compse up -d`  resulted in "lost users" see other issue
2. error came up after purging container as well as image and new start with `docker compose up -d `
3. additional information: The referred directory `/usr/local/etc/ncp-templates/mysql` is missing

theCalcaholic, 2022-09-27:
Did you previously attempt to upgrade ncp using `ncp-update`?

BstAA, 2022-09-27:
> Did you previously attempt to upgrade ncp using `ncp-update`?
> 
Yes. Not this time but some time ago.

Howto get back to a running system?

See #1577 "To fix this, please restart the container and restore the latest backup (a backup should have been created during the latest nc update, which you can find in /data/nextcloud/data/ncp-update-backups)." doesn't work. 

Should I start with a new, empty "data" directory (`i.e. including ./bin, ./database, ...`) and than lateron restore latest backup and data?
I've got 
```
- nextcloud-bkp_20220402_1648931186-22.2.2.0.tar.gz
- nextcloud-bkp_20220726_1658838851-23.0.2.1.tar.gz
- nextcloud-bkp_20220909_1662720410-23.0.6.1.tar.gz
```

ZendaiOwl, 2022-09-27:
@BstAA 
I'm currently working at the moment, I had time off previously to catch up in school and now I'm back at work (whole still studying at the same time) so I don't have very much time over at the moment, jsyk 🙏

Could you please reply with the steps you took during your attempt to restore your instance, which commands you ran, in what order, please? 🙏

I think you need to remove your image and volume as well, taking out your backup and user data beforehand, to successfully restore a working instance without the layer cached from the previous image that was upgraded

When it is up running, you can restore the settings with the backup tar, then move your files back, finally scan them with the `ncc` or `occ` command

Could you try and see if that works?

BstAA, 2022-09-27:
@ZendaiOwl 
Thanks for your advice! My steps to restore a running system:
1, Stop container and remove the running container 
2. Remove unused ncp image
3. Create a new host directory for the data volume
4. modify docker-compose.yml to use the new data volume
5. `docker-compose up -d`
6. initial startup etc. / trusted domains ...
7. Result a running system with user ncp System Info:
- NextCloudPi version 1.49.1
- Nextcloud version 24.0.3.2
8. create backup via Web-Panel
9. try to restore `nextcloud-bkp_20220909_1662720410-23.0.6.1.tar.gz` --> seems to work including scan for users but
10. Final result "Somthing when wrong. Try refreshing this page"
11. System Info now gives (as expected) Nextcloud Version 23.0.6.1 but HPB service down
12. After restart everything seams to be o.k.

Moved User files to new volume, Files scan for `--all` with `occ` 

System is up an running ... but with ncp 1.49.1 and nc 23.0.6.1

So advice is?
1. Never update ncp Docker via ncp-config / ncp-web
2. But update nc via ncp-config / ncp-web

theCalcaholic, 2022-09-27:
Glad to hear that you could successfully restore your instance!

>So advice is?
>
>    Never update ncp Docker via ncp-config / ncp-web
>    But update nc via ncp-config / ncp-web

Yes, basically. Starting from version 1.50.1, it isn't possible anymore to update the docker container using ncp-config/ncp-update to prevent this kind of issue

ZendaiOwl, 2022-09-28:
@BstAA 
> So advice is?
> 
> Never update ncp Docker via ncp-config / ncp-web
> But update nc via ncp-config / ncp-web

theCalcaholic already answered this in his post above you :slightly_smiling_face:  

To elaborate a little bit however..

It is not recommended to upgrade your container internally like this, it quickly creates bloat in the layers of the container, making it larger slowly and filled with stuff that doesn't have to be there. This doesn't mean you can't do this, as evidenced by the upgrades have succeeded for some but you have to remember that you are using a docker image. Docker images are pulled from the Docker registry, on the Docker Hub, which means you have to use that repository as a reference when you decide to get or check for an upgraded image, using the ones that are available there. If you update or upgrade without pushing your changes to a repository on docker, so you can use that repository as a reference when you want to launch your container again, the upgrade will fail because the image you're pulling an image from to launch the container, does not have the changes you made in it. 

It is instead better to build a new image and push that one to the repository, then use that to upgrade, keeping it as small as possible. A Docker container is designed to do "one task" or have 1 "binary" in them, which when the image is launched to run as a container, this entrypoint process becomes PID 1 of your container. Of course this is just my current basic understanding and I really liked the way [this Google article][article] defined it you care for a read 

_From the article_
> Note: In the context of this best practice, an "app" is considered to be a single piece of software, with a unique parent process, and potentially several child processes.
> When you start working with containers, it's a common mistake to treat them as virtual machines that can run many different things simultaneously. A container can work this way, but doing so reduces most of the advantages of the container model. For example, take a classic Apache/MySQL/PHP stack: you might be tempted to run all the components in a single container. However, the best practice is to use two or three different containers: one for Apache, one for MySQL, and potentially one for PHP if you are running PHP-FPM.
> 
> Because a container is designed to have the same lifecycle as the app it hosts, each of your containers should contain only one app. When a container starts, so should the app, and when the app stops, so should the container. The following diagram shows this best practice.

[article]: https://cloud.google.com/architecture/best-practices-for-building-containers

ZendaiOwl, 2022-09-28:
Just a quick follow-up, I checked the hashesh of the images on the docker hub and the `:latest` should be correct, they have the same hash value

ZendaiOwl, 2022-09-28:
I'm re-opening this issue primarily to make it more visible, so users unfamiliar with GitHub have an easier time to locate it and the information above pray we can close it again once the docker issues are fully resolved and have calmed down

theCalcaholic, 2022-09-29:
> Should I start with a new, empty "data" directory (i.e. including ./bin, ./database, ...) and than lateron restore latest backup and data?

Yes, exactly. Move your backups to a safe location on the host, purge your docker volume (or mounted directory), start from a fresh container (ideally the same version of NCP from which the backup was created) and restore the backup

theCalcaholic, 2023-01-21:
Closing due to inactivity

closed 2023-01-21 by theCalcaholic