nextcloud/nextcloudpi #1589: Containerize

ZendaiOwl, 2022-09-27:
# Containerize NCP

_A project that started from a brainstorming in the Matrix Wiki chat room._  
First post & preliminary information and all that sort of stuff :pray: 

## Project idea 

Convert NCP and it's tool into containers and services capable of being integrated with others, also making it possible to update/upgrade parts of the whole instead  

### End goal
 
Containerize NCP completely

### Starting point & proof-of-concept

Convert `ncp-config`'s various scripts into individual containers & `ncp-config` to a container as well, being used as the master container, to control the others. 

- [ ] [NCP-Config][ncp-config]   _Check me when the conversion is complete._
  - [ ] Category re-design/re-structuring   _Check me when the category re-design is done._

TODO
  1. - [ ] Add links and script names to the list until completed
      1.  - [X] [BACKUPS][dirBackups]  _:heavy_check_mark: ZendaiOwl_
      2.  - [X] [CONFIG][dirConfig]  _:heavy_check_mark: ZendaiOwl_
      3.  - [ ] [NETWORKING][dirNetworking]
      4.  - [ ] [SECURITY][dirSecurity]
      5.  - [ ] [SYSTEM][dirSystem]
      6.  - [ ] [TOOLS][dirTools]
      7.  - [ ] [UPDATES][dirUpdates]
  2. - [ ] Expand explanations 
  3. - [ ] What else? ..

---

[ncp-config]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp-config

### Related Help articles & Documentation information

[Google - Best practice, Building containers][1]
[Google - Best practice, Operating containers][2]
[Docker - Best practice, Dockerfile][3]
[Docker - Best practice, Development][4]
[Docker - Build enhancements][5]
[Docker - Choosing a build driver][6]
[Docker - Manage images][7]
[Docker - Create a base image][8]

[1]: https://cloud.google.com/architecture/best-practices-for-building-containers#signal-handling
[2]: https://cloud.google.com/architecture/best-practices-for-operating-containers
[3]: https://docs.docker.com/develop/develop-images/dockerfile_best-practices/
[4]: https://docs.docker.com/develop/dev-best-practices/
[5]: https://docs.docker.com/develop/develop-images/build_enhancements/
[6]: https://docs.docker.com/build/building/drivers/
[7]: https://docs.docker.com/develop/develop-images/image_management/
[8]: https://docs.docker.com/develop/develop-images/baseimages/

---

#### [CATEGORIES][dirCategories]

* Conversion progress, check the box when the category has been completely converted to individual container images

1.  - [ ] [BACKUPS][dirBackups]
2.  - [ ] [CONFIG][dirConfig]
8.  - [ ] [NETWORKING][dirNetworking]
9.  - [ ] [SECURITY][dirSecurity]
10.  - [ ] [SYSTEM][dirSystem]
11.  - [ ] [TOOLS][dirTools]
12.  - [ ] [UPDATES][dirUpdates]

[dirCategories]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp
[dirBackups]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/BACKUPS
[dirConfig]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/CONFIG
[dirNetworking]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/NETWORKING
[dirSecurity]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/SECURITY
[dirSystem]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/SYSTEM
[dirTools]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/TOOLS
[dirUpdates]: https://github.com/nextcloud/nextcloudpi/tree/master/bin/ncp/UPDATES

---

#### [BACKUPS][dirBackups]
 
1. - [ ] [nc-backup-auto.sh][nc-backup-auto.sh]
2. - [ ] [nc-backup.sh][nc-backup.sh]
4. - [ ] [nc-export-ncp.sh][nc-export-ncp.sh]
5. - [ ] [nc-import-ncp.sh][nc-import-ncp.sh]
6. - [ ] [nc-restore-snapshot.sh][nc-restore-snapshot.sh]
7. - [ ] [nc-restore.sh][nc-restore.sh]
8. - [ ] [nc-rsync-auto.sh][nc-rsync-auto.sh]
13. - [ ] [nc-rsync.sh][nc-rsync.sh]
14. - [ ] [nc-snapshot-auto.sh][nc-snapshot-auto.sh]
15. - [ ] [nc-snapshot-sync.sh][nc-snapshot-sync.sh]
16. - [ ] [nc-snapshot.sh][nc-snapshot.sh]

[nc-backup-auto.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup-auto.sh
[nc-backup.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup.sh
[nc-export-ncp.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-export-ncp.sh
[nc-import-ncp.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-import-ncp.sh
[nc-restore-snapshot.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-restore-snapshot.sh
[nc-restore.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-restore.sh
[nc-rsync-auto.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-rsync-auto.sh
[nc-rsync.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-rsync.sh
[nc-snapshot-auto.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-snapshot-auto.sh
[nc-snapshot-sync.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-snapshot-sync.sh
[nc-snapshot.sh]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-snapshot.sh

#### [CONFIG][dirConfig]

- [ ] 1. [nc-admin.sh][nc-admin]
- [ ] 2. [nc-database.sh][nc-database]
- [ ] 3. [nc-datadir.sh][nc-datadir]
- [ ] 4. [nc-httpsonly.sh][nc-httpsonly]
- [ ] 5. [nc-init.sh][nc-init]
- [ ] 6. [nc-limits.sh][nc-limits]
- [ ] 7. [nc-nextcloud.sh][nc-nextcloud]
- [ ] 8. [nc-passwd.sh][nc-passwd]
- [ ] 9. [nc-prettyURL.sh][nc-prettyURL]
- [ ] 10. [nc-previews-auto.sh][nc-previews-auto]
- [ ] 11. [nc-scan-auto.sh][nc-scan-auto]
- [ ] 12. [nc-trusted-domains.sh][nc-trusted-domains]
- [ ] 13. [nc-webui.sh][nc-webui]

[nc-admin]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-admin.sh
[nc-database]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-database.sh
[nc-datadir]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-datadir.sh
[nc-httpsonly]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-httpsonly.sh
[nc-init]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-init.sh
[nc-limits]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-limits.sh
[nc-nextcloud]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-nextcloud.sh
[nc-passwd]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-passwd.sh
[nc-prettyURL]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-prettyURL.sh
[nc-previews-auto]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-previews-auto.sh
[nc-scan-auto]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-scan-auto.sh
[nc-trusted-domains]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-trusted-domains.sh
[nc-webui]: https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/CONFIG/nc-webui.sh

#### [NETWORKING][dirNetworking]

- [ ] 1. 
- [ ] 2. 
- [ ] 3. 
- [ ] 4. 
- [ ] 5. 
- [ ] 6. 
- [ ] 7. 
- [ ] 8. 
- [ ] 9. 
- [ ] 10. 
- [ ] 11. 
- [ ] 12. 

#### [SECURITY][dirSecurity]

- [ ] 1. 
- [ ] 2. 
- [ ] 3. 
- [ ] 4. 
- [ ] 5. 
- [ ] 6. 
- [ ] 7. 
- [ ] 8. 
- [ ] 9. 
- [ ] 10. 
- [ ] 11. 
- [ ] 12. 

#### [SYSTEM][dirSystem]

- [ ] 1. 
- [ ] 2. 
- [ ] 3. 
- [ ] 4. 
- [ ] 5. 
- [ ] 6. 
- [ ] 7. 
- [ ] 8. 
- [ ] 9. 
- [ ] 10. 
- [ ] 11. 
- [ ] 12. 

#### [TOOLS][dirTools]

- [ ] 1. 
- [ ] 2. 
- [ ] 3. 
- [ ] 4. 
- [ ] 5. 
- [ ] 6. 
- [ ] 7. 
- [ ] 8. 
- [ ] 9. 
- [ ] 10. 
- [ ] 11. 
- [ ] 12. 

#### [UPDATES][dirUpdates]

- [ ] 1. 
- [ ] 2. 
- [ ] 3. 
- [ ] 4. 
- [ ] 5. 
- [ ] 6. 
- [ ] 7. 
- [ ] 8. 
- [ ] 9. 
- [ ] 10. 
- [ ] 11. 
- [ ] 12.

sunjam, 2022-09-27:
Do you want to also make a linked copy of this on the forum for wiki collab and further discussion?

ZendaiOwl, 2022-09-27:
Hmm ... I don't know :thinking: 

I don't mind if you want to link to it in the forum to start a discussion as such, I just hope it won't be regarded as a promise or anything like that :D it's still very early as I just started writing this up and I don't know how far it'll go, how long it'll take or even if it'll all work out in the end fully yet. But if that is understood (or those expectations are set, to say it in another way), then I don't mind at all, you can go ahead and link it in the forum :pray:

sunjam, 2022-09-27:
Great, in that case we'll discuss things on the forum if it makes sense.  Just thinking of ways to keep collaborating on it.  Thanks for writing this up!

ZendaiOwl, 2022-09-30:
Moving this to a readme instead within the branch

szaimen, 2022-10-02:
Hello, just had a look and the approach sounds interesting! 

A possible approach that I could imagine would be building Nextcloudpi in the future around AIO which would allow to split-up the maintenance work. You could simply run AIO in reverse proxy mode and implement further features than that NextlcoudPi currently offers, but AIO can't. Going the same path like AIO and introducing a mastercontainer that manages other docker containers would make NextcloudPi easier to maintain but would introduce some limits regarding what NextcloudPis mastercontainer can do so it should be carefully discussed.

Also as a sidenote: modularization using docker containers is a good approach for maintainability but creating ~50 docker containers that need to get mounted and run a specific way does not make it more maintainable.

theCalcaholic, 2022-10-02:
@szaimen I did not have time yet to do a write up on my ideas and criteria regarding this topic, but I agree - my vision encompasses just one container that runs ncp tasks and overwrites the entry point or command for the various scripts and one or two othwr containers for the web interface and maybe a cron-like service (we'll have to see what's a good approach here). And then of course we need the kind of services that AIO contains (or maybe just integrate with AIO). :) 

Some thoughts I have about this (a complete writeup will follow):

Generally, I want to explore how well we can abstract script environments that way to generalize them for testing.
E. g. a script can always operate on fixed paths if running and docker and selecting the correct part could be implemented by how the container is run - effectively making scripts more agnostic to the system environment.

A big challenge will be to provide our users with working upgrade paths (seemless upgrades from both, the current ncp docker container as well as bare metal installations to a multi-container setup) - I'm actually not sure yet, if it will be feasible at all, but I will at least explore that

szaimen, 2022-10-02:
> A big challenge will be to provide our users with working upgrade paths (seemless upgrades from both, the current ncp docker container as well as bare metal installations to a multi-container setup) - I'm actually not sure yet, if it will be feasible at all, but I will at least explore that

Just discussed this with @st3iny and came to the conclusion that this should indeed be carefully considered 👍

szaimen, 2022-10-02:
> And then of course we need the kind of services that AIO contains (or maybe just integrate with AIO). :)

Would be really cool if you would integrate AIO into NextcloudPi :)

theCalcaholic, 2022-10-02:
> > And then of course we need the kind of services that AIO contains (or maybe just integrate with AIO). :)
> 
> Would be really cool if you would integrate AIO into NextcloudPi :)

That would be my preferable option - I just don't know enough about AIO yet to understand all technical implications :)

szaimen, 2022-10-02:
> That would be my preferable option - I just don't know enough about AIO yet to understand all technical implications :)

Feel free to ask me anything about it ;)

ZendaiOwl, 2022-10-02:
@theCalcaholic @szaimen 

Thank you 🙏 I'm glad you think the idea is interesting and for expressing your interest 🥳

I had also come to similar thoughts/conclusions but haven't really written it down yet in the README ^^ 

To use one container, ncp-config is entrypoint, possibly install it as plugin or only `nc-encrypt` which needs admin permissions. 

Then combine that with nextcloud-aio, Nextcloud, PHP, mariaDB or a database and Caddy as front-end or reverse proxy, which is how I've used Caddy the most (reverse-proxy)

If possible, could I also ask you questions about docker when I have some, or as they come up when I try things out? 🙏 

I'm currently working right now, making it difficult to write a proper response on my phone 😅 but something like described above is about where I'm at in my thoughts right now

szaimen, 2022-10-03:
@theCalcaholic regarding version pinning, this would be easily possible if we add the build time to the mastercontainer and publish that to docker hub as tag like we already do with all the other containers. However this will break mastercontainer updates as watchtower is not able to update to a different docker tag so you would need to take care of that youself. E.g. by stopping the mastercontainer, removing it and recreating it with the new docker tag. Afterwards, you can use the `AUTOMATIC_UPDATES` flag of https://github.com/nextcloud/all-in-one#how-to-stopstartupdate-containers-or-trigger-the-daily-backup-from-a-script-externally to update all the other containers as well. Please tell me if this would be sufficient for you. I could then add this to the build script as well.

---

@ZendaiOwl AIO already includes the parts that are important for running a full-fletched Nextcloud instance so NextcloudPi should probably rather all additional options that are not already included in AIO should be added here.

theCalcaholic, 2022-10-03:
@szaimen Sounds good to me. That's basically how I imagined that it could work. 👍

ZendaiOwl, 2022-10-04:
I had a little thought, while I've been writing java apps for Android and reading about docker I also read your responses here which got me thinking..  

Since the focus then would be to mainly containerize the features of ncp perhaps implementing the ncp features within an API is the easiest way to get it working? I don't know, it seems that is the standard way for containers to communicate between each other within a docker network, it's also possible to use an alias for that within the containers, which I've tested out. 

Anyway .. I had a thought if it would be possible to use Java (maybe with Spring) or Go (or something else) to make that API framework and I tested it out now in a quick test, which was successful :partying_face:, so I wanted to share and see what you think? 

I created a very simple Java program which uses Java's Runtime.getRuntime().exec() command to run `ping -c 2 google.com`, which is similar to bash's `exec` and is the recommended execution command inside a container or `command` 

If it's built using the GraalVM, it can be made into a single file basically, which is perfect for a docker container, or to use Go which also makes it into a single binary. 
 
```bash
┌─ 🖼🖌️ 🎨♬ 🌸#️⃣ 🔮
│ ~/AndroidStudioProjects/Test 
└࿓❯ javac ExecuteCommandTest.java 
┌─ 🖼🖌️ 🎨♬ 🌸#️⃣ 🔮
│ ~/AndroidStudioProjects/Test 
└࿓❯ java ExecuteCommandTest 
PING google.com(ams16s32-in-x0e.1e100.net (2a00:1450:400e:80c::200e)) 56 data bytes
64 bytes from ams16s32-in-x0e.1e100.net (2a00:1450:400e:80c::200e): icmp_seq=1 ttl=118 time=10.9 ms
64 bytes from ams16s32-in-x0e.1e100.net (2a00:1450:400e:80c::200e): icmp_seq=2 ttl=118 time=9.91 ms

--- google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 2ms
rtt min/avg/max/mdev = 9.905/10.407/10.910/0.512 ms

```

ZendaiOwl, 2022-10-05:
Since the web interface already implements an API to execute the scripts, perhaps putting them all together with the web interface and mounting the volumes from MariaDB & Nextcloud will work

szaimen, 2022-10-14:
> @szaimen Sounds good to me. That's basically how I imagined that it could work. 👍

Done. Should work now :)

E.g. `nextcloud/all-in-one:20221014_110407-latest-arm64` for arm64 and `nextcloud/all-in-one:20221014_110407-latest` for amd64

szaimen, 2022-10-14:
BTW, https://github.com/Nitrokey/nextbox does have an interesting architecture as well. Maybe it is worth having a look at that project :)

sunjam, 2022-10-14:
I looked at it in person.  It uses a USB connected spinning disk.  Same
basic layout as ourselves, but already setup for the user.

szaimen, 2022-10-14:
> I looked at it in person. It uses a USB connected spinning disk. Same basic layout as ourselves, but already setup for the user.

Thanks for the insights! :)

However I meant the software-architecture ;)

sunjam, 2022-10-14:
All I remember offhand is it is built on a minimal Debian image.  I'll ask them, as I remember Jan from Nitrokey had questions about our project software architecture as well.

theCalcaholic, 2022-10-14:
Our main challenges are the ones where we differ from NitroKey, though (i. e. integrating the admin web UI and NCP apps into the docker architecture).

theCalcaholic, 2022-10-14:
However, I talked to the NotroKey guys and I'll certainly have a look at their DynDNS integration, which is very neat

szaimen, 2022-10-14:
> Our main challenges are the ones where we differ from NitroKey, though (i. e. integrating the admin web UI and NCP apps into the docker architecture).

true

daringer, 2022-10-18:
> However I meant the software-architecture ;)

I'll shortly outline the main components and ideas:

* Nextcloud (official docker image, Apache) is running from within a `docker-compose` set, which adds MariaDB and Redis
* The Admin UI is build as a Nextcloud App, which has the neat advantage that it can rely on Nextcloud authentification and security. But this also means, that if Nextcloud is not starting/broken one can also not access the NextBox App.
* Updates are introduced via `unattended-updates` and a `nextbox` package brings in all NextBox components
* The backend is a so-called `nextbox-daemon`, which is a systemd service 
* Communication between the NextBox App and the daemon is done through a forwarding mechanism through Nextcloud to the daemon through a socket and a REST API

The dynamic DNS integration is based on deSEC, they provide an API which we utilize and as they also provide a certbot hook there is no need for a port forwarding as authentication is done through a DNS entry during the certbot run.

szaimen, 2022-11-10:
@theCalcaholic Since we talked about this: it is starting with v3.0.0 of AIO possible to disable imagick in AIO. See https://github.com/nextcloud/all-in-one#how-to-add-packets-permanently-to-the-nextcloud-container and https://github.com/nextcloud/all-in-one#how-to-add-php-extensions-permanently-to-the-nextcloud-container

theCalcaholic, 2022-11-10:
@szaimen Yeah, I saw this and it's great news! I already tried (unsuccessfully) to set it up with my private instance :)

szaimen, 2022-11-10:
> @szaimen Yeah, I saw this and it's great news! I already tried (unsuccessfully) to set it up with my private instance :)

What exactly did you try unsuccessfully? Feel free to open a new thread here :)
https://github.com/nextcloud/all-in-one/discussions/new?category=questions

szaimen, 2022-11-17:
AIO v3.0.0 stable is released as of today :)

szaimen, 2023-02-09:
Hi @theCalcaholic, there is now this guide: https://nextcloud.com/blog/how-to-install-the-nextcloud-all-in-one-on-linux/. Maybe it helps you to figure out what went wrong when you wanted to try it out :)

theCalcaholic, 2023-02-11:
@szaimen Thank you :)

My failed attempt was to integrate the imaginary container with a bare metal installation of nextcloud (so not AIO) though.

However we will soon start to migrate our services to docker one by one and will attempt to integrate AIO in NCP once we're done with that.

So we will probably need these resources within the next months :D

szaimen, 2023-03-07:
> > @szaimen Sounds good to me. That's basically how I imagined that it could work. 👍
> 
> Done. Should work now :)
> 
> E.g. `nextcloud/all-in-one:20221014_110407-latest-arm64` for arm64 and `nextcloud/all-in-one:20221014_110407-latest` for amd64

BTW, I just wanted to mention that the containers are now multiarch since v4.4.0 so e.g. `nextcloud/all-in-one:20221014_110407-latest` will work on both x64 and arm64 :)

szaimen, 2023-04-17:
Hey, I just thought I inform you about an idea that I had a while ago as you may be interested in this: https://github.com/nextcloud/all-in-one/discussions/1581

theCalcaholic, 2023-04-18:
@szaimen Looks very interesting indeed!

szaimen, 2023-04-19:
> Thank you for sharing!

No problem! :)

> It looks really interesting, allowing containers to be added & configured through json will be incredibly useful

Yes, I thought so as well. That was why I came up with this idea in the first place :D

> I don't know, it's not quite the same but it reminds me of something I've been slowly working on, a Docker dashboard to manage containers and images.
> 
> The backend is somewhat there, excluding swarm functionality at the moment [link](https://github.com/ZendaiOwl/Build/tree/master/Rust%2Fhttp_rs%2Fbackend%2Fsrc)

Indeed sounds comparable. Actually we are already using such a json for all our containers since the beginning of AIO: https://github.com/nextcloud/all-in-one/blob/main/php/containers.json. The idea would simply exposing this functionality for further containers that could be made and added by the community :)

ZendaiOwl, 2024-04-07:
@theCalcaholic Since you mentioned you are doing technical exploration I'll close this issue :)

closed 2024-04-07 by ZendaiOwl

szaimen, 2025-02-11:
Looks like some work started here: https://github.com/nextcloud-atomic