nextcloud/nextcloudpi #1590: "Latest" does not pull latest image, but "v1.50.1" does

sunjam, 2022-09-28:
When pulling on docker-compose `latest` fails but `v1.50.1` succeeds to pull the image.

ZendaiOwl, 2022-09-28:
Should work as shown with my test above :pray:

ZendaiOwl, 2022-09-28:
I'm re-opening this issue primarily to make it more visible, so users unfamiliar with GitHub have an easier time to locate it and the information above :pray: we can close it again once the docker issues are fully resolved and have calmed down

ZendaiOwl, 2022-09-28:
I found another issue on that version, wrote to you in Matrix chat

The `:latest` tag is only build using the `amd64` platform tag, use the version itself when pulling from the docker hub to get the correct architecture

theCalcaholic, 2022-09-28:
@ZendaiOwl Thanks, should be fixed now

theCalcaholic, 2022-09-28:
I deleted the v1.50.1 release for now

closed 2022-09-28 by theCalcaholic