nextcloud/nextcloudpi #1591: V1.50.1 Docker - dbdir now /data-ro/database

JDC8077, 2022-09-28:
Watchtower triggered an update from v1.50 to v1.51 8 hours ago which did not go smoothly on existing setup and database.

- Re-installed a fresh system container v1.50.1 (no existing ncp or user data) using my compose.yml which resulted in a functional ncp system.
- System info is now reporting that the dbdir is in the container (/data-ro/database) rather than the ncpdata mounted location.
- The ncpdata mounted location contains a database directory and mariadb files from 27 Sept with timestamp ~23:25 CET
- The /etc/mysql/mariadb.conf.d/90-ncp.cnf in the container reports: datadir = /data-ro/database
[image]

Not clear to me how to redirect this to the ncpdata/database folder.

ZendaiOwl, 2022-09-28:
I have to start working in 8 minutes

You can see how the database is moved in the Dockerfile 

Here

https://github.com/nextcloud/nextcloudpi/blob/master/build/docker/Dockerfile#L49

theCalcaholic, 2022-09-28:
@JDC8077 I'm very sorry for your issue. To fix it, please run the following commands and then restore the latest backup

```sh
sed -i 's|/data-ro|/data|' /etc/mysql/mariadb.conf.d/90-ncp.cnf
chown -R mysql: /data/database
service mariadb restart
```