nextcloud/nextcloudpi #1593: After updating to v1.50.1 HPB service shows "down", nextcloud unreachable

cb-studi, 2022-09-28:
Hi all,
just made an update to v1.50.1 of nextcloudpi on my Raspberry Pi 4B 8GB, as I do the updates regularly. Since I did the update, my nextcloud instance is now unreachable, the browser shows Error 503 "Service unavailable (possibly maintenance mode)" and the desktop client does not connect either. The configuration page, which can be reached via the IP address and port 4443, can be called up, however. Here you can see under System Info that the HPB service is down. Maintenance mode is off, as far as I can see. A reboot doesn't help either, and I always do that after an update anyway. So far, nextcloudpi has always worked without any problems. 

The complete excerpt from ncp-report:

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.1
NextCloudPi image    NextCloudPi_10-08-21
OS                   Debian GNU/Linux 11. 5.10.92-v8+ (aarch64)
automount            no
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      40G/440G
rootfs usage         40G/440G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
Uptime               12min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "192.168.178.24",
            "3": "nextcloudpi",
            "11": "77.20.38.19",
            "1": "192.168.178.24",
            "20": "nextcloudpi.local",
            "": "cloudius3117.ddns.net",
            "21": "cloudius3117.ddns.net",
            "14": "nextcloudpi",
            "2": "cloudius3117.ddns.net",
            "12": "cloudius3117.ddns.net"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
[... 92 lines cut ...]
```

{"reqId":"IZVlH0FeLjb0H6fktK6a","level":4,"time":"2022-09-28T11:07:39+00:00","remoteAddr":"","user":"--","app":"privacy","method":"","url":"--","message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","userAgent":"--","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":207,"function":"class_exists"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":172,"function":"isBootable","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":140,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":119,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away"}}
{"reqId":"IZVlH0FeLjb0H6fktK6a","level":4,"time":"2022-09-28T11:07:39+00:00","remoteAddr":"","user":"--","app":"serverinfo","method":"","url":"--","message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","userAgent":"--","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"Redis server /var/run/redis/redis.sock:0 went away","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/autoloader.php","line":154,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"function":"load","class":"OC\\Autoloader","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":207,"function":"class_exists"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":172,"function":"isBootable","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":140,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":119,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away"}}
{"reqId":"XoQ6gKlWKf0fPmkl8Mg6","level":4,"time":"2022-09-28T11:20:55+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Could not boot accessibility: read error on connection to /var/run/redis/redis.sock:0","userAgent":"--","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"read error on connection to /var/run/redis/redis.sock:0","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":48,"function":"get","class":"Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/CachingRouter.php","line":54,"function":"get","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/var/www/nextcloud/lib/private/URLGenerator.php","line":91,"function":"generate","class":"OC\\Route\\CachingRouter","type":"->"},{"file":"/var/www/nextcloud/apps/accessibility/lib/AppInfo/Application.php","line":79,"function":"linkToRoute","class":"OC\\URLGenerator","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":67,"function":"injectCss","class":"OCA\\Accessibility\\AppInfo\\Application","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/BootContext.php","line":51,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->"},{"file":"/var/www/nextcloud/apps/accessibility/lib/AppInfo/Application.php","line":58,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\BootContext","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":190,"function":"boot","class":"OCA\\Accessibility\\AppInfo\\Application","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":207,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":140,"function":"loadApp","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/lib/private/Console/Application.php","line":119,"function":"loadApps","class":"OC_App","type":"::"},{"file":"/var/www/nextcloud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Could not boot accessibility: read error on connection to /var/run/redis/redis.sock:0","exception":{},"Cust [... cut ...] loud/console.php","line":99,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"message":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away","exception":{},"CustomMessage":"Error during app loading: Redis server /var/run/redis/redis.sock:0 went away"}}

```
</details>

theCalcaholic, 2022-09-28:
Hi, seems like there's an issue with your redis server. Please post the output of

```sh
service redis-server status
sudo cat /var/log/redis/redis-server.log
```

cb-studi, 2022-09-28:
Thank you for the answer @theCatholic, here is the output:

pi@nextcloudpi:~ $ sudo cat /var/log/redis/redis-server.log
2052:M 25 Sep 2022 00:05:01.980 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:05:01.981 * Background saving started by pid 159209
159209:C 25 Sep 2022 00:05:02.073 * DB saved on disk
159209:C 25 Sep 2022 00:05:02.075 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:05:02.082 * Background saving terminated with success
2052:M 25 Sep 2022 00:10:03.048 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:10:03.049 * Background saving started by pid 159688
159688:C 25 Sep 2022 00:10:03.132 * DB saved on disk
159688:C 25 Sep 2022 00:10:03.134 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:10:03.150 * Background saving terminated with success
2052:M 25 Sep 2022 00:15:04.098 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:15:04.100 * Background saving started by pid 160559
160559:C 25 Sep 2022 00:15:04.183 * DB saved on disk
160559:C 25 Sep 2022 00:15:04.185 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:15:04.200 * Background saving terminated with success
2052:M 25 Sep 2022 00:20:05.094 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:20:05.095 * Background saving started by pid 200612
200612:C 25 Sep 2022 00:20:05.178 * DB saved on disk
200612:C 25 Sep 2022 00:20:05.180 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:20:05.196 * Background saving terminated with success
2052:M 25 Sep 2022 00:30:02.018 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:30:02.019 * Background saving started by pid 200935
200935:C 25 Sep 2022 00:30:02.112 * DB saved on disk
200935:C 25 Sep 2022 00:30:02.113 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:30:02.119 * Background saving terminated with success
2052:M 25 Sep 2022 00:40:01.534 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:40:01.536 * Background saving started by pid 201321
201321:C 25 Sep 2022 00:40:01.653 * DB saved on disk
201321:C 25 Sep 2022 00:40:01.655 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:40:01.738 * Background saving terminated with success
2052:M 25 Sep 2022 00:50:01.871 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 00:50:01.872 * Background saving started by pid 201462
201462:C 25 Sep 2022 00:50:01.966 * DB saved on disk
201462:C 25 Sep 2022 00:50:01.968 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 00:50:01.972 * Background saving terminated with success
2052:M 25 Sep 2022 01:00:02.303 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:00:02.304 * Background saving started by pid 201787
201787:C 25 Sep 2022 01:00:02.395 * DB saved on disk
201787:C 25 Sep 2022 01:00:02.397 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:00:02.404 * Background saving terminated with success
2052:M 25 Sep 2022 01:10:01.580 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:10:01.581 * Background saving started by pid 202185
202185:C 25 Sep 2022 01:10:01.672 * DB saved on disk
202185:C 25 Sep 2022 01:10:01.674 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:10:01.683 * Background saving terminated with success
2052:M 25 Sep 2022 01:15:02.035 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:15:02.037 * Background saving started by pid 202207
202207:C 25 Sep 2022 01:15:02.172 * DB saved on disk
202207:C 25 Sep 2022 01:15:02.173 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:15:02.238 * Background saving terminated with success
2052:M 25 Sep 2022 01:20:03.004 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:20:03.005 * Background saving started by pid 202236
202236:C 25 Sep 2022 01:20:03.089 * DB saved on disk
202236:C 25 Sep 2022 01:20:03.091 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:20:03.106 * Background saving terminated with success
2052:M 25 Sep 2022 01:25:04.017 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:25:04.018 * Background saving started by pid 202537
202537:C 25 Sep 2022 01:25:04.101 * DB saved on disk
202537:C 25 Sep 2022 01:25:04.102 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:25:04.119 * Background saving terminated with success
2052:M 25 Sep 2022 01:35:01.614 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:35:01.615 * Background saving started by pid 202568
202568:C 25 Sep 2022 01:35:01.708 * DB saved on disk
202568:C 25 Sep 2022 01:35:01.710 * RDB: 0 MB of memory used by copy-on-write
2052:M 25 Sep 2022 01:35:01.716 * Background saving terminated with success
2052:M 25 Sep 2022 01:45:02.108 * 10 changes in 300 seconds. Saving...
2052:M 25 Sep 2022 01:45:02.109 * Background saving started by pid 203000
[... 1817 lines cut ...]
2005:M 26 Sep 2022 19:05:23.270 * Background saving terminated with success
2005:M 26 Sep 2022 19:10:24.063 * 10 changes in 300 seconds. Saving...
2005:M 26 Sep 2022 19:10:24.066 * Background saving started by pid 16866
16866:C 26 Sep 2022 19:10:24.213 * DB saved on disk
16866:C 26 Sep 2022 19:10:24.215 * RDB: 0 MB of memory used by copy-on-write
2005:M 26 Sep 2022 19:10:24.267 * Background saving terminated with success
2005:M 26 Sep 2022 19:15:25.066 * 10 changes in 300 seconds. Saving...
2005:M 26 Sep 2022 19:15:25.068 * Background saving started by pid 16874
16874:C 26 Sep 2022 19:15:25.214 * DB saved on disk
[... cut ...]
2005:M 26 Sep 2022 19:25:27.034 * Background saving started by pid 17201
17201:C 26 Sep 2022 19:25:27.122 * DB saved on disk
17201:C 26 Sep 2022 19:25:27.124 * RDB: 0 MB of memory used by copy-on-write
2005:M 26 Sep 2022 19:25:27.135 * Background saving terminated with success
2005:M 26 Sep 2022 19:30:28.024 * 10 changes in 300 seconds. Saving...
2005:M 26 Sep 2022 19:30:28.025 * Background saving started by pid 17223
17223:C 26 Sep 2022 19:30:28.113 * DB saved on disk
17223:C 26 Sep 2022 19:30:28.115 * RDB: 0 MB of memory used by copy-on-write

theCalcaholic, 2022-09-28:
Hm, nothing out of the ordinary there... Does `sudo service redis-server restart` or a reboot help?

ZendaiOwl, 2022-09-28:
@cb-studi 
Could you check the things described in this forum article? 🙏

https://help.nextcloud.com/t/nextcloudpi-hpb-service-down-after-dist-upgrade/144566/7?u=zendaiowl

cb-studi, 2022-09-28:
@ZendaiOwl Everything is already enabled, the three lines to be added according to this tutorial are also present  
So I did not have to change anything. Everything is as described there, only unfortunately it does not work anymore.

theCalcaholic, 2022-09-28:
@cb-studi Would you be willing to give me ssh access to your ncp instance? If so, please contact me on Matrix (@thecalcaholic:private-cloud.network) or Telegram (theCalcaholic)

theCalcaholic, 2022-09-28:
Unfortunately I did not manage to find the cause of this just yet. I recommend to disable redis, until I managed to provide a fix. To do so, comment the following two lines in /var/www/nextcloud/config/config.php:

```sh
  'memcache.local' => '\\OC\\Memcache\\Redis',
  'memcache.locking' => '\\OC\\Memcache\\Redis',
```
so that they look like this:
```sh
  // 'memcache.local' => '\\OC\\Memcache\\Redis',
  // 'memcache.locking' => '\\OC\\Memcache\\Redis',
```

mteuscher, 2022-09-28:
> Unfortunately I did not manage to find the cause of this just yet. I recommend to disable redis, until I managed to provide a fix. To do so, comment the following two lines in /var/www/nextcloud/config/config.php:
> 
> ```php
>   'memcache.local' => '\\OC\\Memcache\\Redis',
>   'memcache.locking' => '\\OC\\Memcache\\Redis',
> ```
> so that they look like this:
> ```php
>   // 'memcache.local' => '\\OC\\Memcache\\Redis',
>   // 'memcache.locking' => '\\OC\\Memcache\\Redis',
> ```
> Afterwards, reboot (or restart your container)

I had the same Problem and this fixed it for me (for now)

theCalcaholic, 2022-09-28:
@mteuscher Would you mind to leave your hardware information? It might help me to understand the pattern of who is affected

mteuscher, 2022-09-28:
@theCalcaholic 
Raspberry Pi 4 Model B Rev 1.2 - 4 GB 
OS: Debian GNU/Linux 11. 5.15.61-v7l+ (armv7l)

Do you need something else?

Maybe interesting for you: On 1.50.1 (Docker release) everything except HTTPD and PHP was down. After the downgrade to 1.49.1 everything was up, except HPB service according to ncp-report

theCalcaholic, 2022-09-29:
@mteuscher thank you

jiyang1018, 2022-09-29:
> 
I am running docker on Unraid 6.10.2, and experience the same issue. Please let me know what information I can provide to help.
`[2022-09-28 23:35:40 0 [Note] mysqld (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 224 ...
2022-09-28 23:35:40 0 [ERROR] mysqld: File '/data/database/aria_log_control' not found (Errcode: 13 "Permission denied")
2022-09-28 23:35:40 0 [ERROR] mysqld: Got error 'Can't open file' when trying to use aria control file '/data/database/aria_log_control'
2022-09-28 23:35:40 0 [ERROR] Plugin 'Aria' init function returned error.
2022-09-28 23:35:40 0 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
2022-09-28 23:35:40 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
2022-09-28 23:35:40 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
2022-09-28 23:35:40 0 [ERROR] InnoDB: The innodb_system data file 'ibdata1' must be writable
2022-09-28 23:35:40 0 [ERROR] InnoDB: The innodb_system data file 'ibdata1' must be writable
2022-09-28 23:35:40 0 [ERROR] Plugin 'InnoDB' init function returned error.
2022-09-28 23:35:40 0 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
2022-09-28 23:35:40 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-09-28 23:35:40 0 [ERROR] Could not open mysql.plugin table: "Table 'mysql.plugin' doesn't exist". Some plugins may be not loaded
2022-09-28 23:35:40 0 [ERROR] Failed to initialize plugins.
2022-09-28 23:35:40 0 [ERROR] Aborting
`

theCalcaholic, 2022-09-29:
@jiyang1018 the config.php file is inside the container. You can edit it with `docker exec -it nextcloudpi nano /var/www/nextcloud/config/config.php`

mteuscher, 2022-09-29:
@jiyang1018 i changed the file in the <datadirectory>/nextcloud/config/config.php which was also accessible outside the container. You might need to sudo it, to be able to access it.

cb-studi, 2022-09-30:
@theCalcaholic Since I updated my ncp to the newly released v1.50.2, the same error occurs again. My config.php does not even contain the lines anymore you mentioned earlier for a temporary bug fix:

  'memcache.local' => '\\OC\\Memcache\\Redis',
  'memcache.locking' => '\\OC\\Memcache\\Redis',

theCalcaholic, 2022-10-01:
@cb-studi What's your NC version (`ncc --version`)?
Can you share the part from your config where it says `'redis' => array(...)` (without the password, please)?

Also, what's the output of `sudo grep -i redis /var/www/nextcloud/config/config.php` (again, make sure to remove anything that looks like a password)?

KlausKinsky, 2022-10-01:
> Unfortunately I did not manage to find the cause of this just yet. I recommend to disable redis, until I managed to provide a fix. To do so, comment the following two lines in /var/www/nextcloud/config/config.php:
> 
> ```
>   'memcache.local' => '\\OC\\Memcache\\Redis',
>   'memcache.locking' => '\\OC\\Memcache\\Redis',
> ```
> 
> so that they look like this:
> 
> ```
>   // 'memcache.local' => '\\OC\\Memcache\\Redis',
>   // 'memcache.locking' => '\\OC\\Memcache\\Redis',
> ```
> 
> Afterwards, reboot (or restart your container)

I just tried this. I have been running ncp v1.50.2 at least since yesterday and my config.php still contained those lines. Edited + saved, rebooted via web UI -> failed. Rebooted by pulling plug of raspi, worked, restarted normally. HBP service still down :-( situation unchanged. Checked config.php and the 2 lines were gone! Dunno why. Now config.php looks like this

`<?php
 $CONFIG = array (
 'passwordsalt' => '***',
 'secret' => '***',
 'trusted_domains' =>
 array (
 0 => 'localhost',
 5 => 'nextcloudpi.local',
 7 => 'nextcloudpi',
 8 => 'nextcloudpi.lan',
 11 => '***',
 1 => '***',
 3 => '***',
 2 => '***',
 12 => '***',
 22 => 'nextcloudpi',
 '' => '***',
 14 => 'nextcloudpi',
  ),
 'datadirectory' => '/media/NextCloudPi/ncdata',
 'dbtype' => 'mysql',
 'version' => '23.0.6.1',
 'overwrite.cli.url' => '***',
 'dbname' => 'nextcloud',
 'dbhost' => 'localhost',
 'dbport' => '',
 'dbtableprefix' => 'oc_',
 'mysql.utf8mb4' => true,
 'dbuser' => 'ncadmin',
 'dbpassword' => '***',
 'installed' => true,
 'instanceid' => '***',
 'redis' =>
 array (
 'host' => '/var/run/redis/redis.sock',
 'port' => 0,
 'timeout' => 0.0,
 'password' => '***',
  ),
 'tempdirectory' => '/media/NextCloudPi/ncdata/tmp',
 'mail_smtpmode' => 'sendmail',
 'mail_smtpauthtype' => 'LOGIN',
 'mail_from_address' => 'admin',
 'mail_domain' => 'ownyourbits.com',
 'preview_max_x' => '2048',
 'preview_max_y' => '2048',
 'jpeg_quality' => '60',
 'overwriteprotocol' => 'https',
 'maintenance' => false,
 'logfile' => '/media/NextCloudPi/ncdata/nextcloud.log',
 'loglevel' => '2',
 'log_type' => 'file',
 'trusted_proxies' =>
 array (
   11 => '127.0.0.1',
   12 => '::1',
   13 => '***',
   14 => '***',
  ),
 'data-fingerprint' => '***',
); `

I really hope I haven't f**ked something up?!

> @cb-studi What's your NC version (`ncc --version`)? Can you share the part from your config where it says `'redis' => array(...)` (without the password, please)?
>
> Also, what's the output of `sudo [...]

It says 

`pi@nextcloudpi:~ $ ncc --version
Could not open input file: /var/www/nextcloud/occ
`

`pi@nextcloudpi:~ $ sudo grep -i redis /var/www/nextcloud/config/config.php
  'redis' =>
    'host' => '/var/run/redis/redis.sock', `

Nothing more

KlausKinsky, 2022-10-01:
> @cb-studi What's your NC version (`ncc --version`)? Can you share the part from your config where it says `'redis' => array(...)` (without the password, please)?
>
> Also, what's the output of `sudo [...]

Sorry, it wanted a sudo for that comment
`pi@nextcloudpi:~ $ sudo ncc --version
NextCloudPi 23.0.6
`

mteuscher, 2022-10-01:
@theCalcaholic Just FYI: I was on Nectcloud 24.0.3.2 while i had the problems. Today I upgraded ncp (via new docker image) to 1.50.2 and everything worked. Then I updated Nextcloud to 24.0.5.1 and still everything works. So my case is completely fixed

theCalcaholic, 2022-10-01:
@mteuscher That's great to hear!

Does anyone still have this issue on 1.50.2?

KlausKinsky, 2022-10-01:
> @mteuscher That's great to hear!
> 
> Does anyone still have this issue on 1.50.2?

Yes, I still cannot access my cloud. I'm on ncp 1.50.2 and nc 23.06.6 as I stated above. I understand now that my nc version obviously is behind. I have all automatic updates enabled and only update via ncp Web UI. Do I have to update nc manually or will the new nc version soon reach me via automatic updates? I guess it should work then...?
What can I do in the meantime?

KlausKinsky, 2022-10-01:
I'm really worried now: when I go to nc-update-nextcloud in the web UI and start the update it tells me:  
Current Nextcloud version 23.0.6.1
Available Nextcloud version 24.0.5
**ERROR 1146 (42S02) at line 1: Table 'nextcloud.oc_jobs' doesn't exist**

That would actually be the version I need. I guess it's not good that nextcloud.oc_jobs is missing? 

And when I try to nc-update-nc-apps it tells me

In ExceptionConverter.php line 47:

An exception occurred while executing a query: SQLSTATE[42S02]: Base table
or view not found: 1146 Table 'nextcloud.oc_storages' doesn't exist

In Exception.php line 26:

SQLSTATE[42S02]: Base table or view not found: 1146 Table 'nextcloud.oc_sto
rages' doesn't exist

In Statement.php line 92:

SQLSTATE[42S02]: Base table or view not found: 1146 Table 'nextcloud.oc_sto
rages' doesn't exist

app:update [--all] [--showonly] [--allow-unstable] [--] [<app-id>]

So I'm also missing nextcloud.oc_storages? I'm not aware to have done anything to cause this. Can this be solved?

mteuscher, 2022-10-01:
@theCalcaholic  I'm sorry, i totally forgot something: Thank you very very much for your hard work and your help!!!

theCalcaholic, 2022-10-01:
@KlausKinsky Unfortunately that sounds like your database was corrupted by the faulty NCP update. Do you have a recent backup?

If so, please *don't* restore it right away, but move your data directory to a safe location first (there is a bug that your data directory gets overwritten if it's in the default location, even when restoring a dataless backup).

So to be safe, do `mv /data/nextcloud/data /data/data_bk`, then restore the backup, then run `rm -r /data/nextcloud/data; mv /data/data_bk /data/nextcloud/data`.

KlausKinsky, 2022-10-02:
@theCalcaholic Thank you very much! Especially for the hint regarding the database bug. Obviously I should be happy that it worked for almost 2 years without major problems. I'll go for a fresh installation now though.

cb-studi, 2022-10-05:
> @cb-studi What's your NC version (`ncc --version`)? Can you share the part from your config where it says `'redis' => array(...)` (without the password, please)?
>
> Also, what's the output of `sudo [...]

Hi @theCalcaholic 
I had the problem that these commands had stopped working because I had tried to change the SSH user to ncp according to your release note. This resulted in errors that have already been described here. So I decided to do a fresh reinstall with the v1.49.1 image. My cloud is working as it should again. As soon as a new release is out, I will upgrade. Is there anything to be considered then?

theCalcaholic, 2022-10-05:
@cb-studi Alright :)
You should actually be fine then. If you uograde now, you should receive version v1.50.3 and that should work alright

ksdrummer, 2022-10-08:
Hello, 
I have the same problem since today (I think, but I'm not 100% sure since I don't see anything that have been updated today except push_notify, but the error seems to appear before the update).

- NextCloudPi version | v1.50.3
- Nextcloud 24.0.5

It's nextcloudpi running on a raspberry pi 4.

I can't access the Nextcloud instance from the browser (503 Service unavailable), but the ncp Web UI is available. HPB appears as down too.

From the logs : 
`[ nc-previews-auto ]
[ nc-limits ] (Sat 8 Oct 04:00:50 BST 2022)
Using 6143192064 for PHP max memory
System config value trusted_domains => 3 set to string kuluse.mooo.com
Using 6 PHP threads
System config value overwrite.cli.url set to string https://kuluse.mooo.com/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string kuluse.mooo.com
An unhandled exception has been thrown:
RedisException: read error on connection to /var/run/redis/redis.sock:0 in /var/www/nextcloud/lib/private/RedisFactory.php:143
Stack trace:
#0 /var/www/nextcloud/lib/private/RedisFactory.php(143): Redis->auth()
#1 /var/www/nextcloud/lib/private/RedisFactory.php(178): OC\RedisFactory->create()
#2 /var/www/nextcloud/lib/private/Memcache/Redis.php(43): OC\RedisFactory->getInstance()
#3 /var/www/nextcloud/lib/private/Memcache/Factory.php(118): OC\Memcache\Redis->__construct()
#4 /var/www/nextcloud/lib/private/Server.php(1106): OC\Memcache\Factory->createLocking()
#5 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#6 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#7 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#8 /var/www/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#10 /var/www/nextcloud/lib/private/Server.php(2082): OC\AppFramework\Utility\SimpleContainer->get()
#11 /var/www/nextcloud/lib/private/Files/View.php(122): OC\Server->getLockingProvider()
#12 /var/www/nextcloud/lib/private/Server.php(454): OC\Files\View->__construct()
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#14 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#15 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#16 /var/www/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#17 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#18 /var/www/nextcloud/lib/private/Server.php(1445): OC\AppFramework\Utility\SimpleContainer->get()
#19 /var/www/nextcloud/lib/base.php(602): OC\Server->boot()
#20 /var/www/nextcloud/lib/base.php(1111): OC::init()
#21 /var/www/nextcloud/console.php(48): require_once('...')
#22 /var/www/nextcloud/occ(11): require_once('...')
#23 {main}
✓ redis is configured
🗴 can't connect to push server: Server error: `GET https://kuluse.mooo.com/push/test/cookie` resulted in a `503 Service Unavailable` response:

Sat 8 Oct 06:25:02 BST 2022 - Running /etc/cron.daily/ncp-autoupdate...
[ nc-update-nc-apps-auto ]
checking for updates...
notify_push new version available: 0.5.0
notify_push updated
[ncp-update-nc]
Current Nextcloud version 24.0.5.1
Available Nextcloud version 24.0.5
Nothing to update
`

The fix (disabling redis in config.php and reboot) doesn't work for me.

ksdrummer, 2022-10-08:
The redis section after the commented lines : 

```
 'redis' => 
  array (
    'host' => '/var/run/redis/redis.sock',
    'port' => 0,
    'timeout' => 0.0,
    'password' => 'SomePasswooooorrd',
  ),
```

ksdrummer, 2022-10-08:
Something interesting maybe (?), I can access the files from the nextcloud android app, and stream my music from the android dsub app.

br6, 2022-10-10:
I am having similar issues but in docker, specifically php 8.1 incompatibility any chance of a fresh docker image?

ksdrummer, 2022-10-11:
Ok, I solved it by : 
Changing the following values in /etc/php/8.1/fpm/php.ini (I already did that with php7.4 but didn't report it when php updated to 8.1) : 
```
memory_limit = 512M
max_execution_time = 3600
max_input_time = 3600
upload_max_filesize 16G
post_max_size 16G
```

Then (and I think this is what did the trick), I changed the following value in /etc/php/8.1/fpm/pool.d/www.conf (accroding to #1439 and #1624 )
```
pm.max_children = 32
```

Then of course, restart the apache2 and php8.1-fpm services.

ksdrummer, 2022-10-11:
HPB Service is still down though

ksdrummer, 2022-10-11:
Update : After a reboot, I still have the 503 service unreachable problem :'(

fcarp10, 2022-10-15:
The same problem here with HPB service down. The instance is new using docker image v1.50.2 on Rpi 4. After the initial synchronization went fine, the server now responds 503 when modifying a files.

Netrieb, 2022-10-18:
I am on docker NCPi image (v1.50.2 which is the newest right now) with same issue, the above solution (commenting)
```
  // 'memcache.local' => '\\OC\\Memcache\\Redis',
  // 'memcache.locking' => '\\OC\\Memcache\\Redis',
```
works for me. Just the NC Settings>Administration>Overview complains:
`No memory cache has been configured. To enhance performance, please configure a memcache, if available. Further information can be found in the [documentation ↗](https://docs.nextcloud.com/server/24/go.php?to=admin-performance).`

jiyang1018, 2022-10-20:
This version of Nextcloud is not compatible with > PHP 8.0. You are currently running 8.1.10.
When I get to the web UI, mine is set to port 444, I get this message:
This version of Nextcloud is not compatible with > PHP 8.0.
You are currently running 8.1.10.

Log:

```
Starting Redis
Starting Cron
Starting Postfix
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 12543200256 for PHP max memory
Using 8 PHP threads
INFO: Restoring template to default settings
2022-09-30  0:03:19 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2022-09-30  0:03:19 0 [Note] Event Scheduler: Purging the queue. 0 events
2022-09-30  0:03:19 0 [Note] InnoDB: FTS optimize thread exiting.
2022-09-30  0:03:20 0 [Note] InnoDB: Starting shutdown...
2022-09-30  0:03:20 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2022-09-30  0:03:20 0 [Note] InnoDB: Restricted to 4032 pages due to innodb_buf_pool_dump_pct=25
2022-09-30  0:03:20 0 [Note] InnoDB: Buffer pool(s) dump completed at 220930  0:03:20
2022-09-30  0:03:22 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2022-09-30  0:03:22 0 [Note] InnoDB: Shutdown completed; log sequence number 13067456854; transaction id 19688950
2022-09-30  0:03:22 0 [Note] mysqld: Shutdown complete

Stopping MariaDB database server: mariadbd.
Starting MariaDB database server: mariadbd.
Clearing opcache...
This can take some time. Please don't interrupt the process/close your browser tab.
Done.
Stopping redis-server: redis-server.
Starting redis-server: /etc/init.d/redis-server: 51: ulimit: error setting limit (Operation not permitted)
redis-server.
Starting notify_push daemon
Configuring Domain
This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.Starting PHP-fpm
Starting Apache
INFO: Docker container detected.
Starting mariaDB
2022-10-02 23:14:34 0 [Note] mysqld (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 252 ...
2022-10-02 23:14:34 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
2022-10-02 23:14:34 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
2022-10-02 23:14:34 0 [Note] InnoDB: Uses event mutexes
2022-10-02 23:14:34 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-10-02 23:14:34 0 [Note] InnoDB: Number of pools: 1
2022-10-02 23:14:34 0 [Note] InnoDB: Using crc32 + pclmulqdq instructions
2022-10-02 23:14:34 0 [Note] InnoDB: Using Linux native AIO
2022-10-02 23:14:34 0 [Note] InnoDB: Initializing buffer pool, total size = 4160749568, chunk size = 134217728
2022-10-02 23:14:34 0 [Note] InnoDB: Completed initialization of buffer pool
2022-10-02 23:14:34 0 [Note] InnoDB: Starting crash recovery from checkpoint LSN=13067456854,13067456854
2022-10-02 23:14:35 0 [Note] InnoDB: 128 rollback segments are active.
2022-10-02 23:14:35 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2022-10-02 23:14:35 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-10-02 23:14:35 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-10-02 23:14:35 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-10-02 23:14:35 0 [Note] InnoDB: 10.5.15 started; log sequence number 13067456866; transaction id 19688949
2022-10-02 23:14:35 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-10-02 23:14:35 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
Starting Redis
2022-10-02 23:14:35 0 [Note] Server socket created on IP: '127.0.0.1'.
Starting Cron
Starting Postfix
2022-10-02 23:14:36 0 [Note] Reading of all Master_info entries succeeded
2022-10-02 23:14:36 0 [Note] Added new Master_info '' to hash table
[... 100 lines cut ...]
#2 /data/nextcloud/lib/private/Memcache/Redis.php(46): OC\RedisFactory->getInstance()
#3 /data/nextcloud/lib/private/Memcache/Factory.php(120): OC\Memcache\Redis->__construct()
#4 /data/nextcloud/lib/private/Server.php(1080): OC\Memcache\Factory->createLocking()
#5 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#6 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#7 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#8 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#9 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#10 /data/nextcloud/lib/private/Server.php(2040): OC\AppFramework\Utility\SimpleContainer->get()
#11 /data/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#12 /data/nextcloud/lib/private/Server.php(439): OC\Files\View->__construct()
#13 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
[... cut ...]
#18 /data/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
#19 /data/nextcloud/lib/base.php(594): OC\Server->boot()
#20 /data/nextcloud/lib/base.php(1087): OC::init()
#21 /data/nextcloud/console.php(48): require_once('/data/nextcloud...')
#22 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#23 {main}
```
I updated from 1.49.1 to latest, and it is doing the same PHP 8.0 error.
Mine is running on Unraid 6.11.1 currently. Please let me know what info to provide. Thanks in advance.

PeterPilley, 2022-11-22:
Hi I have just started experiencing this with the docker version I have tried 1.50.2 and am now trying a downgrade to 1.48.3 to see what if anything the cause can be.

Is there any idea on what causes this or how to resolve

PeterPilley, 2022-11-22:
> 

I can see in the apache2/error.log 
[Wed Nov 23 10:03:07.528165 2022] [authz_host:error] [pid 120:tid 140203221001984] [client 172.80.15.1:58996] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name

Netrieb, 2022-11-23:
For me this helps: `sudo docker exec -it CONTAINER_NAME service redis-server restart` (perhaps followed by container restart). In my case (on fresh 1.50.2 installation) it is a reaction to "_RedisException:_ ..." (eg. _WRONGPASS invalid username-password pair_ etc.) in nextcloud.log

Netrieb, 2022-11-29:
> For me this helps: `sudo docker exec -it CONTAINER_NAME service redis-server restart` (perhaps followed by container restart). In my case (on fresh 1.50.2 installation) it is a reaction to "_RedisException:_ ..." (eg. _WRONGPASS invalid username-password pair_ etc.) in nextcloud.log

Well it doesn't help anymore. It is like black magic, sometime it worked, sometime it didn't, with no clues in nextcloud.log. Now it stopped working completely, so I gave up. I do also have Raspberry Pi 4 as mentioned above, maybe it is related to it. Now I am trying similar project - Nextcloud All In One container, I'll see if it works for me.

theCalcaholic, 2023-01-21:
This issue seems to have accumulated a bunch of different and unrelated issues by now. Because of this, because the issue of the original issue creator has been solved and because of the age of this issue, I'm closing it now.

If you have still issues, please ask for help at https://help.nextcloud.com or create a new issue if none exists for your problem yet.

closed 2023-01-21 by theCalcaholic