nextcloud/nextcloudpi #1595: RaspberryPi Image1.50.1 - It's not possible to login as user pi (This account is currently not available)

noisie86, 2022-09-28:
Hi all,
after a complete new setup with image 1.50.1 for raspi, it is neither possible to get access via ssh nor to login directly in terminal as pi user.
I use Raspberry Pi 4B 8GB with 32GB flash.

When i tried to login with user pi, system tells "This account is currently not available"

Wrote the image several times new on flash, with previous image version it works!

theCalcaholic, 2022-09-28:
Yes, the pi user has been disabled for security reasons. Please use the ncp user to login (after activating via https://nextcloudpi.local) :)

noisie86, 2022-09-28:
Thanks for quick response @theCalcaholic,
but even after activating and using ncp user with generated pw I'm not able to login!
Same reply: "This account is currently not available"

Is there any way to login in terminal to change things like mount devices or other configs?

thanks in advanced.

ZendaiOwl, 2022-09-28:
I think @theCalcaholic meant that you should use the NCP user to login in the browser. You can configure ssh with the ssh part in the admin web interface and activate ssh for the pi user by changing password, iirc

noisie86, 2022-09-28:
I'm still not able to login as user pi or ncp, even after activating ssh in the nc panel admin settings.
[image]

Even when I'm trying to connected directly to raspi with monitor and keybord its not possible to login as user pi or any other way ;-(

Steps after first boot after image installation:
-> Activate NCP via https://nextcloudpi.local)https://nextcloudpi.local/ web interface
-> Go through the ncp first installation wizard
-> Activate ssh in ncp network setting (User pi, new password test1234)
-> Reboot raspi

Its a clean new image installation, i am wondering that others dont have this behavior.
No clue what is going wrong there, would appreciate all helpful ideas...

theCalcaholic, 2022-09-29:
Alright, I'll investigate this. Others don't have this issue, because it was a fairly recent change :)

pcomputerhelp, 2022-09-29:
On the first boot can't log in as user: pi and password: raspberry. It looks like this [image]

ZendaiOwl, 2022-09-30:
@theCalcaholic 
@pcomputerhelp 

The `pi` user has been set with `/no/login` shell, without any other user to use instead after initial install it seems 🙏

https://help.nextcloud.com/t/nextcloudpi-ssh-enable-not-working-probably/146444/3?u=zendaiowl

theCalcaholic, 2022-09-30:
> @theCalcaholic @pcomputerhelp
> 
> The `pi` user has been set with `/no/login` shell, without any other user to use instead after initial install it seems pray
> 
> https://help.nextcloud.com/t/nextcloudpi-ssh-enable-not-working-probably/146444/3?u=zendaiowl

Oh :D
That's unfortunate... I'll remove the last two RPI releases until this is fixed

ZendaiOwl, 2022-09-30:
Good question 🤔 .. could maybe mount the ISO and check /etc/passwd and see. Working today tho, not mistaken this time 😂, so can't really check for a while

theCalcaholic, 2022-09-30:
> I'm still not able to login as user pi or ncp, even after activating ssh in the nc panel admin settings. [image]
>
> Even when I'm trying to connected directly to raspi with monitor and keybord its not [...]

You would need to use user `ncp` in the SSH section. Then you should be able to login as that user

theCalcaholic, 2022-09-30:
Yeah, I think the actual issue is that the pi user is configured by default in the SSH section on a raspberry pi.

Fix is prepared in https://github.com/nextcloud/nextcloudpi/pull/1604

theCalcaholic, 2022-09-30:
I added a not on the release page

xp1ode, 2022-10-01:
I don't mean to hijack the thread but is there a way to fix this? It seems my issue is exactly the same with the added issue that even the NCP account doesn't work. I can't login in through SSH with neither pi nor ncp. I gave up trying with pi after finding this thread yesterday but i have been trying with ncp to no avail. I tried with an install on a pi (tried installing 3 times) and now i moved to a VM install on proxmox. No matter how many times i change the pw through the webui and SSH i cannot get past the "this account is currently not available. Connection to XX.XX.XX closed"

nimnananuk, 2022-10-01:
@xp1ode If you are also currently setting up your nextcloud, try using the 1.49 image. I just used this one and everything works. In 1.50 I was also not able to add trusted proxies

theCalcaholic, 2022-10-01:
I'm building a test image with the fix, should be ready in about an hour. Would anyone be willing to test if it resolves the issue?

xp1ode, 2022-10-01:
I installed with the curl script, i'd be willing to test it out.

noisie86, 2022-10-01:
I could test it tot...

noisie86, 2022-10-01:
I could test it too...

noisie86, 2022-10-01:
@theCalcaholic  pi user works again with test image!
But unfortunately ncp user still cannot be activated. (Yes I activated ncp user in pi panel admin section first!)
[image]

ZendaiOwl, 2022-10-01:
Could you please confirm by checking the login shell in `/etc/passwd` for `ncp` perhaps? 🙏

It will say `/no/login` if no login shell is configured. 

Ex. `grep ncp /etc/passwd`

noisie86, 2022-10-02:
Yes /nologin shell is configured for ncp.

theCalcaholic, 2022-10-02:
@noisie86 Thanks a lot for testing! You can just stay on that image. It will update to the next release like a normal release

theCalcaholic, 2022-10-02:
@xp1ode Are you sure you enabled ssh in the admin interface for user ncp?

noisie86, 2022-10-02:
Don’t mention it, happy that guy's like you support and provide stuff like nextcloudpi <3.

xp1ode, 2022-10-03:
> @xp1ode Are you sure you enabled ssh in the admin interface for user ncp?

Yes i'm very certain of it, I am able to change the pw through the webui but when i log in through SSH, it'll accept the password change but it will automatically log me out with the error message that the account is unavailable .

theCalcaholic, 2022-10-03:
The fix is now in master and will be included in the image v1.50.3
@xp1ode You can run an update via the web ui and then reapply your SSH settings (for either user). Afterwards, ssh access should work.

PR #1604 (2022-10-03): SSH.sh: Reenable pi user if selected in SSH
Fixes #1595 by:

- Reenabling pi user if used in ncp-config/SSH
- Setting ncp user as default in ncp-config/SSH
closed 2022-10-03 by theCalcaholic