nextcloud/nextcloudpi #1602: NCP creates backups including data although "include data" is disabled

svenb1234, 2022-09-29:
When CP creates a backup it includes the data folder although including data is disabled.

[image]

Related question. With 1.50.2 a backup is created in /data/docker-startup-backups although a different path is set for backups and this one also includes data.

Including data can easily make a disk get full and thus cause issues.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.14.21-150400.24.21-default (x86_64)
automount            no
USB devices          sdc sde 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      2.9T/3.6T
rootfs usage         18G/30G
swapfile             /dev/md3
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               4days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "***REMOVED SENSITIVE VALUE***",
            "11": "***REMOVED SENSITIVE VALUE***",
            "1": "172.17.0.3",
            "14": "***REMOVED SENSITIVE VALUE***",
            "20": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.5.1",
        "overwrite.cli.url": "https:\/\/yxc.selfhost.eu\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/data\/nextcloud\/data\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "maintenance": true,
        "loglevel": 2,
        "theme": "",
        "twofactor_enforced": "true",
        "twofactor_enforced_groups": [],
        "twofactor_enforced_excluded_groups": [],
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpauth": 1,
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "logfile": "\/data\/nextcloud\/data\/nextcloud.log",
        "data-fingerprint": "e1d03246583073a9c18b6f82ab8dd6e2"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Sep 29 17:44:16.122046 2022] [ssl:warn] [pid 118:tid 140472875134272] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu Sep 29 17:44:16.150868 2022] [ssl:warn] [pid 119:tid 140472875134272] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu Sep 29 17:44:16.157058 2022] [mpm_event:notice] [pid 119:tid 140472875134272] AH00489: Apache/2.4.54 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Thu Sep 29 17:44:16.157159 2022] [core:notice] [pid 119:tid 140472875134272] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

svenb1234, 2022-09-29:
`www-data  1068  0.0  0.0  22840  5832 ?        Sl   17:44   0:00 /var/www/nextcloud/apps/notify_push/bin/x86_64/notify_push --allow-self-signed /var/www
root      1318  0.0  0.0   2428   536 ?        S    17:44   0:00 tail -f -n0 /data/nextcloud/data/nextcloud.log
root      1325  0.0  0.0   4812  3992 ?        S    17:44   0:00 /bin/bash /usr/local/bin/ncp-backup /data/docker-startup-backups no yes 5
root      1395  5.6  0.0   6856  5148 ?        D    17:45   2:17 tar -I pigz -cf /data/docker-startup-backups/nextcloud-bkp_20220929_1664473497.tar.gz n
root      1396  0.0  0.0   2484   564 ?        S    17:45   0:00 /bin/sh -c pigz
root      1397  187  0.0 376172  7596 ?        Sl   17:45  76:16 pigz
www-data  1764  0.0  0.0   2484   568 ?        S    17:53   0:00 sh -c sudo /home/www/ncp-backup-launcher.sh bkp '/data/backups//nextcloud-bkp_20220929_
root      1765  0.0  0.0   7004  4352 ?        S    17:53   0:00 sudo /home/www/ncp-backup-launcher.sh bkp /data/backups//nextcloud-bkp_20220929_1664420
root      1766  0.0  0.0   3820  2684 ?        S    17:53   0:00 /bin/bash /home/www/ncp-backup-launcher.sh bkp /data/backups//nextcloud-bkp_20220929_16
root      1768  6.0  0.0   3984  2332 ?        R    17:53   1:56 tar -I pigz -tf /data/backups//nextcloud-bkp_20220929_1664420403.tar.gz data
root      1769 48.4  0.0  28756  1104 ?        Rl   17:53  15:40 pigz -d`

theCalcaholic, 2022-09-29:
Thanks for the report. That definitely needa to be fixed. You can disable the startup backup by adding the parameter `-e NOBACKUP=true` to your docker run command (before the image name)

theCalcaholic, 2022-09-29:
@svenb1234 Actually, the backups are supposed to contain parts of the data directory, but not the actual files. The point of this is to backup everythin that's related to NC apps, I suppose (apart from actual files that are shown as files in NC). Here's the command that creates the backups:

```sh
tar $compress_arg -cf "$destfile" \
    "$dbbackup" \
    --exclude "$data/.opcache" \
    --exclude "$data/{access,error,nextcloud}.log" \
    --exclude "$data/access.log" \
    --exclude "$data/ncp-update-backups" \
    -C "$(dirname "$datadir"/)" $data \
\
    --exclude "nextcloud/data/*/files/*" \
    --exclude "nextcloud/data/.opcache" \
    --exclude "nextcloud/data/{access,error,nextcloud}.log" \
    --exclude "nextcloud/data/access.log" \
    --exclude "nextcloud/data/appdata_*/preview/*" \
    --exclude "nextcloud/data/*/uploads/*" \
    --exclude "nextcloud/data/*/cache/*" \
    --exclude "nextcloud/data/*/files_trashbin/*" \
    --exclude "nextcloud/data/*/files_versions/*" \
    --exclude "nextcloud/data/ncp-update-backups" \
    --exclude "nextcloud/data/__groupfolders" \
    -C $basedir nextcloud/
```

Can you please recheck, if there is an issue here?

theCalcaholic, 2022-09-29:
To reiterate: "include data" is supposed to mean "include files" - maybe it should be renamed..?

svenb1234, 2022-09-29:
Ah, wait, I think I see my mistake.

svenb1234, 2022-09-29:
User error. Closing.

closed 2022-09-29 by svenb1234

theCalcaholic, 2022-09-29:
Alright, thank you :)