nextcloud/nextcloudpi #1605: On a fresh system sql is missing

swifty99, 2022-10-01:
Thanks for providing this awesome Software

I found a solution, however it did not seem the right way.

I had a buster system and the dist-upgrade failed. No one to blame I am used in setting up ncp again and again while not reading the right document at the right time.

Finally it was time to move to docker. Docker works like a charm already with pihole after a complete SSD erase.
Now time to start ncp.
Download and setup according to docs no problem. Hitting "Activate" would not finish. ncp-report was gibberish.
manually restoring with ncp-config would start hover this was the result:
```
Running nc-restore
check free space...
extracting backup file /data/nc_backups_jk/nextcloud-bkp_20220929_1664413203.tar...
restore files...
Stopping redis-server: redis-server.
Starting redis-server: redis-server.
restore database...
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)
Cleanup...
Done. Press any key...
```

A check on docker and base system showed, mariadb and mysql were totaly missing.

Solution (one or all of em?):
- full-upgrade
- reboot
- install ncp with curl until "downloading nextcloud"
- remove container completely
- reinstall container

hope this helps, it took me some hours...

**System information**
Raspi4 4G, Clean Bullseye on SSD

The output was just gibberish as no connection to db was possible.

theCalcaholic, 2022-10-01:
Hi, thanks for the report.
Can you please include the output of `docker exec nextcloudpi ncp-report` (assuming your container is named nextcloudpi)?

swifty99, 2022-10-02:
sure,

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.15.70-v8+ (aarch64)
automount            no
USB devices          sda
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      384G/917G
rootfs usage         384G/917G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               18:09

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "cloud.DOMAIN.de",
            "1": "172.17.0.3",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "9": "home.DOMAIN.de",
            "10": "cloud.DOMAIN.de",
            "20": "cloud.DOMAIN.de",
            "22": "192.168.1242.50",
            "12": "cloud.DOMAIN.de",
            "13": "home.DOMAIN.de",
            "3": "cloud.DOMAIN.de",
            "23": "home.DOMAIN.de",
            "": "cloud.DOMAIN.de",
            "14": "dbd90180e0e9"
[... 86 lines cut ...]
```

{"reqId":"YzkAlOupvu2nDNUbtCeojwAAANU","level":2,"time":"2022-10-02T03:08:04+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/inicio.shtml","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAlMEuVBnqpuXMBMOQwgAAAIw","level":2,"time":"2022-10-02T03:08:04+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/pools","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAlMEuVBnqpuXMBMOQwwAAAJA","level":2,"time":"2022-10-02T03:08:04+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/__Additional","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAlMEuVBnqpuXMBMOQxAAAAIk","level":2,"time":"2022-10-02T03:08:04+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/menu.aspx","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAlsEuVBnqpuXMBMOQxQAAAIQ","level":2,"time":"2022-10-02T03:08:06+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAl8EuVBnqpuXMBMOQxgAAAJM","level":2,"time":"2022-10-02T03:08:07+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/base.pl","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAmcEuVBnqpuXMBMOQxwAAAIA","level":2,"time":"2022-10-02T03:08:09+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/indice.pl","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkAmsEuVBnqpuXMBMOQyAAAAI8","level":2,"time":"2022-10-02T03:08:10+00:00","remoteAddr":"141.105.66.148","user":"--","app":"core","method":"GET","url":"/localstart.pl","message":"Trusted domain error. \"141.105.66.148\" tried to access using \"i59F52B83.versanet.de\" as host.","userAgent":"curl/7.54.0","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkNm-upvu2nDNUbtCeoqQAAAM8","level":2,"time":"2022-10-02T04:03:39+00:00","remoteAddr":"87.236.176.187","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"87.236.176.187\" tried to access using \"89.245.43.131\" as host.","userAgent":"Mozilla/5.0 (compatible; InternetMeasurement/1.0; +https://internet-measurement.com/)","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzkW78EuVBnqpuXMBMORCwAAAIA","level":2,"time":"2022-10-02T04:43:27+00:00","remoteAddr":"34.77.127.183","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"34.77.127.183\" tried to access using \"89.245.43.131\" as host.","userAgent":"python-requests/2.28.1","ver [... cut ...] .x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzlEIeupvu2nDNUbtCepFQAAANU","level":2,"time":"2022-10-02T07:56:17+00:00","remoteAddr":"64.62.197.77","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"64.62.197.77\" tried to access using \"89.245.43.131\" as host.","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/104.0.5112.102 Safari/537.36","version":"24.0.5.1","data":{"app":"core"}}

```
</details>

theCalcaholic, 2022-10-02:
Hm, that doesn't really tell much. Can you please supply the output of `docker exec nextcloudpi mysqld` as well?

swifty99, 2022-10-03:
yepp:
`Error response from daemon: Container dbd90180e0e9e51e2a4cc3758eab33cfc6996440267e960be18818452e0a9cdf is not running`

theCalcaholic, 2022-10-03:
Sorry, I should have read your first message more thoroughly (I was writing from my phone back then).

What you tried, won't work. You tried to install ncp twice, once on the host (with the curl installer) and once via docker.
Please setup either version on a fresh system and try that again (you can, of course, import your data directory and restore a backup).

swifty99, 2022-10-05:
I guess I will close this,

I did the same thing (I wish i could prove), now it is finally working.
The only difference was installation order after starting with a fresh raspbian 64 lite image:

Working:

1. docker
2. ncp
3. unifi
4. pihole

not working:

1. docker 
2. pihole
3. unifi
4. ncp

super strange,
anyway, thx for having a look, 
bad to use so much time for nextcloud..

cheers
Jan

closed 2022-10-05 by swifty99