nextcloud/nextcloudpi #1608: memcache file locking very unreliable

Hyperspeed1313, 2022-10-02:
With memcache file locking, files of any meaningful size are completely unable to be updated, returning either either a 423 or 503 error. If I fall back to software file locking (remove memcache.locking from the config, keep filelocking.enabled => true,) the server functions completely fine.

When I restarted the container all comments were stripped from config.php, but before that the memcache.locking line read `"memcache.locking" => "\\OC\\Memcache\\Redis",` same as the memcache.local property.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.10.0-0.bpo.8-amd64 (x86_64)
automount            no
USB devices          sdf 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      1.1T/1.8T
rootfs usage         8.2G/218G
swapfile             /dev/sde3
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    24.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "hyperspeed1313.mooo.com",
            "11": "47.185.55.206",
            "1": "172.17.0.5",
            "14": "445a2cab9355"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "24.0.5.1",
        "overwrite.cli.url": "https:\/\/hyperspeed1313.mooo.com\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
[... 77 lines cut ...]
```

{"reqId":"JyFC9znBR0909cYOVAdv","level":0,"time":"2022-10-02T21:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 123","userAgent":"--","version":"24.0.5.1","data":{"app":"cron"}}
{"reqId":"JyFC9znBR0909cYOVAdv","level":0,"time":"2022-10-02T21:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 124","userAgent":"--","version":"24.0.5.1","data":{"app":"cron"}}
{"reqId":"JyFC9znBR0909cYOVAdv","level":0,"time":"2022-10-02T21:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 125","userAgent":"--","version":"24.0.5.1","data":{"app":"cron"}}
{"reqId":"JyFC9znBR0909cYOVAdv","level":0,"time":"2022-10-02T21:00:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"CLI cron call has selected job with ID 126","userAgent":"--","version":"24.0.5.1","data":{"app":"cron"}}
{"reqId":"Yzn72WFrAaV-pOzNFWiU_AAAAAw","level":0,"time":"2022-10-02T21:00:09+00:00","remoteAddr":"192.168.1.1","user":"hyper","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/hyper/Documents/","message":"System is in maintenance mode.","userAgent":"Mozilla/5.0 (Windows) mirall/3.6.0stable-Win64 (build 20220906) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.5.1","exception":{"Exception":"OCA\\DAV\\Exception\\ServerMaintenanceMode","Message":"System is in maintenance mode.","Code":0,"Trace":[{"file":"/data/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/lib/Server.php","line":358,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/remote.php","line":166,"args":["/data/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/data/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":86,"CustomMessage":"--"}}
{"reqId":"Yzn72mFrAaV-pOzNFWiU_QAAAA4","level":0,"time":"2022-10-02T21:00:10+00:00","remoteAddr":"192.168.1.1","user":"hyper","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/hyper/Digital-Daydreamer/","message":"System is in maintenance mode.","userAgent":"Mozilla/5.0 (Windows) mirall/3.6.0stable-Win64 (build 20220906) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.5.1","exception":{"Exception":"OCA\\DAV\\Exception\\ServerMaintenanceMode","Message":"System is in maintenance mode.","Code":0,"Trace":[{"file":"/data/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/lib/Server.php","line":358,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/remote.php","line":166,"args":["/data/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/data/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":86,"CustomMessage":"--"}}
{"reqId":"Yzn72mFrAaV-pOzNFWiU_gAAABA","level":0,"time":"2022-10-02T21:00:10+00:00","remoteAddr":"192.168.1.1","user":"hyper","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/hyper/Pictures/","message":"System is in maintenance mode.","userAgent":"Mozilla/5.0 (Windows) mirall/3.6.0stable-Win64 (build 20220906) (Nextcloud, windows-10.0.19044 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"24.0.5.1","exception":{"Exception":"OCA\\DAV\\Exception\\ServerMaintenanceMode","Message":"System is in maintenance mode.","Code":0,"Trace":[{"file":"/data/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/lib/Server.php","line":358,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/remote.php","line":166,"args":["/data/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/data/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":86,"CustomMessage":"--"}}
[... cut ...] ,"type":"->"},{"file":"/data/nextcloud/apps/dav/lib/Server.php","line":358,"function":"exec","class":"Sabre\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->"},{"file":"/data/nextcloud/remote.php","line":166,"args":["/data/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}

```
</details>

feutl, 2023-03-09:
I am running into the same problem, took me some time to figure out what and why and my endeavor started somewhere else, and I link all of the sites to have a little completeness at the end of the story ;)

I do see issues with the contacts web app - so not contacts show and this is where I  started, at the github repo of the contacts web app. I have not seen any other issues on the installation I have.
I do run the latest NCP Docker Build 1.51.0 and NC 25.0.3

I found this and removed memcache form the config.php and voila all started to work again. What I removed was
```
"memcache.locking" => "\\OC\\Memcache\\Redis"
"memcache.local" => "\\OC\\Memcache\\Redis"
```
If this was too much, let me know. I am now struggling a little with me knowing if Redis is still doing its job now, or did it got useless? `filelocking.enabled => 'true'` is still set, confuses me even more.

I also try to understand the following admin articles
https://docs.nextcloud.com/server/stable/admin_manual/configuration_server/caching_configuration.html#id3
https://docs.nextcloud.com/server/stable/admin_manual/configuration_files/files_locking_transactional.html
and the impact for me now, but not an expert on this at all.

So perhaps this problem can be resolved somehow and perhaps someone helps me do understand the implications right now after removed the two lines mentioned above.

Links
- https://github.com/nextcloud/contacts/issues/3253
- https://github.com/nextcloud/nextcloudpi/issues/1728
- https://github.com/nextcloud/nextcloudpi/issues/1721
- https://help.nextcloud.com/t/contacts-menu-links-to-contacts-app-with-no-contacts/153092
- https://help.nextcloud.com/t/contacts-do-not-show-in-web-app-for-all-users-any-more/156852

feutl, 2023-03-10:
Ok, I turn back - tested removing the memcache option and it did not fix it. I had high hopes, but perhaps it is still somehow a hint?
So from today, the contacts app is again empty showing the error mentioned already every time I do open it. Once again - clueless.