nextcloud/nextcloudpi #1609: docker: backup functionality not configureable

svenb1234, 2022-10-03:
With the latest releases a backup is created on every restart of docker.

This means that one can get a lot of useless backups if one restarts a few times, due to whatever reason.

After a docker restart, nextcloud is available for a few seconds and then goes into maintenance mode without the user having any clue as to why, since the docker start-up backup was not configured or enabled by the user.

The docker backups are not shown in the backup list of the web UI.

The normal backup seems to not work anymore, at least no backups are added to the list.

There is no way to disable "on every start" and return to "daily" or whatever the user thinks is useful.

Expected behaviour: make the functionality configureable and integrate it with the normal backup.

- [ ] add option "on start" in the backup module for docker installations. Enable/disable
- [ ] enable the user to run normal backups on docker, i.e. every X days
- [ ] make sure that "on start-up" and "every X days" backup do not run at the same time
- [ ] Show docker backups in the list
- [ ] Add a hint that "on start" will cause nextcloud to not be available for some time after every restart. If data would be included, every restart would be delayed by quite a bit.

BTW, why is this a docker only feature and what's the reason to not backup docker installations the same way asnot containerised installations?

theCalcaholic, 2022-10-03:
I can't reproduce that normally created backups are not visible in the admin interface. If that's the case for you, it might be an unrelated bug

svenb1234, 2022-10-04:
> startup backups are just an additional precaution to reduce the risk of losing data during image updates. data backups are not included, as those could become very large and take hours to complete.

Then the feature is buggy because it is forces abackup on the user on every _restart_ of the container, not only when the image is updated.

Further in that case it does not make sense to keep several backups and fill-up space without the user's knowledge. How many start-up backups are kept?

theCalcaholic, 2022-10-04:
This is not a bug, it is working as intended. There is currently no way to detect whether the image has been updated.

If you want to avoid these backups, you are free to disable them.

This is really mostly a feature to protect users from themselves (i. e. users who didn't setup a proper backup strategy).

theCalcaholic, 2022-10-04:
This is not a bug, it is working as intended. There is currently no way to detect whether the image has been updated.

If you want to avoid these backups, you are free to disable them.

This is really mostly a feature to protect users from themselves (i. e. users who didn't setup a proper backup strategy).

I might include an option in ncp-config, though, to disable the startup backups in order to make it more accessible to users who don't know about the environment variable.

luismauriz1, 2022-11-07:
In my case the Backup is 80GB Big with a 5-pers Nextcloud and not a big amount of data! It seems that is saves the data as well ;)

HenrysCat, 2023-01-05:
> @svenb1234 You can disable the backups by adding the parameter `-e NOBACKUP=true` It is a docker only feature, because docker containers are upgraded by replacing the container and this gives some additional security. On non-docker installations, backups are created automatically during updates, therefore this is not needed. You can use the normal backup system on docker installations, just like on all other variants of ncp - it has not been altered.
> 
> startup backups are just an additional precaution to reduce the risk of losing data during image updates. data backups are not included, as those could become very large and take hours to complete.

Where do we add '-e NOBACKUP=true' to prevent this? one backup and my SD card is full now, pointless feature really.

DanielVolz, 2023-01-13:
@HenrysCat "-e NOBACKUP=true" can be added to a docker run command. 

Or if you are using a docker-compose.yml you can add it like that:

```yaml
version: '3'
services:
  nextcloudpi:
    image: ownyourbits/nextcloudpi-arm64:latest
    volumes:
      - /ext_hdd/nextcloud64:/data
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    container_name: nextcloudpi64
    environment: # this is the relevant part
      - NOBACKUP=true
```