nextcloud/nextcloudpi #1615: After upgrade to 1.50.3 nc-update-nc-apps gives error

GaryGate15, 2022-10-04:
After upgrade to 1.50.3 the command nc-update-nc-apps gives error:
```
Running nc-update-nc-apps
/usr/local/bin/ncc: line 3: 27282 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
Done. Press any key...
```
Clearing opcache did not change situation, nor did reboot.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncc: line 3: 32144 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
NextCloudPi version  v1.50.3
NextCloudPi image    NextCloudPi_11-24-20
OS                   Debian GNU/Linux 11. 5.15.61-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /media/MX500/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      76G/427G
rootfs usage         9.1G/31G
swapfile             /var/swap
dbdir                /media/MX500/ncdatabase
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4 & ipv6)
Port check 443       open (ipv4 & ipv6)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               3:43

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48: 32264 Segmentation fault      sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue Oct 04 16:46:50.532296 2022] [ssl:error] [pid 740:tid 548618482752] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: 72A81754EF8CC672655F45B8D3A04332C1FFB0E0 / notbefore: Nov 24 05:04:15 2020 GMT / notafter: Nov 22 05:04:15 2030 GMT]
[Tue Oct 04 16:46:50.532385 2022] [ssl:error] [pid 740:tid 548618482752] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Tue Oct 04 16:46:50.597465 2022] [ssl:warn] [pid 828:tid 548618482752] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue Oct 04 16:46:50.597844 2022] [ssl:error] [pid 828:tid 548618482752] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: 72A81754EF8CC672655F45B8D3A04332C1FFB0E0 / notbefore: Nov 24 05:04:15 2020 GMT / notafter: Nov 22 05:04:15 2030 GMT]
[Tue Oct 04 16:46:50.604765 2022] [ssl:error] [pid 828:tid 548618482752] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[... 40 lines cut ...]
```

{"reqId":"Yzwououuz9LfzczZttKS5QAAAE8","level":2,"time":"2022-10-04T12:36:10+00:00","remoteAddr":"71.6.167.142","user":"--","app":"core","method":"GET","url":"/sitemap.xml","message":"Trusted domain error. \"71.6.167.142\" tried to access using \"92.132.29.30\" as host.","userAgent":"--","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzwou9XBk8jUm4v8S52ofQAAAA4","level":2,"time":"2022-10-04T12:36:11+00:00","remoteAddr":"71.6.167.142","user":"--","app":"core","method":"GET","url":"/.well-known/security.txt","message":"Trusted domain error. \"71.6.167.142\" tried to access using \"92.132.29.30\" as host.","userAgent":"--","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzws9Iuuz9LfzczZttKS9gAAAEo","level":2,"time":"2022-10-04T12:54:12+00:00","remoteAddr":"142.93.144.55","user":"--","app":"core","method":"GET","url":"/ab2g","message":"Trusted domain error. \"142.93.144.55\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzws9dXBk8jUm4v8S52ohgAAAAU","level":2,"time":"2022-10-04T12:54:13+00:00","remoteAddr":"142.93.144.55","user":"--","app":"core","method":"GET","url":"/ab2h","message":"Trusted domain error. \"142.93.144.55\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzws-Iuuz9LfzczZttKS9wAAAFE","level":2,"time":"2022-10-04T12:54:16+00:00","remoteAddr":"142.93.144.55","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"142.93.144.55\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzwyBIuuz9LfzczZttKTFwAAAFQ","level":2,"time":"2022-10-04T13:15:48+00:00","remoteAddr":"106.75.134.236","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"106.75.134.236\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 (Windows NT 6.1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.112 Safari/537.36","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzwyD9XBk8jUm4v8S52oqAAAAAE","level":2,"time":"2022-10-04T13:15:59+00:00","remoteAddr":"106.75.134.236","user":"--","app":"core","method":"GET","url":"/js/core/merged-template-prepend.js?v=1add2072","message":"Trusted domain error. \"106.75.134.236\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 (Windows NT 6.1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/49.0.2623.112 Safari/537.36","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzw62Yuuz9LfzczZttKTcwAAAFQ","level":2,"time":"2022-10-04T13:53:29+00:00","remoteAddr":"152.89.196.23","user":"--","app":"core","method":"GET","url":"/remote/fgt_lang?lang=/../../../..//////////dev/cmdb/sslvpn_websession","message":"Trusted domain error. \"152.89.196.23\" tried to access using \"92.132.29.30:443\" as host.","userAgent":"Python-urllib/3.8","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzw_Oouuz9LfzczZttKTmQAAAEY","level":2,"time":"2022-10-04T14:12:10+00:00","remoteAddr":"159.203.29.154","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"159.203.29.154\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxI0qQrPTiBBWWeunPL8AAAAFY","level":2,"time":"2022-10-04T14:53:06+00:00","remoteAddr":"54.147.24.127","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"54.147.24.127\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 (Windows NT 6.2;en-US) AppleWebKit/537.32.36 (KHTML, live Gecko) Chrome/52.0.3058.107 Safari/537.32","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxMgc3BaXYOCqA7KCtOJAAAAAg","level":2,"time":"2022-10-04T15:08:49+00:00","remoteAddr":"201.150.42.134","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"201.150.42.134\" tried to access using \"92.132.29.30\" as host.","userAgent":"libwww-perl/6.67","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxPB6QrPTiBBWWeunPMJgAAAEs","level":2,"time":"2022-10-04T15:19:35+00:00","remoteAddr":"192.241.218.58","user":"--","app":"core","method":"GET","url":"/owa/auth/logon.aspx","message":"Trusted domain error. \"192.241.218.58\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxPis3BaXYOCqA7KCtOPwAAAAw","level":2,"time":"2022-10-04T15:21:46+00:00","remoteAddr":"192.241.212.172","user":"--","app":"core","method":"GET","url":"/ecp/Current/exporttool/microsoft.exchange.ediscovery.exporttool.application","message":"Trusted domain error. \"192.241.212.172\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxQUqQrPTiBBWWeunPMNgAAAFM","level":2,"time":"2022-10-04T15:25:06+00:00","remoteAddr":"192.241.205.22","user":"--","app":"core","method":"GET","url":"/owa/auth/x.js","message":"Trusted domain error. \"192.241.205.22\" tried to access using \"92.132.29.30\" as host.","userAgent":"Mozilla/5.0 zgrab/0.x","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"YzxZWM3BaXYOCqA7KCtOjgAAAAg","level":2,"time":"2022-10-04T16:03:36+00:00","remoteAddr":"5.8.10.202","user":"--","app":"core","method":"GET","url":"/aaa9","message":"Trusted domain error. \"5.8.10.202\" tried to access using \"92.132.29.30:443\" as host.","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTM [... cut ...]
{"reqId":"Yzx046QrPTiBBWWeunPNHAAAAEQ","level":2,"time":"2022-10-04T18:01:07+00:00","remoteAddr":"78.225.59.183","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"78.225.59.183\" tried to access using \"92.132.29.30\" as host.","userAgent":"--","version":"24.0.5.1","data":{"app":"core"}}
{"reqId":"Yzx7y6QrPTiBBWWeunPNdgAAAFE","level":2,"time":"2022-10-04T18:30:35+00:00","remoteAddr":"190.163.172.70","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"190.163.172.70\" tried to access using \"92.132.29.30\" as host.","userAgent":"--","version":"24.0.5.1","data":{"app":"core"}}

```
</details>

sirdrakeistda, 2022-10-08:
I get the segmentation fault error for every occ command in Nextcloud. I'm unsure if this happened with the 1.50.3 update. I searched for the error, but I haven't found a solution. "apc.enable_cli=1" did not fixed this problem for me.

sirdrakeistda, 2022-10-09:
It seems like opcache got damaged. After deleting all files in ncdata/.opcache my nextcloud worked again as expected.

GaryGate15, 2022-10-09:
I ran these two commands today, and after a reboot everything seems to work again.
`sudo bash -c 'source /usr/local/etc/library.sh && clear_opcache' 
sudo service php8.1-fpm restart
`

closed 2022-10-09 by GaryGate15

monsieurhannes, 2022-11-14:
I have the same error and smoe more... (1.50.2 on docker). And it doesn't go away with neither of your solutions. Any other idea?

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncc: line 3:  1564 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.10.147-meson64 (aarch64)
automount            no
USB devices          sda sdb 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      btrfs
data disk usage      2.9T/3.7T
rootfs usage         16G/466G
swapfile             none
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48:  1641 Segmentation fault      sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Nov 14 21:16:53.743304 2022] [proxy_fcgi:error] [pid 122:tid 281472451715456] [client 172.17.0.1:54900] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:16:53.743434 2022] [proxy_fcgi:error] [pid 122:tid 281472451715456] (104)Connection reset by peer: [client 172.17.0.1:54900] AH01075: Error dispatching request to : 
[Mon Nov 14 21:17:07.888856 2022] [proxy_fcgi:error] [pid 122:tid 281472443322752] [client 172.17.0.1:47790] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:17:07.889109 2022] [proxy_fcgi:error] [pid 122:tid 281472443322752] (104)Connection reset by peer: [client 172.17.0.1:47790] AH01075: Error dispatching request to : 
[Mon Nov 14 21:17:07.954622 2022] [proxy_fcgi:error] [pid 122:tid 281472216850816] [client 172.17.0.1:47798] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:17:07.954740 2022] [proxy_fcgi:error] [pid 122:tid 281472216850816] (104)Connection reset by peer: [client 172.17.0.1:47798] AH01075: Error dispatching request to : 
[Mon Nov 14 21:17:20.941121 2022] [proxy_fcgi:error] [pid 121:tid 281472736936320] [client 172.17.0.1:49892] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:17:20.941202 2022] [proxy_fcgi:error] [pid 121:tid 281472736936320] (104)Connection reset by peer: [client 172.17.0.1:49892] AH01075: Error dispatching request to : 
[Mon Nov 14 21:17:37.478203 2022] [proxy_fcgi:error] [pid 121:tid 281472720150912] [client 172.17.0.1:57630] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:17:37.478329 2022] [proxy_fcgi:error] [pid 121:tid 281472720150912] (104)Connection reset by peer: [client 172.17.0.1:57630] AH01075: Error dispatching request to : 
[Mon Nov 14 21:17:44.194243 2022] [proxy_fcgi:error] [pid 121:tid 281472711758208] [client 172.17.0.1:57636] AH01067: Failed to read FastCGI header
[Mon Nov 14 21:17:44.194352 2022] [proxy_fcgi:error] [pid 121:tid 281472711758208] (104)Connection reset by peer: [client 172.17.0.1:57636] AH01075: Error dispatching request to : 
[Mon Nov 14 21:18:07.776201 2022] [proxy_fcgi:error] [pid 121:tid 281472241996160] [client 172.17.0.1:38608] AH01067: Failed to read FastCGI header
[... 24 lines cut ...]
```

{"reqId":"Y3KvKqpZnmwzecOJsvLtpwAAAAA","level":3,"time":"2022-11-14T21:12:10+00:00","remoteAddr":"172.17.0.1","user":"--","app":"core","method":"DELETE","url":"/index.php/apps/bookmarks/public/rest/v2/lock","message":"WRONGPASS invalid username-password pair","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:106.0) Gecko/20100101 Firefox/106.0","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"WRONGPASS invalid username-password pair","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":143,"function":"auth","class":"Redis","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file": [... cut ...]
{"reqId":"QEudtFwrgw9wmBMtgHQ0","level":0,"time":"2022-11-14T21:14:30+00:00","remoteAddr":"","user":"--","app":"drawio","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.5.1","data":{"app":"drawio"}}
{"reqId":"QEudtFwrgw9wmBMtgHQ0","level":0,"time":"2022-11-14T21:14:30+00:00","remoteAddr":"","user":"--","app":"files_mindmap","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.5.1","data":{"app":"files_mindmap"}}

```
</details>

monsieurhannes, 2022-11-15:
> I have the same error and some more... (1.50.2 on docker). And it doesn't go away with neither of your solutions. Any other idea?
> NextCloudPi diagnostics
> Nextcloud configuration
> HTTPd logs
> Database logs
> Nextcloud logs

I was able to solve it by updating all the apps. Seems to run stable now. Cleaning opcache might have helped as well...