nextcloud/nextcloudpi #1629: ncc and occ segmentation fault

ksdrummer, 2022-10-11:
**System information**
Raspberry pi 4 8G

Since a week, I could access the WebUI (error 503, see #1593 ). I was able to fix it but now I have a new error, every time I try to run ncc or occ, I get a segmentation fault : 

``` # ncc notify_push:self-test
/usr/local/bin/ncc: line 3:  6511 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
```

```

```<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncc: line 3:  6401 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
NextCloudPi version  v1.50.3
NextCloudPi image    NextCloudPi_11-27-20
OS                   Debian GNU/Linux 11. 5.15.32-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      297G/470G
rootfs usage         297G/470G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               10min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48:  6485 Segmentation fault      sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue Oct 11 16:42:34.989418 2022] [authz_host:error] [pid 611:tid 547550687616] [client 192.168.1.16:54098] AH01753: access check of 'localhost' to /img/info-white.svg failed, reason: unable to get the remote host name, referer: http://192.168.1.16:4443/css/ncp.css
[Tue Oct 11 16:42:34.989418 2022] [authz_host:error] [pid 611:tid 547567489408] [client 192.168.1.16:54100] AH01753: access check of 'localhost' to /img/wizard.svg failed, reason: unable to get the remote host name, referer: http://192.168.1.16:4443/css/ncp.css
[... 43 lines cut ...]
```

{"reqId":"Y0Ht6aebeC7jde0c5rxeEAAAAEg","level":3,"time":"2022-10-08T21:38:49+00:00","remoteAddr":"37.173.6.115","user":"--","app":"music","method":"GET","url":"/apps/music/subsonic/rest/getCoverArt.view?u=kuluse&p=enc:76633675766e79773872786b&v=1.2.0&c=DSub&id=pl-1","message":"Requested cover not found for OCA\\Music\\Db\\Album entity 72139, coverId=5325","userAgent":"DSub","version":"24.0.5.1","data":{"app":"music"}}
{"reqId":"Y0Ht6aebeC7jde0c5rxeEAAAAEg","level":3,"time":"2022-10-08T21:38:49+00:00","remoteAddr":"37.173.6.115","user":"--","app":"music","method":"GET","url":"/apps/music/subsonic/rest/getCoverArt.view?u=kuluse&p=enc:76633675766e79773872786b&v=1.2.0&c=DSub&id=pl-1","message":"Requested cover not found for OCA\\Music\\Db\\Album entity 71927, coverId=13632","userAgent":"DSub","version":"24.0.5.1","data":{"app":"music"}}
{"reqId":"Y0Ht6aebeC7jde0c5rxeEAAAAEg","level":3,"time":"2022-10-08T21:38:49+00:00","remoteAddr":"37.173.6.115","user":"--","app":"PHP","method":"GET","url":"/apps/music/subsonic/rest/getCoverArt.view?u=kuluse&p=enc:76633675766e79773872786b&v=1.2.0&c=DSub&id=pl-1","message":"Trying to access array offset on value of type null at /var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php#356","userAgent":"DSub","version":"24.0.5.1","exception":{"Exception":"Error","Message":"Trying to access array offset on value of type null at /var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php#356","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":356,"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":379,"function":"OCA\\Music\\Utility\\{closure}","class":"OCA\\Music\\Utility\\CoverHelper","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":92,"function":"createMosaic","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":72,"function":"getCoverMosaic","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Controller/SubsonicController.php","line":430,"function":"getCover","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Controller/SubsonicController.php","line":179,"function":"getCoverArt","class":"OCA\\Music\\Controller\\SubsonicController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":225,"function":"handleRequest","class":"OCA\\Music\\Controller\\SubsonicController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":133,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":172,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":298,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1030,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}
{"reqId":"Y0Ht6aebeC7jde0c5rxeEAAAAEg","level":3,"time":"2022-10-08T21:38:49+00:00","remoteAddr":"37.173.6.115","user":"--","app":"index","method":"GET","url":"/apps/music/subsonic/rest/getCoverArt.view?u=kuluse&p=enc:76633675766e79773872786b&v=1.2.0&c=DSub&id=pl-1","message":"getimagesizefromstring(): Argument #1 ($string) must be of type string, null given in file '/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php' line 356","userAgent":"DSub","version":"24.0.5.1","exception":{"Exception":"Exception","Message":"getimagesizefromstring(): Argument #1 ($string) must be of type string, null given in file '/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php' line 356","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":172,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->"},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":298,"function":"main","class":"OC\\AppFramework\\App","type":"::"},{"file":"/var/www/nextcloud/lib/base.php","line":1030,"function":"match","class":"OC\\Route\\Router","type":"->"},{"file":"/var/www/nextcloud/index.php","line":36,"function":"handleRequest","class":"OC","type":"::"}],"File":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","Line":165,"Previous":{"Exception":"TypeError","Message":"getimagesizefromstring(): Argument #1 ($string) must be of type string, null given","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":356,"function":"getimagesizefromstring"},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":379,"function":"OCA\\Music\\Utility\\{closure}","class":"OCA\\Music\\Utility\\CoverHelper","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":92,"function":"createMosaic","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Utility/CoverHelper.php","line":72,"function":"getCoverMosaic","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Controller/SubsonicController.php","line":430,"function":"getCover","class":"OCA\\Music\\Utility\\CoverHelper","type":"->"},{"file":"/var/www/nextcloud/apps/music/lib/Controller/SubsonicController.php","line":179,"function":"getCoverArt","class":"OCA\\Music\\Controller\\SubsonicController","type":"->"},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":225,"function":"handleRequest","class":"OCA\\Music\\Controller\\SubsonicController","type":"->"},{"file":"/var/www/nextcl [... cut ...] e":"/var/www/nextcloud/lib/private/legacy/OC_Template.php","line":78,"function":"initTemplateEngine","class":"OC_Template","type":"::"},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Template.php","line":324,"function":"__construct","class":"OC_Template","type":"->"},{"file":"/var/www/nextcloud/index.php","line":74,"function":"printExceptionErrorPage","class":"OC_Template","type":"::"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":48,"CustomMessage":"--"}}

```
</details>

monsieurhannes, 2022-10-17:
Hi @ksdrummer , I have the same issue. What helped in your case?
Some background information:
- Odroid N2 with NCP in Docker
- Caddy on system as reverse proxy
- Nextcloud is available, sometimes I get to the login screen, then I get different HTTP errors in the 500-area. Sometimes it changes when I reload the page...

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncc: line 3:  1434 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.10.147-meson64 (aarch64)
automount            no
USB devices          sda sdb 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      btrfs
data disk usage      2.9T/3.7T
rootfs usage         14G/466G
swapfile             none
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        closed
Port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               5:37

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

/usr/local/bin/ncp-report: line 48:  1520 Segmentation fault      sudo -u www-data php /var/www/nextcloud/occ config:list system

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Oct 17 18:44:18.618173 2022] [proxy_fcgi:error] [pid 121:tid 281473223483776] [client 172.17.0.1:34944] AH01067: Failed to read FastCGI header
[Mon Oct 17 18:44:18.618256 2022] [proxy_fcgi:error] [pid 121:tid 281473223483776] (104)Connection reset by peer: [client 172.17.0.1:34944] AH01075: Error dispatching request to : 
[Mon Oct 17 18:45:13.208110 2022] [proxy_fcgi:error] [pid 121:tid 281473206698368] [client 172.17.0.1:35470] AH01067: Failed to read FastCGI header
[Mon Oct 17 18:45:13.208210 2022] [proxy_fcgi:error] [pid 121:tid 281473206698368] (104)Connection reset by peer: [client 172.17.0.1:35470] AH01075: Error dispatching request to : 
[Mon Oct 17 18:45:13.231048 2022] [proxy_fcgi:error] [pid 121:tid 281473181520256] [client 172.17.0.1:35478] AH01067: Failed to read FastCGI header
[Mon Oct 17 18:45:13.231143 2022] [proxy_fcgi:error] [pid 121:tid 281473181520256] (104)Connection reset by peer: [client 172.17.0.1:35478] AH01075: Error dispatching request to : 
[Mon Oct 17 18:45:20.602010 2022] [proxy_fcgi:error] [pid 121:tid 281473198305664] [client 172.17.0.1:35482] AH01067: Failed to read FastCGI header
[Mon Oct 17 18:45:20.602086 2022] [proxy_fcgi:error] [pid 121:tid 281473198305664] (104)Connection reset by peer: [client 172.17.0.1:35482] AH01075: Error dispatching request to : 
[Mon Oct 17 18:46:13.169532 2022] [proxy_fcgi:error] [pid 121:tid 281472938262912] [client 172.17.0.1:52852] AH01067: Failed to read FastCGI header
[... 28 lines cut ...]
```

{"reqId":"Y02hyUWiAgREitu_NtgP0AAAABg","level":3,"time":"2022-10-17T18:41:13+00:00","remoteAddr":"172.17.0.1","user":"--","app":"index","method":"DELETE","url":"/index.php/apps/bookmarks/public/rest/v2/lock","message":"WRONGPASS invalid username-password pair","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:105.0) Gecko/20100101 Firefox/105.0","version":"24.0.5.1","exception":{"Exception":"RedisException","Message":"WRONGPASS invalid username-password pair","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":143,"function":"auth","class":"Redis","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/RedisFactory.php","line":178,"function":"create","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Redis.php","line":43,"function":"getInstance","class":"OC\\RedisFactory","type":"->"},{"file":"/data/nextcloud/lib/private/Memcache/Factory.php","line":118,"function":"__construct","class":"OC\\Memcache\\Redis","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":1106,"function":"createLocking","class":"OC\\Memcache\\Factory","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":162,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":122,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":129,"function":"offsetGet","class":"Pimple\\Container","type":"->"},{"file":"/data/nextcloud/lib/private/ServerContainer.php","line":136,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":57,"function":"query","class":"OC\\ServerContainer","type":"->"},{"file":"/data/nextcloud/lib/private/Server.php","line":2082,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->"},{"file":"/data/nextcloud/l [... cut ...]
{"reqId":"nlJLrG1K4INWor5WZLVl","level":0,"time":"2022-10-17T18:47:38+00:00","remoteAddr":"","user":"--","app":"drawio","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.5.1","data":{"app":"drawio"}}
{"reqId":"nlJLrG1K4INWor5WZLVl","level":0,"time":"2022-10-17T18:47:38+00:00","remoteAddr":"","user":"--","app":"files_mindmap","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"24.0.5.1","data":{"app":"files_mindmap"}}

```
</details>

monsieurhannes, 2022-10-18:
Thank you for your reply @ksdrummer! 
I actually got it running again as well. Not by changing anything in the `php.ini` but I ran `apt update` and `apt upgrade` inside the docker container (which is not recommended usually, I know) But so far my NC works again :)

Maybe something @theCalcaholic is interested to know, as it happened at least for us two.

amongesimpasta, 2022-11-07:
I don't know why but it fixed it for me too, thanks.

madscientist42, 2022-11-18:
Still broken...

Running nc-trusted-domains
/usr/local/bin/ncc: line 3: 64598 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
/usr/local/bin/ncc: line 3: 64601 Segmentation fault      ${SUDO} php /var/www/nextcloud/occ "$@"
Done. Press any key...

As best as I can tell, it's due to something in the latest PHP version that's BROKEN...  Reverting to 7.4 fixes the issues...

ksdrummer, 2022-11-23:
Yeah I still have the issue, but it seems to appear randomly

ksdrummer, 2022-11-23:
I just desperately tried to run ncp-limits and set up everything to default and the segmentation fault is gone for now... Try that and see if it helps

MB-Finski, 2022-12-01:
I've also had this issue for a couple of days now. Ran apt update && apt upgrade and several php 8.1 packages were updated. After that the issue was resolved (for now).

saschabrockel, 2022-12-28:
Same here. I am also running Odroid N2 with NCP in Docker like @monsieurhannes. I also ran `apt update` and `apt upgrade` and it works again. It is so random... it worked before. I didn't updated NCP. I can only say that I have restarted Docker completely because I adjusted my `daemon.json`. Maybe somebody of you with this issue has also changed something there? I enabled TCP port 2375 for an external connection to Docker. But it does not seem to be related to this issue IMHO.

monsieurhannes, 2022-12-28:
Without knowing exactly how everything interconnects between Docker and the host, it might be that the host OS has problems that create problems inside the Docker container? 

I don't know if NCP get's the same kind of focus like before? There haven't been any major updates, commits or pull requests since September - especially for Docker. I know it's a community run project, but it's behaving a bit unstable at the moment and I wonder if I should make plans to switch to one of the other versions. 
However, I'm very thankful for the efforts that everyone puts into this project!

theCalcaholic, 2023-02-07:
This is related to issues with the JIT compiler in PHP 8.1. It should be better with the latest release of NCP (also, the bug that caused php 8.2 to be installed when running `apt upgrade` manully has been fixed).

Please report back, if you still encounter segmentation faults/PHP crashes

closed 2023-02-07 by theCalcaholic

tusch001, 2023-06-20:
Came here bc of same problem, probably after php upgrade to 8.1, even php -v dropped segmentation fault error.
[On Stackoverflow](https://stackoverflow.com/questions/57618320/why-does-php-version-return-segmentation-fault-core-dumped-after-install) was a suggestion to repair install of a FreeBSD server, the recommended command as root was `pkg upgrade -f`.
On my Ubuntu Server installation I was successful with `sudo apt upgrade -f`
Hope that helps.