nextcloud/nextcloudpi #1630: Error "This account is not currently available"

Gratts27, 2022-10-11:
**System information**

Once I enter the 
NextCloud login: pi
password: raspberry

It doesn't do anything but keeps asking for login with a message:
"This account is not currently available"

I saw in other forums it's a bug in the latest version.
My version: 1.50.3

Plinsboorg, 2022-10-13:
Same issue

Plinsboorg, 2022-10-13:
You need to open the web-ui first and enable SSH from there.
It's written in the previous release notes.
I set new password for pi user and it worked fine. 

note from web UI:
In order to enable SSH, the password for user 'pi' can NOT remain set to the default 'raspberry'. 
You HAVE to create a NEW password for 'pi' if you want this program to enable SSH, it will fail if you dont!
The same will happen with user 'root' and password '1234'

theCalcaholic, 2022-10-15:
Yes, the default pi user has been disabled for security reasons and in accordance with non-rapsberry-pi installations.

I'll keep this in mind and will consider providing better documentation on this.

closed 2022-10-15 by theCalcaholic