nextcloud/nextcloudpi #1632: Please don't capitalize the c in Nextcloud 😭

jospoortvliet, 2022-10-13:
Hi team,

I know Nextcloud Pi is a special thing, but the capitalization makes my eyes bleed. Can't we just call it NextcloudPi and not capitalize the c in there? We never did that for Nextcloud and it's just confusing as FSCK to see this used everywhere...

ZendaiOwl, 2022-10-13:
I agree 😅

It's also the way I wrote it on the replicated website at https://nextcloudpi.com 

Have been changing it on the few places I come across it but haven't really had the time to look it over properly

I don't know, I think we talked about this at one point in a video call a couple of us 🤔 but I can't quite recall what was mentioned 🙇

@theCalcaholic do you remember? 🙏

theCalcaholic, 2022-10-13:
Totally valid point, let's change that! 👍

sunjam, 2022-10-14:
Never actually seen a single person write write a capital C.  Where has that been a thing?

theCalcaholic, 2022-10-15:
> Never actually seen a single person write the name with a capital C. Where has that been a thing?

All over the project (including Readme and nextcloudpi.com) :D

theCalcaholic, 2022-10-15:
I've started a PR (#1634) where I recplaced all occurrences of "NextCloudPi", but I still have to check which ones might break anything

schoetju, 2022-10-15:
@sunjam Do you have any smart idea how to change from NextCloudPi to NextcloudPi inside all/many articles of the docs.nextcloudpi.com based on forum-articles [https://help.nextcloud.com/c/ncpdocs/178]  ?
Maybe there is something better than updating articles one by one...?

BTW: Even a banner image is using the capitalized C .... :-)
[image]

michuvon, 2022-10-16:
I originally designed the NCP docs banner with a lower case 'c' but someone suggested to use a upper case 'C'.  (It's still in the ncp docs draft discussion in the forum)

If I have time I will edit the banner. However, I can't promisse it will happen this week, so if you have time feel free to take the files attatched GIMP and change it.
(Not sure if GIMP saves history as well, if yes you might be able just reverse the changes and get to my original version with lower case 'c'.)
I'm on the road right now and only got my phone, so no way to open the GIMP files to see which one is the correct one, so here are all the files I have:

https://drive.proton.me/urls/YAS7F80HW0#aQGwZnzIo2kQ

ZendaiOwl, 2022-10-16:
@michuvon 
You need to pull a branch of the repository and add them in there then make a pull request back to the repository on GitHub, hang on and I'll create a branch for you

jospoortvliet, 2022-10-17:
> Never actually seen a single person write the name with a capital C. Where has that been a thing?
> 
> @jospoortvliet would you be so kind as to re-add us to the new website under https://nextcloud.com/devices/ See the former listing at http://web.archive.org/web/20190714123728/https://nextcloud.com/devices/

Ah, done. It's a bit different as the other projects sell hardware pre-installed, or is that something NextcloudPi does these days?

Thanks all for removing the capital C!

I was reading the page as a colleague asked for a nice way to run Nextcloud at home and others in the company recommended a Raspberry Pi with NextcloudPi - and I was, like, hey, I haven't looked at NextcloudPi for a while, let's see what the status is with regards to the new Raspberry Pi 4 devices. 

I'm personally also thinking a bit of moving from my current old-desktop-turned-server to a Raspberry to save electricity... A Raspberry Pi 4 with 8gb ram would be faster (I guess) than my Core i5 860 downclocked to 1GHZ, and use a lot less power. But I have to figure out how to put all my hard drives into USB cases etc, so it's a project that will take a bit :D

Anyhow, have a great day and thanks all!

michuvon, 2022-10-17:
> @michuvon You need to pull a branch of the repository and add them in there then make a pull request back to the repository on GitHub, hang on and I'll create a branch for you
> 
> Edit: Here, I've created the branch `feature/add-banner-images` that you can clone/fork and add the images to and then make a PR to that one slightly_smiling_face
> 
> https://github.com/nextcloud/nextcloudpi/tree/feature/add-banner-images

Never done that.. so I had to BraveSearch it. But I think I succeed.

ZendaiOwl, 2022-10-17:
@michuvon 
You did good 🥳, I've merged the PR into `feature/add-banner-images` and the images are available there now

https://github.com/nextcloud/nextcloudpi/pull/1640

sunjam, 2022-11-04:
> Hi team,
> 
> I know Nextcloud Pi is a special thing, but the capitalization makes my eyes bleed. Can't we just call it NextcloudPi and not capitalize the c in there? We never did that for Nextcloud and it's just confusing as FSCK to see this used everywhere...

I realized you have done exactly the same thing in your sub-reddit.  :rofl: 
https://www.reddit.com/r/NextCloud/

ZendaiOwl, 2023-01-13:
> @sunjam Do you have any smart idea how to change from NextCloudPi to NextcloudPi inside all/many articles of the docs.nextcloudpi.com based on forum-articles [...]

It would be so great if we could change the text on the forums programmatically .. Unfortunately I think we need an API key for that unless I'm mistaken, if that's the case we'll need to edit each one individually 😥

sunjam, 2023-01-13:
That would be a question for meta.discourse.org
afaik it will be a manual process.

ZendaiOwl, 2023-01-13:
It's fully doable with an API-key with discourse :)

https://docs.discourse.org/#tag/Topics

However I don't know if Nextcloud allows that kind of access on their forum, I know it's possible to register an API key for a web app, which I've tested out before, I don't know if it will get the permissions for editing topics &/or posts tho, probably not without some kind of admin privileges or similar permissions as the user that have access to edit the topic/post in the first place