nextcloud/nextcloudpi #1633: This version of Nextcloud is not compatible with > PHP 8.0. You are currently running 8.1.10.

jiyang1018, 2022-10-13:
I can open panel, mine is set to port 4444, and see this. There isn't much in it.
[image]

When I get to the web UI, mine is set to port 444, I get this message:
This version of Nextcloud is not compatible with > PHP 8.0.
You are currently running 8.1.10.

Log:
```
Starting Redis
Starting Cron
Starting Postfix
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 12543200256 for PHP max memory
Using 8 PHP threads
INFO: Restoring template to default settings
2022-09-30  0:03:19 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2022-09-30  0:03:19 0 [Note] Event Scheduler: Purging the queue. 0 events
2022-09-30  0:03:19 0 [Note] InnoDB: FTS optimize thread exiting.
2022-09-30  0:03:20 0 [Note] InnoDB: Starting shutdown...
2022-09-30  0:03:20 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2022-09-30  0:03:20 0 [Note] InnoDB: Restricted to 4032 pages due to innodb_buf_pool_dump_pct=25
2022-09-30  0:03:20 0 [Note] InnoDB: Buffer pool(s) dump completed at 220930  0:03:20
2022-09-30  0:03:22 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2022-09-30  0:03:22 0 [Note] InnoDB: Shutdown completed; log sequence number 13067456854; transaction id 19688950
2022-09-30  0:03:22 0 [Note] mysqld: Shutdown complete

Stopping MariaDB database server: mariadbd.
Starting MariaDB database server: mariadbd.
Clearing opcache...
This can take some time. Please don't interrupt the process/close your browser tab.
Done.
Stopping redis-server: redis-server.
Starting redis-server: /etc/init.d/redis-server: 51: ulimit: error setting limit (Operation not permitted)
redis-server.
Starting notify_push daemon
Configuring Domain
This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.This version of Nextcloud is not compatible with > PHP 8.0.<br/>You are currently running 8.1.10.Starting PHP-fpm
Starting Apache
INFO: Docker container detected.
Starting mariaDB
2022-10-02 23:14:34 0 [Note] mysqld (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 252 ...
2022-10-02 23:14:34 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
2022-10-02 23:14:34 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
2022-10-02 23:14:34 0 [Note] InnoDB: Uses event mutexes
2022-10-02 23:14:34 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-10-02 23:14:34 0 [Note] InnoDB: Number of pools: 1
2022-10-02 23:14:34 0 [Note] InnoDB: Using crc32 + pclmulqdq instructions
2022-10-02 23:14:34 0 [Note] InnoDB: Using Linux native AIO
2022-10-02 23:14:34 0 [Note] InnoDB: Initializing buffer pool, total size = 4160749568, chunk size = 134217728
2022-10-02 23:14:34 0 [Note] InnoDB: Completed initialization of buffer pool
2022-10-02 23:14:34 0 [Note] InnoDB: Starting crash recovery from checkpoint LSN=13067456854,13067456854
2022-10-02 23:14:35 0 [Note] InnoDB: 128 rollback segments are active.
2022-10-02 23:14:35 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2022-10-02 23:14:35 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-10-02 23:14:35 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-10-02 23:14:35 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-10-02 23:14:35 0 [Note] InnoDB: 10.5.15 started; log sequence number 13067456866; transaction id 19688949
2022-10-02 23:14:35 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-10-02 23:14:35 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
Starting Redis
2022-10-02 23:14:35 0 [Note] Server socket created on IP: '127.0.0.1'.
Starting Cron
Starting Postfix
2022-10-02 23:14:36 0 [Note] Reading of all Master_info entries succeeded
2022-10-02 23:14:36 0 [Note] Added new Master_info '' to hash table
2022-10-02 23:14:36 0 [Note] mysqld: ready for connections.
Version: '10.5.15-MariaDB-0+deb11u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 11
2022-10-02 23:14:37 0 [Note] InnoDB: Buffer pool(s) load completed at 221002 23:14:37
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 12543200256 for PHP max memory
Using 8 PHP threads
INFO: Restoring template to default settings
Stopping redis-server: redis-server.
Starting redis-server: /etc/init.d/redis-server: 51: ulimit: error setting limit (Operation not permitted)
redis-server.
Starting notify_push daemon
Configuring Domain
```

Please let me know what info to provide. Thanks in advance.

theCalcaholic, 2022-10-13:
Please run an earlier docker container (e.g. 1.49.1) and upgrade your nextcloud from there. I'm working on a way to prevent this situation.

theCalcaholic, 2022-10-13:
Compare https://github.com/nextcloud/nextcloudpi/issues/1628

jiyang1018, 2022-10-13:
> Please run an earlier docker container (e.g. 1.49.1) and upgrade your nextcloud from there. I'm working on a way to prevent this situation.

I do not know how to run an earlier container. I got most dockers installed and configured by following tutorials.

jiyang1018, 2022-10-13:
I have have found a way. I put :1.49.1, and it failed. Then I put :v1.49.1 and it's pulling.

jiyang1018, 2022-10-13:
I managed to go back to v1.49.1, and encountered 
```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```
Here is log:

```
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 12544880640 for PHP max memory
bash: /usr/local/etc/ncp-templates/php/90-ncp.ini.sh: No such file or directory
/usr/local/etc/library.sh: line 220: ${3:}: bad substitution
Starting notify_push daemon
Configuring Domain
An unhandled exception has been thrown:
RedisException: WRONGPASS invalid username-password pair in /data/nextcloud/lib/private/RedisFactory.php:130
Stack trace:
#0 /data/nextcloud/lib/private/RedisFactory.php(130): Redis->auth()
#1 /data/nextcloud/lib/private/RedisFactory.php(164): OC\RedisFactory->create()
#2 /data/nextcloud/lib/private/Memcache/Redis.php(46): OC\RedisFactory->getInstance()
#3 /data/nextcloud/lib/private/Memcache/Factory.php(120): OC\Memcache\Redis->__construct()
#4 /data/nextcloud/lib/private/Server.php(1080): OC\Memcache\Factory->createLocking()
#5 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#6 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#7 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#8 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#9 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#10 /data/nextcloud/lib/private/Server.php(2040): OC\AppFramework\Utility\SimpleContainer->get()
#11 /data/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#12 /data/nextcloud/lib/private/Server.php(439): OC\Files\View->__construct()
#13 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#14 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#15 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#16 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#17 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#18 /data/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
#19 /data/nextcloud/lib/base.php(594): OC\Server->boot()
#20 /data/nextcloud/lib/base.php(1087): OC::init()
#21 /data/nextcloud/console.php(48): require_once('/data/nextcloud...')
#22 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#23 {main}An unhandled exception has been thrown:
RedisException: WRONGPASS invalid username-password pair in /data/nextcloud/lib/private/RedisFactory.php:130
Stack trace:
#0 /data/nextcloud/lib/private/RedisFactory.php(130): Redis->auth()
#1 /data/nextcloud/lib/private/RedisFactory.php(164): OC\RedisFactory->create()
#2 /data/nextcloud/lib/private/Memcache/Redis.php(46): OC\RedisFactory->getInstance()
#3 /data/nextcloud/lib/private/Memcache/Factory.php(120): OC\Memcache\Redis->__construct()
#4 /data/nextcloud/lib/private/Server.php(1080): OC\Memcache\Factory->createLocking()
#5 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#6 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#7 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#8 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#9 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#10 /data/nextcloud/lib/private/Server.php(2040): OC\AppFramework\Utility\SimpleContainer->get()
#11 /data/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#12 /data/nextcloud/lib/private/Server.php(439): OC\Files\View->__construct()
#13 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#14 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#15 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#16 /data/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#17 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#18 /data/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
[... 8 lines cut ...]
#1 /data/nextcloud/lib/private/RedisFactory.php(164): OC\RedisFactory->create()
#2 /data/nextcloud/lib/private/Memcache/Redis.php(46): OC\RedisFactory->getInstance()
#3 /data/nextcloud/lib/private/Memcache/Factory.php(120): OC\Memcache\Redis->__construct()
[... cut ...]
#17 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#18 /data/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
#19 /data/nextcloud/lib/base.php(594): OC\Server->boot()
#20 /data/nextcloud/lib/base.php(1087): OC::init()
#21 /data/nextcloud/console.php(48): require_once('/data/nextcloud...')
#22 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#23 {main}
```
I updated from 1.49.1 to latest, and it is doing the same PHP 8.0 error.

jiyang1018, 2022-10-18:
> Compare #1628

Please reopen this. I tried an earlier version, and it is still not working.

servernoise, 2022-11-14:
I have the exact same issue. This issue is still unresolved, so please reopen it.

amongesimpasta, 2023-01-19:
same for me

theCalcaholic, 2023-01-20:
Okay looks like your container was corrupted in some way by the update.
Please try the following:

1. restore from a backup with the image version v1.49.1
2. upgrade nextcloud by running ncp-update-nextcloud from either the web UI or `docker exec -it <container-name> ncp-config`
3. Then stop the container and replace it with the latest version

Sorry for your issues, we urgently need a way to check dependencies on container startup to avoid this kind of situation. While there are only a few cases where this can happen (basically PHP upgrades and OS upgrades), we need to find a way to handle them better for the docker version of NCP.

@ZendaiOwl the `/usr/local/etc/library.sh: line 220: ${3:}: bad substitution` error looks familiar. I found it in [this issue](https://github.com/nextcloud/nextcloudpi/issues/1568), that you were working on - do you remember what exactly caused it and when it was fixed?

Haselhirse, 2023-01-22:
The same happened to me. I moved from a plain ncp installation to docker:v1.48.2, restored my backup successfully and wanted to update to the latest image version (v1.50.5) . I also got the PHP compatibility message, so I removed the 1.50.5 container and went back to the (previously working) 1.48.2 which showed now the server error (Unavailable). I restarted the container and now I get an ERR_CONNECTION REFUSED. The log shows dB connection issues and in config.php the values for dbpassword and redis password are overwritten with 'default'.

facorazza, 2023-01-24:
@jiyang1018 what was your nc version?

jrwagh333, 2023-01-30:
Same here, how is this still an issue months later? 

Updated NC, got the PHP error.
Rolled back to a previous version and now the server is unavailable.

jiyang1018, 2023-02-13:
> @jiyang1018 what was your nc version?
I have tried:
1.50.2
1.49.1

PR #1726 (2023-03-26): Fix various issues related to updating Nextcloud
Fixes #1715
Fixes #1633
Fixes #1694