nextcloud/nextcloudpi #1636: pulling docker image 1.50.2, redis service not found, unable to start mariadb service

Colfenor, 2022-10-14:
Hi unfortunately pulling the current docker image (v 1.50.2) & running it results in services not starting,
mariadb, redis won't start, and interestingly `nano` does not seem to be installed on this image ?

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.50.2
NextCloudPi image    NextCloudPi_docker_09-29-22
OS                   Debian GNU/Linux 11. 5.10.103-v7l+ (armv7l)
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      btrfs
data disk usage      78G/932G
rootfs usage         5.6G/59G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        down
HPB service          down
Postfix service      down
Internet check       ok
public IP            ***REMOVED SENSITIVE VALUE***
public IP            ***REMOVED SENSITIVE VALUE***
Port check 80        open (ipv4)
Port check 443       open (ipv4)
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
Interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
Uptime               16min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data-ro/nextcloud/lib/private/DB/Connection.php:139
Stack trace:
#0 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data-ro/nextcloud/lib/private/DB/Connection.php(261): Doctrine\DBAL\Connection->executeQuery()
#3 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /data-ro/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(281): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data-ro/nextcloud/lib/private/AppConfig.php(411): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data-ro/nextcloud/lib/private/AppConfig.php(176): OC\AppConfig->loadConfigValues()
#7 /data-ro/nextcloud/lib/private/AppConfig.php(367): OC\AppConfig->getApps()
#8 /data-ro/nextcloud/lib/private/legacy/OC_App.php(983): OC\AppConfig->getValues()
#9 /data-ro/nextcloud/lib/private/Server.php(713): OC_App::getAppVersions()
#10 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\Server->OC\{closure}()
#11 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(122): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#13 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(57): OC\ServerContainer->query()
#15 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(184): OC\AppFramework\Utility\SimpleContainer->get()
#16 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(162): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#17 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(129): Pimple\Container->offsetGet()
#19 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
[... 88 lines cut ...]
Starting Apache
INFO: Docker container detected.
Starting mariaDB
2022-10-14 20:49:31 0 [Note] mysqld (mysqld 10.5.15-MariaDB-0+deb11u1) starting as process 248 ...
2022-10-14 20:49:31 0 [Warning] The parameter innodb_file_format is deprecated and has no effect. It may be removed in future releases. See https://mariadb.com/kb/en/library/xtradbinnodb-file-format/
2022-10-14 20:49:31 0 [Warning] The parameter innodb_buffer_pool_instances is deprecated and has no effect.
2022-10-14 20:49:31 0 [Note] InnoDB: Uses event mutexes
2022-10-14 20:49:31 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-10-14 20:49:31 0 [Note] InnoDB: Number of pools: 1
2022-10-14 20:49:31 0 [Note] InnoDB: Using generic crc32 instructions
2022-10-14 20:49:31 0 [Note] mysqld: O_TMPFILE is not supported on /tmp (disabling future attempts)
2022-10-14 20:49:31 0 [Note] InnoDB: Using Linux native AIO
2022-10-14 20:49:31 0 [Note] InnoDB: Initializing buffer pool, total size = 268435456, chunk size = 134217728
2022-10-14 20:49:31 0 [Note] InnoDB: Completed initialization of buffer pool
2022-10-14 20:49:31 0 [ERROR] InnoDB: Upgrade after a crash is not supported. The redo log was created with MariaDB 10.3.36.
2022-10-14 20:49:31 0 [ERROR] InnoDB: Plugin initialization aborted with error Generic error
2022-10-14 20:49:31 0 [Note] InnoDB: Starting shutdown...
2022-10-14 20:49:31 0 [ERROR] Plugin 'InnoDB' init function returned error.
2022-10-14 20:49:31 0 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
2022-10-14 20:49:31 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-10-14 20:49:31 0 [ERROR] Unknown/unsupported storage engine: InnoDB
2022-10-14 20:49:31 0 [ERROR] Aborting

```
</details>

theCalcaholic, 2022-10-15:
The issue is actually that your mariadb database crashed while the last container was running (or stopping). You need to fix this, before the container will start normally. You can run arbitrary commands inside the container with `docker run --rm --entrypoint '' <your usual parameters> ownyourbits/nextcloudpi:latest <command>`.

You could try the advice given here https://bbs.archlinux.org/viewtopic.php?id=259364 (e.g. `docker run --rm --entrypoint '' <your usual parameters> ownyourbits/nextcloudpi:latest mysql_upgrade`).

Colfenor, 2022-10-15:
> The issue is actually that your mariadb database crashed while the last container was running (or stopping). You need to fix this, before the container will start normally. You can run arbitrary [...]

Thanks a lot for your reply :) hm the problem is I cannot get mariadb to run inside the container. After pulling the docker nextcloudpi image from the hub, I started the container using this cmd: 

`docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v my/media/path --name nextcloudpi --ip=my_ip ownyourbits/nextcloudpi:latest my_domain`

after that the `docker logs CONTAINER_ID` cmd reveals that mariadb (v10.5.15) failed to start properly.
Then I spawned a bash shell inside the container and tried to restart mariadb manually without success (`service mariadb start`), interestingly `ls /var/lib/` reveals that there is no mysql folder at all and running `systemctl` yields:
```
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```

> You could try the advice given here https://bbs.archlinux.org/viewtopic.php?id=259364 (e.g. `docker run --rm --entrypoint '' <your usual parameters> ownyourbits/nextcloudpi:latest mysql_upgrade`).

unfortunately mysql_upgrade doesn't work since it requires mariadb to be running

Colfenor, 2022-10-16:
I could fix the issue of the broken mariadb & redis by first backing up and then deleting the folders `ncdata` and `ncdatabase` in the nextcloudpi used mounted external drive path, removing all docker containers and starting off fresh with a new `docker run <various params> ...` command

closed 2023-01-07 by theCalcaholic