nextcloud/nextcloudpi #1638: Update to nextcloud 24.0.6

svenb1234, 2022-10-15:
**Description**  
Update NCP to nextcloud 24.0.6.

The update went well on my docker installation.

Somebody else with a different experience?

Tekgnome-svg, 2022-10-24:
I went ahead and updated to 24.0.6 and php 7.4 to 8.0.24

I'm not sure if it's just me but 24.0.6 feels more responsive. Maybe the PHP upgrade helped page response time?
In the time it took to copy these logs/warnings over I can definitely feel an increase in page load times. It wasn't slow or anything before but it feels extra snappy now.

Everything seems to have gone okay but I do have a system warning and some log warnings. 

`Some columns in the database are missing a conversion to big int. Due to the fact that changing column types on big tables could take some time they were not changed automatically. By running "occ db:convert-filecache-bigint" those pending changes could be applied manually. This operation needs to be made while the instance is offline. For further details read [the documentation page about this ↗](https://docs.nextcloud.com/server/24/go.php?to=admin-bigint-conversion).
file_metadata.id`

As for logs so far I've only got this.
`Error | backup | OCA\Backup\Tools\Exceptions\SignatureException: signature issue`

I'll have to look into these as I didn't have them before.

svenb1234, 2022-11-05:
Update to 24.0.7 seems fawless as well.

svenb1234, 2022-12-09:
Same for the update to 24.0.8

svenb1234, 2023-01-16:
Done with 1.50.5

closed 2023-01-16 by svenb1234