nextcloud/nextcloudpi #1639: Terminal repeatedly closes immediately after nc-passwd command - Rpi4 install with install script

EagleDTW, 2022-10-15:
**Problem**
Terminal on both Rpi and ssh login close immediately

Steps to reproduce...
1. Rpi4 install with install script
2. after changing nextcloud password using the nc-passwd
3. terminal stops working

**System information**
sudo ncp-report non-existant as terminal is not accessible
Rpi4 4GB running Raspbian OS Bullseye

EagleDTW, 2022-10-22:
> **Problem** Terminal on both Rpi and ssh login close immediately
> 
> Steps to reproduce...
> 
> 1. Rpi4 Raspbian OS 64-bit Bullseye install with nextcloudpi install script
> 2. after changing nextcloud password using the nc-passwd
> 3. terminal login on ssh upon reboot says nextcloud up to date and that the connection was closed and terminal from inside the OS screen stops working and closes down immediately after opening for a second.
> 
> **System information** sudo ncp-report non-existant as terminal is not accessible Rpi4 4GB running Raspbian OS Bullseye 64-bit

EagleDTW, 2022-10-30:
Any indication why this happens?

schoetju, 2022-10-31:
What userid are you using for terminal login (SSH or physically connected)?
Are you using userid 
- pi?
- ncp?

nc-passwd takes care of userid ncp only (as fas as i can see) - so i guess you are using ncp then?

EagleDTW, 2022-10-31:
Hi Schoetju,

I'm using both ssh into console with pi account and with logged in directly to pi with pi userid on the HW accessing account, but on the Nextcloud account I'm using ncp userid - do I need to shift anything around there? Every time I finally get to the step where I issue the nc-passwd command upon reboot it automatically closes the terminal after a brief opening for a second. I can probably not issue the command to change the password for the nextcloudpi login and nextcloud login, but was having another issue where the screen with the activation page was not coming up even when I added the ip address to the config.php, thus forcing me to have to issue the nc-passwd command - just to get into the system. Maybe I just need a few pointers to alleviate the workflow that is getting me into trouble, that would be appreciated!

EagleDTW, 2022-11-06:
> What userid are you using for terminal login (SSH or physically connected)? Are you using userid
> 
>     * pi?
> 
>     * ncp?
> 
> 
> nc-passwd takes care of userid ncp only (as fas as i can see) - so i guess you are using ncp then?

Hi Schoetju,

The install script seems to be turning off my terminal window on raspbian and it happens before even logging into nextcloud.

Is there any way to get back the terminal?

philipp2310, 2022-11-06:
Had an old raspbian installation, just installed nextcloud with ssh on top with the script. now my ssh terminal is closing right after login and I can't access my own projects on there anymore..
ftp got broken as well.

Only way seems to be removing the SD and changing some files - but what files was nextcloud messing with? (user pi, kitty ssh)

EagleDTW, 2022-11-06:
> Had an old raspbian installation, just installed nextcloud with ssh on top with the script. now my ssh terminal is closing right after login and I can't access my own projects on there anymore.. ftp [...]

Which files did you replace and get working again...would love to know as this has been very frustrating?

philipp2310, 2022-11-06:
> Which files did you replace and get working again...would love to know as this has been very frustrating?

Nothing so far, sorry. I'm just as frustrated as it bricked not only a new install but the heart of my house automation, assistant, piHole etc.
Will let you know when I figure out what settings are messed up

EagleDTW, 2022-11-06:
> Nothing so far, sorry. I'm just as frustrated as it bricked not only a new install but the heart of my house automation, assistant, piHole etc. Will let you know when I figure out what settings are [...]

Appreciate it...I was thinking that you might have a method there of replacing boot, but not sure that that will bring back terminal as that's part of main OS operation - are you able to look at anything past boot drive inserting the sd into another computer?

philipp2310, 2022-11-06:
Just trying to access it via https://www.paragon-software.com/home/linuxfs-windows/ 10 day trial - otherwise maybe https://www.diskinternals.com/linux-reader/ but I haven't tried so far.
First one at least mounts the whole sd, no idea yet what to look for though..

philipp2310, 2022-11-06:
seems like user "pi" is disabled in the setup process but "ncp" wasn't enabled:
`    # Disable raspberry pi default user
    usermod pi -s /sbin/nologin`
will continue after dinner break..
pi and ncp close the terminal immediatelly.

philipp2310, 2022-11-06:
@EagleDTW workaround: find a way to execute:
`usermod pi -s /bin/bash`
For me it was the personal assistant that has a console in the webUI. If that wouldn't have worked, I would recommend the creation of some .sh file that is executed after boot (would have to look up filename/location etc. I guess you can figure that one out faster than me.. I'm not a linux guy at all)

I think the issue was, that the setup wasn't completely successfully, as there was a later point in the setup that would have re-enabled that user again.

schoetju, 2022-11-07:
> Is there any way to get back the terminal?

I am still not 100% sure, if I understand correctly...
Anyway - it might be related to 
https://help.nextcloud.com/t/how-to-add-user-to-sudoers/147655/8?u=schoetju

At least this Web-UI-SSH trick will reactivate user pi.

EagleDTW, 2022-11-07:
> usermod pi -s /bin/bash

This seems to have fixed the issue!! Many thanks!! Glad you got back up and running too!

closed 2022-11-07 by EagleDTW