nextcloud/nextcloudpi #1641: Open ports check fail due to a url change in ncp-diag file

CarlosCastaneda, 2022-10-18:
It's nearly the very same problem as in #1486

Now, the solution is to replace check by checking in the portchecker's web site. Here is a diff between modified and original ncp-diag file.

```
86,87c86,87
<   readonly ipv4_portcheck_args=(-T2 -t1 -qO- --load-cookies "${tmp_file}" "${port_url}/checking" --post-data "target_ip=${publicIPv4}&port=${port}&_csrf=${token::-1}")
<   readonly ipv6_portcheck_args=(-T2 -t1 -qO- --load-cookies "${tmp_file}" "${port_url}/checking" --post-data "target_ip=${publicIPv6}&port=${port}&_csrf=${token::-1}")
---
>   readonly ipv4_portcheck_args=(-T2 -t1 -qO- --load-cookies "${tmp_file}" "${port_url}/check" --post-data "target_ip=${publicIPv4}&port=${port}&_csrf=${token::-1}")
>   readonly ipv6_portcheck_args=(-T2 -t1 -qO- --load-cookies "${tmp_file}" "${port_url}/check" --post-data "target_ip=${publicIPv6}&port=${port}&_csrf=${token::-1}")

```
This works for me. 🙂

CarlosCastaneda, 2022-10-18:
In addition there seem tbe time related issue, because I ran ncp-diag 3 times with the following port results:
1)

1. Port check 80|closed
Port check 443|open (ipv4)

ZendaiOwl, 2022-10-21:
Fix is in PR below, this issue will be closed automatically when it has been merged

- #1647

CarlosCastaneda, 2022-10-28:
> Fix is in PR below, this issue will be closed automatically when it has been merged
> 
> * [Update ncp-diag - Fix portchecker URL #1647](https://github.com/nextcloud/nextcloudpi/pull/1647)
> 
> Edit: Thanks for reporting it 🙏

You're welcome :-). But there is still another issue: if I call [https://portchecker.co/checking](url) directly port 443 is **always** reported to be open, whereas running ncp-diag the result is nont consistent: sometimes it is closed, sometimes open (see following 2 executions directly one after the other - port 80 is closed in my environment).

**sudo bash ncp-diag**
NextCloudPi version|v1.50.3
NextCloudPi image|NextCloudPi_07-21-19
OS|Raspbian GNU/Linux 11. 5.15.61-v7l+ (armv7l)
automount|yes
USB devices|sda sdb sdc
datadir|/media/myCloudDrive1/ncdata
data in SD|no
data filesystem|btrfs
data disk usage|51G/1.9T
rootfs usage|6.6G/15G
swapfile|/var/swap
dbdir|/var/lib/mysql
Nextcloud check|ok
Nextcloud version|24.0.5.1
HTTPD service|up
PHP service|up
MariaDB service|up
Redis service|up
HPB service|up
Postfix service|up
Internet check|ok
public IPv4|x.xxx.xxx.xxxx
public IPv6|xxxx:xxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx
Port check 80|closed
**Port check 443|closed**
IP|192.168.xxx.xxx
gateway|192.168.xxx.xxx
Interface|eth0
certificates|none
NAT loopback|no
Uptime|4days
pi@nextcloudpi:~ $ **sudo bash ncp-diag**
NextCloudPi version|v1.50.3
NextCloudPi image|NextCloudPi_07-21-19
OS|Raspbian GNU/Linux 11. 5.15.61-v7l+ (armv7l)
automount|yes
USB devices|sda sdb sdc
datadir|/media/myCloudDrive1/ncdata
data in SD|no
data filesystem|btrfs
data disk usage|51G/1.9T
rootfs usage|6.6G/15G
swapfile|/var/swap
dbdir|/var/lib/mysql
Nextcloud check|ok
Nextcloud version|24.0.5.1
HTTPD service|up
PHP service|up
MariaDB service|up
Redis service|up
HPB service|up
Postfix service|up
Internet check|ok
public IPv4|xxx.xxx.xxx.xxx
public IPv6|xxxx:xxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx
Port check 80|closed
**Port check 443|open (ipv6)**
IP|192.168.xxx.xxx
gateway|192.168.xxx.xxx
Interface|eth0
certificates|none
NAT loopback|no
Uptime|4days
pi@nextcloudpi:~ $

PS: I don't want to close this issue, but there are only options on the left side of the comment button (Close with comment, Close as not planned).

CarlosCastaneda, 2022-10-28:
> I can't find the file /bin/ncp-diag to edit as shown here: https://github.com/nextcloud/nextcloudpi/pull/1647/files

On my box the file is located in **/usr/local/bin/ncp-diag**.

LittleAlf, 2022-11-16:
I can confirm that using `portchecker.co` with the changed configuration delivers unreliable results, if consecutive requests are send in a short period of time.

_What I did:_ On the website I manually entered different port numbers very quickly and executed it with \<ENTER>. The result: If I had entered two or three requests very quickly, the server responded with `503 Service Temporarily Unavailabe`. It seems like the programmers of the website did not only change the URL, but included a timer or similar which limits the number of consecutive requests from the same IP address!

I believe we have to live with the fact, that the programmers of `portchecker.co` do everything to prevent non-interactive use of their website. This might not even be that they are really against it. But it is clear also, that payment for Google ads (which is linked into their website) will be for ads people actually see. ;)

**I think we need to move away from `portchecker.co` in order to find a reliable way to check, if the ports are open!** We should not play cat-and-mouse with the programmers all the time. **Why don't we use Linux built-in programs to check for open ports?**

I would start with the simplest way of doing it: The `echo` command! This would be something like
`echo > /dev/tcp/[host]/[port] && echo "Port is open"`

I did a quick trial on my system with the IP address of my NCP server and ports 80 and 443, and it nicely answered "Port is open", trials with non-open ports and other IP addresses.

I would be glad to provide a first code snippet for `ncp-diag`, which would do the port check this way. **But I need others to try it out on their systems to confirm it really works reliably!**

Any comment and suggestion is highly appreciated.

Oberlurch3000, 2022-11-23:
> I made these changes that are proposed in the pull request, and this is the result on the webpanel using v1.50.3 :
> 
> [image]
> 
> before, both ports showed closed.

Same issue here. Checking the port 443 directly via portchecker site confirms the port is open.

ZendaiOwl, 2022-12-01:
Sorry, I've been really busy with finishing my undergraduate degree so I haven't been active in the issues here. 

Yes it does seem the developers of the website are changing things up because people are running repeated request on their site without any sleep timers in-between. This is considered bad etiquette on the web btw and should be avoided 🙂, one day your IP might be permanently banned if you keep doing while(true) loop request on various web services, it is essentially the same as a minor DoS attack after all. 

And yes, they would like people to visit their site so they get the ad revenue as well, which they use to pay for the server and the service. 

The issue with using the internal commands on Linux is that you are only testing the ports on your device and not the ports on your router from outside your network, which was the point of using the portchecker website. 

I've got another 2,5 courses and the graduation exam project remaining, I'll see if I can find something appropriate or set up an alternative when I'm done with my degree

LittleAlf, 2022-12-07:
@ZendaiOwl 
> A follow-up on this:
> 
> I've made a portchecker endpoint with Express.js ...

Thank you very much for taking it up! It does not matter if it takes a little bit longer. But then, the cat-and-mouse game is over.

Now: just concentrate on your studies. All the best.

ZendaiOwl, 2022-12-25:
I've deployed an API that is compatible with both IPv4 & IPv6 that we can use for checking if port 80 & 443 is open for external access and made the changes to `ncp-dig` in the branch [fix/portcheck][branch], I made a little bit of change to the layout in the admin web interface to reduce the amount of requests needed per check from 4 to 2, hopefully that will be alright otherwise we can change it again. 

I used a Sanic web server in a Docker image deployed on https://fly.io with a free tier, please don't abuse it? :pray: 

If we go over 100GB outbound data transfer p/month I will be charged for the excess data, I left a comment in the function as well with some info about it. Thank you :pray: :bow: 

We'll see when/if it gets merged into `devel` and then into `master` or if we'll change it a bit more. Also, Merry Christmas everyone :christmas_tree: :santa: :gift: :gift_heart: & a happy New Year :pray: 

[branch]: https://github.com/nextcloud/nextcloudpi/blob/bc577f911eb4d93460edd5617dabeb51322ee9d3/bin/ncp-diag#L67-L107

ZendaiOwl, 2023-01-22:
Closing this since the new port check app has been implemented (it may still need some tweaking but it's no longer an issue related to the old website and URL)

closed 2023-01-22 by ZendaiOwl