nextcloud/nextcloudpi #1642: Users do not have sudo access (sd image)

sunjam, 2022-10-18:
Users do not have sudo access when using the current SD image.  [See this discussion topic from the forum](https://help.nextcloud.com/t/how-to-add-user-to-sudoers/147655)

Might also be related to [recent changes in Raspberry Pi Bullseye](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/)

ZendaiOwl, 2022-10-21:
Indeed, this is something we need to implement in the installer I think :thinking:  

(By "this" I mean this part ↓) 

> There are also mechanisms to preconfigure an image without using Imager. To set up a user on first boot and bypass the wizard completely, create a file called userconf or userconf.txt in the boot partition of the SD card; this is the part of the SD card which can be seen when it is mounted in a Windows or MacOS computer.
> 
> This file should contain a single line of text, consisting of username:encrypted- password – so your desired username, followed immediately by a colon, followed immediately by an encrypted representation of the password you want to use.
> 
> To generate the encrypted password, the easiest way is to use OpenSSL on a Raspberry Pi that is already running – open a terminal window and enter
> 
> echo 'mypassword' | openssl passwd -6 -stdin
> 
> This will produce what looks like a string of random characters, which is actually an encrypted version of the supplied password.

ZendaiOwl, 2022-10-22:
The question is what username to use instead? And what password to generate? 

It needs to be added to [build-SD-rpi.sh](https://github.com/nextcloud/nextcloudpi/blob/master/build/build-SD-rpi.sh) so we create a user that people can use with sudo access, so they can mange their system

schoetju, 2022-10-31:
Currently - after having flashed the RPI-image - _terminal-access_ is not possible (with physical keyboard/display connected to the RPI).

Only workaround currently is to activate SSH, choose user pi and choose a new password for user pi. 
Good enough for a workaround - however confusing for new users.

From my point of view, terminal-access should be possible again (without any workaround) - right from the start.
Security risk is quite low (it needs physical access to the system). There could be a userid like ncp_admin with a fixed default-password (like ownyourbits or similar).

When SSH gets activated the mandatory "new password" will assure, that the fixed default-password has to be changed - latest when it's possible to logon via network.

theCalcaholic, 2022-12-10:
@schoetju Physical access is a very low barrier for an attacker model that we enable by default for all instances.
I disabled this, because NCP does not require you to login via the terminal *ever*, which means, that the default password could remain active for years without the users' knowledge.

All our documentation explains the access via the activation page, so I don't see why we need to provide an undocumented and insecure secondary access vector.

ZendaiOwl, 2023-01-04:
Better we remove the `pi` user and let the users themselves create one using RPi Imager or `userconf.txt` if they wish to create a user. If they don't create any users they can use the admin web interface to activate SSH access using a new user & password instead. 

I think this is the most optimal way for us to go about it, reducing user variables inside the scripts as well since we no longer use the `pi` user. 

#1682

ZendaiOwl, 2023-01-22:
Closing this since we've removed the references to the pi user and users can now create their own user for login either with the userconf.txt, pi imager or ssh activation in the web interface

closed 2023-01-22 by ZendaiOwl