nextcloud/nextcloudpi #1644: Suggestion: (let's use hyphens when tracking issues and pull requests on github)

sunjam, 2022-10-19:
**Description**  
If you link an issue #1531 plus a hyphen in Github you can actually track the status in real time!!!
Let's do this moving forward!

**Example**  
`- #1531 `
becomes
- #1531 

And you'll be able to check actual status of whether the issue is open / closed / merged at a glance.  :heart: 
Green = Open etc.

szaimen, 2022-10-19:
The same also works for checkboxes ;)
- [ ] https://github.com/nextcloud/nextcloudpi/issues/1531

sunjam, 2022-10-19:
Yes, we already use checkmarks.  But the beauty of simply using the method I shared is it auto-updates.  No one needs to checkmark anything, because the color responds when the linked issue/PR is opened/closed/merged.