nextcloud/nextcloudpi #1645: "It seems like you are running a 32-bit PHP version. Nextcloud needs 64-bit to run well. Please upgrade your OS and PHP to 64-bit!

sunjam, 2022-10-19:
Directly related to 
- #1531 

Need help triaging this!  If it is true I cannot say it is a surprise, but I do not have a 32-bit instance to test personally, nor have I found any linked documentation yet.

- [ ] Can we prevent 32-bit users from being able to proceed with the update process to 25, if this is true?

sunjam, 2022-10-20:
https://docs.nextcloud.com/server/stable/admin_manual/installation/system_requirements.html

> "...However we didnt plan to actively remove 32-bit compatibility but it seems like we did for 25 without knowing as we do not test against 32-bit which is why we removed 32-bit support in the first place."

Got this confirmation that 32-bit support is no more in PHP

ZendaiOwl, 2022-10-20:
It should be possible to prevent an update using an architecture check with dpkg before running the updater for those with 32 bit instances

ZendaiOwl, 2022-10-21:
See 

- #1648

ZendaiOwl, 2022-10-22:
@sunjam @theCalcaholic 

As I am interested in the new RISC-V architecture and have purchased a development board with the new JH1100 CPU (still waiting on it to be shipped around February), I've been keeping an eye on the Wiki page from the Debian developers around this architecture and their Wiki page. 

There is some unfortunate data on that page for all 32-bit users, the current Debian repository is slowly becoming unable to compile the 300,000+ packages available in the Debian package repository. 

Ultimately what this means is that 32bit architecture is slowly "dying" or fading away, since it isn't even capable of compiling the tools anymore. Only around 30-35% of the current packages successfully compile on a 32bit architecture, and it has slowly been decreasing every month for a couple years now and getting to the point where it no longer is viable too keep trying to compile the packages for that architecture.

Note that it says x32 is high on that chart, but that is because the x32ABI is no longer a pure "32bit" architecture, it is in fact a mix of 32bit and 64bit due to a request from Linus Torvalds in August 27, 2011. You can read about it on [this](https://en.wikipedia.org/wiki/X32_ABI) Wikipedia page

[image]

QUOTE from https://wiki.debian.org/RISC-V

> What are the goals of this project in particular?
> 
> In this project the goal is to have Debian ready to install and run on systems implementing a variant of the RISC-V ISA:
> 
>     Software-wise, this port targets the Linux kernel
> 
>     Hardware-wise, the port targets the 64-bit variant, little-endian 
> 
> This ISA variant is the "default flavour" recommended by the designers, and the one that seems to attract more interest for planned implementations that might become available in the next few years (development boards, possible consumer hardware or servers).
> 
> While 32-bit and 128-bit implementations are possible, there are problems with this:
> 
>     In the context of RISC-V design, they have not been explored as deeply, and tools and resources (e.g. simulators, research cores) as not as well studied and adapted;
>     For general purpose computers, the focus shifted to 64-bit for many years already, and there isn't a lot of interest in 32-bit architectures except for specific purposes;
>     32-bit ports in Debian already struggle to compile some large packages of the archive in the last few months/years, a problem that will become worse with time;
>     A 128-bit port is simply not realistic at this time. 

Particularly this line:

> 32-bit ports in Debian already struggle to compile some large packages of the archive in the last few months/years, a problem that will become worse with time

Ultimately what this means, in my interpretation, is that 32bit architecture doesn't have much longer left for general use and will not remain as a viable option for that much longer. 

Do I think 32bit will go away completely? 

No, but I do think it will only remain for specific use-cases (like industrial SOC's/Embedded systems for simpler computing tasks) and not general use, and certainly not for general use as a home-server. 

As such, it has become improbable to support 32bit when even the Debian maintainers are struggling to keep up, and if _they_ drop that support, we will be unable to continue with a 32bit image, as we kind of already are in regards to Nextcloud 25 and PHP 8.0+. 

So it's unfortunate, but in my opinion, supporting 32bit architecture is becoming less and less of a viable option, in fact, it kind of already is non-viable with the amount of packages that doesn't compile on it and the fact that almost none of us that write the code here have a 32bit machine to test things out on to make sure they work.

sunjam, 2022-10-22:
My understanding is Tobias has already stopped producing 32bit images.  Our docs state 32bit users ate only supported by the curl script, which now ends with 25.

ZendaiOwl, 2022-10-22:
Alright :slightly_smiling_face:  I have not had the time to check up on the docs so I wasn't aware of that :pray: 

I'll leave this here as a reference then for people that wonder about why 32bit architecture is being dropped

sunjam, 2022-11-03:
> The fix will be available in 25.0.2
- https://github.com/nextcloud/server/pull/34905

sunjam, 2022-12-09:
Need help testing [25.0.2](https://nextcloud.com/blog/maintenance-releases-25-0-2-24-0-8-and-23-0-12-are-out/), which is now in stable and should re-implement 32-bit support.

ZendaiOwl, 2023-01-04:
Closing this since NC decided to continue with 32bit support :slightly_smiling_face:

closed 2023-01-04 by ZendaiOwl